Marus/cortex-debug #732: Feature request: RTOS tab Zephyr support

rosterloh, 2022-08-31:
I'd be happy to help with this but not sure where to start. Any help would be greatly appreciated.

haneefdm, 2022-08-31:
Thanks. Maybe we should write some notes while you are doing this. I do have a pre-req. You have to create a test project on a board I have (or something I can order cheap). My favorite board is the STM32F429ZI_NUCLEO_144

This is where you start
- https://github.com/Marus/cortex-debug/tree/master/src/frontend/rtos
- Here you will find rtos.ts. Very few changes here will be needed
  - You have to [add your OS to this object map](https://github.com/Marus/cortex-debug/blob/1fdef12b6cd84d50a10cfaf2e727be8548ae7388/src/frontend/rtos/rtos.ts#L11)
- The file rtos-common.ts, is something you have become accustomed to. But, first look at an example rtos-freertos.h (or one of the others) to see how the common functions are used. You don't need many of those common methods but you will need most of the types
   - Here we [define a set of Column IDs](https://github.com/Marus/cortex-debug/blob/1fdef12b6cd84d50a10cfaf2e727be8548ae7388/src/frontend/rtos/rtos-freertos.ts#L10)
   - And there are tables below for how to format them, the type of column it is, etc. Also, [look at ucosii because it uses some percentage-type columns with bar graphs kind of thing](https://github.com/Marus/cortex-debug/pull/642)
   - The tryDetect method in each of the methods does an exploration of what symbols are available. Some may be optional. Start with the required ones so we bail fast. As the debugger/user can be unpredictable, it is designed to keep trying until it knows for sure failure or success.
   - Every time the debugger stops, we either continue our discovery process or get the thread info. We get the thread info by first calling refresh() and that works, we call getHTML(). Imagine the user single stepping real fast. Queries can fail randomly and we have to catch it. This is already done for you

If it was not apparent. All of VSCode is done with HTML/CSS/TypeScript/JavaScript, so is this.

So, it is some work and perhaps whole lot of copy-paste which can be error-prone.

I recommend playing with one of the other OSes. I also put some effort in generating Help in freertos case informing users why certain columns are not populated and what they can do in their FW.

rosterloh, 2022-09-01:
Ok, I'll start working on this. As far as test projects go I don't have a STM32F429ZI_NUCLEO_144 but take your pick from:
 - NUCLEO-L452RE
 - B-L475E-IOT01A2
 - STM32L562E-DK
 - NUCLEO-U575ZI-Q
 
or if you want to stick with the F4 family I have an Adafruit Feather F405 Express

haneefdm, 2022-09-01:
I have none of those boards. The trouble nowadays is availability. I will check for the board once you make some forward progress.

haneefdm, 2022-09-01:
I just ordered the NUCLEO-U575ZI-Q while it is still in stock. People should be donating these. I am buying boards just for testing SW and I have no intention of using them for any other purpose

rosterloh, 2022-09-02:
Hopefully that's not directed at me since I'm also giving of my time to make this better for everyone. I've voted for the VSCode issue. 

I've put together the framework files for this feature. Do you have any recommendations for building, deploying and debugging VSCode extensions?

haneefdm, 2022-09-02:
Not directed at you. Just a random rant :-)

If you could setup a github repo for your test case, put some instructions to others to use it (build instructions, etc.).

You can't deploy publicly and you don't need to deploy privately either. If you followed the instructions on our toplevel README.md, the build/debug instructions are there for the extension.

Note that debugging bad XML is very tricky, so be very careful if you are touching any of the XML/CSS stuff.

haneefdm, 2022-09-02:
Oh, and thanks for voting.

rosterloh, 2022-09-02:
Are there any specific features you'd like to see in a test case application? `west build -b nucleo_u575zi_q samples/philosophers` in the zephyr tree would probably do the trick

haneefdm, 2022-09-02:
No, nothing specific. Just have enough threads. And, if there some compile time features to turn on/off we, have to try those as well. For FreeRTOS, there were so many options and with certain options, some info will be missing and I had to deal with all those options on/off and disable certain columns.

What we don't want is crash because of missing info.

I always had all the possible columns but had `??` for missing info with an explanation why that column had no info and FW changes they could make to see that info.

I don't know anything about Zephyr.

haneefdm, 2022-09-06:
@vrepetenko Please see the instructions and requirements above. Use a board that I have, I prefer STM32F429ZI_NUCLEO_144 but I have a NUCLEO-U575ZI-Q on order.

haneefdm, 2022-09-06:
As an advanced notification. In the coming weeks, I will be removing RTOS module from this repo and making it its own repo and a separate extension. None of your work will be lost but this is set to happen.

The main reason we are doing this is so that it can be separate from the debugger and can support other debuggers, like Microsoft's. Microsoft did the same thing for their RTOS viewer and support our debugger.

vrepetenko, 2022-09-07:
> @vrepetenko Please see the instructions and requirements above. Use a board that I have, I prefer STM32F429ZI_NUCLEO_144 but I have a NUCLEO-U575ZI-Q on order.

@haneefdm thanks for instructions, will take a look! All I need now is to find out how to look up for symbols and read memory. 
Board is not a problem, there is RT-STM32F429ZI-NUCLEO144 demo project in ChibiOS, so there will be no problem with testing on your side.

vrepetenko, 2022-09-07:
> @vrepetenko Please see the instructions and requirements above. Use a board that I have, I prefer STM32F429ZI_NUCLEO_144 but I have a NUCLEO-U575ZI-Q on order.

@haneefdm 
there is no thread call stack in current implementation, right?
like this:
[image]

haneefdm, 2022-09-07:
No, thread call stack. That would be redundant and wasteful. We already get that from cortex-debug/Call-stack panel.

How to get symbols are in the instructions once you start reading the code.

vrepetenko, 2022-09-07:
yes, I found how to get symbols and more in code. Thanks!
Now trying to convert my JLink ChibiOS Plugin to more useful rtos-chibios.ts )

vrepetenko, 2022-09-13:
Hi @haneefdm !
How can I add RTOS detection for JLink GDB or force cortex-debug use local RTOS extension, but not JLink RTOS plugin? 
It seems it works now only for OpenOCD.

haneefdm, 2022-09-13:
You can't force cortex-debug to use local RTOS extension. You force JLink/OpenOCD to report threads using whatever mechanism they got. JLink uses dlls/dylibs/so plugins. We always react to threads if any are reported and that is the gdb-server's job.

For JLink (or OpenOCD), it is documented -- `"rtos": "Zephyr"` in launch.json -- case matters.
https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

Note, our RTOS views should work with or without OpenOCD/JLink even enabling RTOS detection. We do not rely on anything a gdb-server does. Our RTOS view HAS to work without any help from the RTOS support from gdb-server. Or else, we are broken. We rely on gdb for help and it in turn only needs to be able to read memory -- nothing else.

We have found that sometimes, OpenOCD and JLink are not as accurate at detecting RTOSes are we are. Not sure why, but I manually verified what we were reporting was correct. Maybe they don't show threads that are in certain states and we do.

vrepetenko, 2022-09-13:
By local RTOS extension I mean rtos-chibios.ts...
I added rtos-chibios.ts with ChibiOS specific tryDetect, modified rtos.ts, and trying to debug rtos-chibios.ts
But I do not see any call of tryDetect or refresh...
It seems there is some missconfig or something else...

launch.json for testing target:
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "showDevDebugOutput": false,
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/test.elf",
            "name": "Debug (J-Link)",
            "device": "STM32WL55CC",
            "interface": "swd",
            "runToMain": true,
            "preLaunchTask": "build",
            "rtos": "ChibiOS"
        }
    ]
}

Debug console output in Extension Dev Host:
Cortex-Debug: VSCode debugger extension version 1.6.3 git(6285985+dirty). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w test.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p test.elf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
Launching gdb-server: JLinkGDBServer -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32WL55CC -rtos GDBServer/RTOSPlugin_ChibiOS.dylib
    Please check TERMINAL tab (gdb-server) for output from JLinkGDBServer
Finished reading symbols from objdump: Time: 59 ms
Finished reading symbols from nm: Time: 159 ms

haneefdm, 2022-09-13:
Also, RTOS detection does not start until the program pauses. In the command panel, you will RTOS related stuff

[image]

RTOS detection is disabled by default, I think.

vrepetenko, 2022-09-14:
Hi!
It works now, thanks!

[image]

haneefdm, 2022-09-14:
What was wrong?

vrepetenko, 2022-09-14:
RTOS panel was not active

vrepetenko, 2022-09-28:
Hi!
First result, but something strange with table rendering:

[image]

haneefdm, 2022-09-28:
Yup, now you got to the painful part. If anything goes wrong in your HTML. It appears the row did not end

You can save your HTML to disk (should already be a mechanism to do that, but disabled) and examine it in a browser/editor. You may have to edit it a bit.

Also, compare the HTML with an existing RTOS

Also, share your repo

vrepetenko, 2022-09-30:
will check HTML, but it is strange, I did not touch anything related to HTML...

repo: https://github.com/vrepetenko/cortex-debug/tree/feature/ChibiOS-support

haneefdm, 2022-10-07:
Hi People, I am moving the RTOS parts to its own extension with support for multiple debuggers and maybe a better way of doing things. My first task is to get it on par with what we got right now, without touching anything but one file (rtos.ts).

Btw, this issue has gotten corrupted. We went from Zephyr to no ChibiOS and I am talking about splitting extensions. I can't keep my head straight. I will start a new issue

Cc: @PhilippHaefele

vrepetenko, 2022-10-10:
> Hi! First result, but something strange with table rendering:
> 
> [image]

I do not see any problem with HTML, it is well-formatted and can be validated by html checker. maybe some style problem...

simple test rendering:
[image]

PhilippHaefele, 2022-10-10:
@vrepetenko Can you please post the html content (including the whole head, script stuff) here and it's maybe good to open your fork to everyone (so also others (me) can have a look).

vrepetenko, 2022-10-10:
> @vrepetenko Can you please post the html content (including the whole head, script stuff) here and it's maybe good to open your fork to everyone (so also others (me) can have a look).

@PhilippHaefele
fork is here and open to everyone: https://github.com/vrepetenko/cortex-debug/tree/feature/ChibiOS-support

rtos.html: https://gist.github.com/vrepetenko/a2a72b743052c9821d96a8cdea7e794c

PhilippHaefele, 2022-10-10:
> @PhilippHaefele fork is here and open to everyone: https://github.com/vrepetenko/cortex-debug/tree/feature/ChibiOS-support

Sorry seems that the GiHub app has a bug which i use to look into issues

> rtos.html: https://gist.github.com/vrepetenko/a2a72b743052c9821d96a8cdea7e794c

Thanks, i will have a look and maybe also try to debug your changes with an example

PhilippHaefele, 2022-10-10:
Just tested with my NUCLEO-F429ZI board and it seems to work in general

[image]

PhilippHaefele, 2022-10-10:
@vrepetenko Just to be sure. Did you cd into the webview folder and did the `npm install` there too?

haneefdm, 2022-10-10:
@PhilippHaefele That may be it!!

haneefdm, 2022-10-10:
I think I found the problem. Not doing an `npm install` should have caused a compiler error. It doesn't because the top-level package.json also has a reference to webview-ui-toolkit which is not needed.

vrepetenko, 2022-10-10:
> Just tested with my NUCLEO-F429ZI board and it seems to work in general
> 
> [image]

cool! will try npm install and update repo with last changes.
there is a problem with stack size detection in current ChibiOS thread_t implementation, no chance to get stack size/stack start, only stack base (bottom) and free size. New ChibiOS version with stack size is coming, but how should we fill stack top , stack start now? I use SP value for stack start/top now, but sometimes SP can have zero value, is it ok?

vrepetenko, 2022-10-10:
it works fine now. thanks!
[image]

haneefdm, 2022-10-10:
Hi folks, I just updated the package.json's and the extension packaging script so we do not need the webview directory anymore.

Please merge with the current master. There will be one change in rtos.ts

haneefdm, 2022-10-10:
Ooooh wait. Build on CI failed. Not sure what is wrong. A bot did also update our package.json due to security vulnerability.

PhilippHaefele, 2022-10-10:
> Ooooh wait. Build on CI failed. Not sure what is wrong. A bot did also update our package.json due to security vulnerability.

I think we need to remove the webview install step from GitHub action also
https://github.com/Marus/cortex-debug/blob/2e91b0e1f2df1c65af78a93f3672988c0c063e48/.github/workflows/main.yaml#L25

haneefdm, 2022-10-10:
Yes, thanks. Bigger problem is that the runner is forcing me to upgrade version of 'nodejs' -- it is the runner that decided to deprecate the version of node we are using as of this week. Finding some incompatibilities one by one.

haneefdm, 2022-10-10:
Everything passing now. Re-enabled tslint for workspace. Nodejs updated as well.

You should be able to merge with master now.

PhilippHaefele, 2022-10-13:
> it works fine now. thanks!
> 
> [image]

Can you push your code and maybe already create a MR against a his repo? I can test/review it with the ChibiOS release i have setup.

@rosterloh I was able to setup a Zephyr project and when i find a little bit more time, i will start implementing the Zephyr support

@haneefdm Do we finish the ChibiOS here and then manually take it over to the mcu repo? For Zephyr i will do it directly in the new mcu repo/extension.

haneefdm, 2022-10-13:
Yes, lets do the integration here and then I will move it to the new repo, may need a little help testing at that time.

I had to make a TON of changes to upgrade to new versions of VSCode, NodeJS, ESLint, TypeScript, ESBuild, etc. To the point where I don't know if I have broken something :-(

rosterloh, 2022-10-14:
@PhilippHaefele that's amazing. I've been bogged down with work at the moment but do let me know if I can help or test.

vrepetenko, 2022-10-14:
> > it works fine now. thanks!
> > [image]
> 
> @vrepetenko Can you push your code and maybe already create a MR against a his repo? I can test/review it with the ChibiOS release i have setup.
> 
> @rosterloh I was able to setup a Zephyr project and when i find a little bit more time, i will start implementing the Zephyr support
> 
> @haneefdm Do we finish the ChibiOS here and then manually take it over to the mcu repo? For Zephyr i will do it directly in the new mcu repo/extension.

@PhilippHaefele 
hi!

I have created PR. Not sure where to find new repo....

Added stack info:

[image]

haneefdm, 2022-10-14:
If you are looking for the new repo, it is here

Organization: https://github.com/mcu-debug (this will have all the new repos)
Repo: https://github.com/mcu-debug/rtos-views

The new repo is in a state where it is about the same as the cortex-debug. But I want to make some changes today because I don't like what I did to conform to ESLint. The meaning of `object` and `any` is vastly different from before and we lost some type-safety.

But let us finish the PR for ChibiOS here. I noticed you already did one.

haneefdm, 2022-10-14:
I finished my work in the new Repo. I will be moving the new ChibiOS os to over there.

How do I test it? Remember, one of my requirements was to have a project that I can use to test?

vrepetenko, 2022-10-14:
> Okay, I finished moving the ChibiIOS from Cortex-Debug to the new repo. I have no way of testing it.
> 
> I request again, I need a repo with an appropriate project setup for testing. I cannot accept PRs that cannot be maintained. I should be able to build and test on my machine. Please, please use STM32F429ZI_NUCLEO_144 for your test.

You can use this demo: https://github.com/ChibiOS/ChibiOS/tree/master/demos/STM32/RT-STM32F429ZI-NUCLEO144
let me know if you need any additional info

haneefdm, 2022-10-14:
Hi @vrepetenko,

I was hoping for a standalone repo I can use for testing, ready for VSCode, including a proper launch.json, task.json, etc. In any case, it fails.

```
    const currentThreadStats = await this.getVarChildrenObj(currentThread['stats-ref'], 'stats');
```
Returns null and then it crashes. I corrected this somewhat in the new repo. We need to deal with exceptions properly. With the PR (unmodified by me), in cortex-debug, I get no output at all -- because it crashes.

in the https://github.com/mcu-debug/rtos-views where I worked around it, I got something but it does not feel right.

[image]

I created a new Issue in the new repo and we need to work over there.

https://github.com/mcu-debug/rtos-views/issues/2

vrepetenko, 2022-10-15:
Hi @haneefdm ,

sure I can create standalone repo for testing, this problem in todo list, there is no optional variables processing now.

haneefdm, 2022-10-15:
@rosterloh Please know that all RTOS work is now active at https://github.com/mcu-debug/rtos-views and I will be shutting this thread down. Wish I could move it...

haneefdm, 2022-10-18:
No, when you are done, just submit a PR over there.

PhilippHaefele, 2022-11-12:
@rosterloh Got a first working state done here https://github.com/mcu-debug/rtos-views/pull/7 It needs some open points to be fixed before we/you can test it further

@haneefdm As we now have a MR to discuss over at the rtos-views repo, i think we can close this one here

rosterloh, 2022-11-14:
@PhilippHaefele awesome work! I'm looking forward to testing it.

PhilippHaefele, 2022-11-26:
@rosterloh As you have already noticed: There's a first test version out in the [marketplace](https://marketplace.visualstudio.com/items?itemName=mcu-debug.rtos-views&ssr=false#overview) you can test. Still having some open points, but should be good enough for a first test. If you encounter any issues please create a Issue on the RTOS Views repo. Thanks and have fun

@haneefdm I think we can close this one here now

haneefdm, 2022-11-26:
We should probably adjust the top-level README for contributors over at the rtos-views repo.

closed 2022-11-26 by haneefdm