Marus/cortex-debug #734: Variable insight for std::vector variables

RobertoRoos, 2022-09-01:
I cannot see the content of `std::vector<T>` type variables. Is that a limitation by Cortex Debug or by VS Code itself?

For me it would be very useful to see the values inside vectors.

Interestingly I see that `std::array<T, N>` variables can be viewed fully.

Screenshot of my debugger:
[image]
`fifo_data` is of type `std::array<uint8_t, 263>` and `impedances_mag` is `std::vector<float>`.

haneefdm, 2022-09-01:
I don't know. Could you do me a favor and use cppdbg and write a small program for a native desktop and see if they can do that. cppdbg ms Micosts c++ debugger in cpptools

https://marketplace.visualstudio.com/items?itemName=ms-vscode.cpptools

If they can do it, we can see how they did and try to something similar. We just give them a bunch of data to VSCode and they interpret/render it.

haneefdm, 2022-09-01:
I hate to ask for this, but could you do me a favor and look at this and vote for the issue on the MS Website if you agree.

https://github.com/Marus/cortex-debug/issues/729#issuecomment-1233293715

RobertoRoos, 2022-09-05:
Good point. This is the debugger result of `std::vector` on VS code running in Windows using GDB:

[image]

So this is not much better.

I can properly see the vector when I build with VS 2022 and use the VS debugger (debug type "Windows" in VS Code, instead of "cppdbg")

And I voted for the issue you linked to.

haneefdm, 2022-09-05:
No, you can't compare VS 2022 to VSCode. VS is proprietary and does not have to follow any rules and VSCode (us and cppdbg) have to follow a spec that serves as a contract between a debug-adapter like ours (and cppdbg) and VSCode core.

haneefdm, 2022-09-05:
I also cannot observe that VS does because it is all close sourced and I have zero access to anything they do.

ehntoo, 2022-11-15:
I believe the display of standard library types is handled by GDB. If your GDB supports python, you can add GDB's STL pretty-printers with `register_libstdcxx_printers`: https://sourceware.org/gdb/wiki/STLSupport https://gist.github.com/daverigby/99dae6997bced43a8fcdbfcb15e36116. The pretty-printers should already be included with most distributions of GDB.

haneefdm, 2023-03-13:
I don't see a good solution for this and it is not on our roadmap.

Closing this for now.