Marus/cortex-debug #737: Add GDB code before loading with OpenOCD

RobertoRoos, 2022-09-05:
I was using VS Code in WSL + OpenOCD (running on Windows native) + Cortex Debug (remote connection) successfully and happily.
But now the microcontroller we are developing has a locked flash and bootloader as feature, so when launching a debug session the file download will always fail.

I already have a GDB script that unlocks the flash. Is there some way I can let Cortex Debug run this script before the file download starts?

I looked at https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md , but the option seems to relate to command to e.g. OpenOCD itself.

haneefdm, 2022-09-05:
You can set `loadFiles` to an empty array ... meaning skip programming. You can set the `overrideLaunchCommands` to and just do a reset. or put your entire script line at a time in the overrideLaunchCommands commands. In the end, the device should be halted at the reset vector or nothing works.

The following is might also be helpful to see how everything comes together.

https://github.com/Marus/cortex-debug/wiki/Cortex-Debug:-Under-the-hood

haneefdm, 2022-09-05:
Please close this issue if you got your answers. We like to keep our issue list short.

RobertoRoos, 2022-09-06:
I ended up using the `preLaunchCommands` config keyword. The full config for debugging on Windows with WSL2 for an stm32 target build with CMake and Ninja:

```
		{
			"name": "openocd Debug",
			"type": "cortex-debug",
			"request": "launch",
			"servertype": "external",
			"cwd": "${workspaceRoot}",
			"executable": "${command:cmake.launchTargetPath}", // `launchTargetPath` will already trigger a build
			"preLaunchCommands": [
				"unlock_bootloader"
			],
			"gdbTarget": "${env:NAME}.local:3333",
			"gdbPath": "gdb-multiarch"
		},
```

And I put the `unlock_bootloader` command in my `~/.gdbinit` file:

```
# ~/.gdbinit

define unlock_bootloader
#...
end
```

closed 2022-09-06 by RobertoRoos