Marus/cortex-debug #739: Error: spawn arm-non-eabi-gdb ENOENT

Marthaarman, 2022-09-07:
I've been trying to setup debug for the raspberry pi pico on windows 10 with vscode.
All of it according to: https://learnembeddedsystems.co.uk/133-2
with the corresponding video: https://www.youtube.com/watch?v=0i2gLeBal9Y

I run into the issue with the description down here:
```
Reading symbols from arm-non-eabi-objdump --syms -C -h -w H:/DevEnv/Pico/debug-blink-example/build/debug-blink-example.elf
Reading symbols from arm-non-eabi-nm --defined-only -S -l -C -p H:/DevEnv/Pico/debug-blink-example/build/debug-blink-example.elf
Launching GDB: arm-non-eabi-gdb -q --interpreter=mi2
1-gdb-version
Error: arm-non-eabi-nm failed! statics/global/functions may not be properly classified: Error: spawn arm-non-eabi-nm ENOENT
    Expecting `nm` next to `objdump`. If that is not the problem please report this.
Could not start GDB process, does the program exist in filesystem?
Error: spawn arm-non-eabi-gdb ENOENT
```

Now, I've seen: https://github.com/Marus/cortex-debug/issues/8
However, the proposed solutions do not help. 

settings.json:
```
{
    "explorer.autoReveal": false,
    "cmake.configureEnvironment": {
        "PICO_SDK_PATH": "H:\\DevEnv\\Pico\\pico-sdk"
    },
    "cmake.generator": "MinGW Makefiles",
    "cortex-debug.openocdPath": "H:/DevEnv/Pico/openocd_picoprobe/openocd.exe",
    "cortex-debug.gdbPath":"arm-non-eabi-gdb",
    //"cortex-debug.armToolchainPath":"C:/VSARM/armcc/11.2 2022.02/bin",
    "cmake.configureOnEdit": false,
}
```

launch.json:
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceRoot}",
            //  set the following variable (most likely in the build folder located)
            "executable": "${workspaceRoot}/build/debug-blink-example.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "device": "Pico2040",
            "runToMain": true,
            "configFiles": ["interface/picoprobe.cfg", "target/rp2040.cfg"],
            //  set the following variable to the openocd tcl folder
            "searchDir": ["H:/DevEnv/Pico/openocd_picoprobe/tcl"],
            //  set the following variable to match the rp2040 (pico) sdk folder's subfile rp2040.svd (src/rp2040/hardware_regs/)
            "svdFile": "H:/DevEnv/Pico/pico-sdk/src/rp2040/hardware_regs/rp2040.svd",
            "showDevDebugOutput":"raw"
        }
    ]
}
```

When I uncomment the line in settings.json with the cortex-debug.armToolchainPath, I get the following error:
```
Reading symbols from C:/VSARM/armcc/11.2 2022.02/bin/arm-none-eabi-objdump.exe --syms -C -h -w H:/DevEnv/Pico/debug-blink-example/build/debug-blink-example.elf
Reading symbols from c:/vsarm/armcc/11.2 2022.02/bin/arm-none-eabi-nm.exe --defined-only -S -l -C -p H:/DevEnv/Pico/debug-blink-example/build/debug-blink-example.elf
Launching GDB: arm-non-eabi-gdb -q --interpreter=mi2
1-gdb-version
Could not start GDB process, does the program exist in filesystem?
Error: spawn arm-non-eabi-gdb ENOENT
```

As can be seen, I am using 11.2-2022.02-mingw-w64-i686-arm-none-eabi

Any suggestions on how to tackle this issue? 
I checked all the paths and they seem to be pointing to an existing folder. 

C:/VSARM/armcc/11.2 2022.02/bin contains all the arm-non-eabi-xxx.exe files

haneefdm, 2022-09-07:
You have a typo? `arm-non-eabi-gdb` should be `arm-none-eabi-gdb` -- `e` missing

Marthaarman, 2022-09-07:
> You have a typo? `arm-non-eabi-gdb` should be `arm-none-eabi-gdb` -- `e` missing

Wauw, so stupid how i never saw that.

closed 2022-09-07 by Marthaarman