Marus/cortex-debug #740: Missing RTT Timestamp in logfile

J-Rios, 2022-09-08:
Hello,

Not sure if this is a missing feature or just something that is not working properly on latest versions, but seems that when you setup the RTT configuration in the launch.json file to enable the Timestamp and also log to file features, the timestamp is shown only in the terminal, but not added to the log file.

Here is the launch.json configuration:

```json
{
    "version": "0.2.0",
    "configurations":
    [
        {
            "name": "Debug JLink",
            "type": "cortex-debug",
            "request": "launch",
            "runToEntryPoint": "main",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/project_name.elf",
            "svdFile": "${workspaceRoot}/project/svd/STM32L4x1.svd",
            "armToolchainPath": "/usr/bin/",
            "servertype": "jlink",
            "interface": "swd",
            "device": "STM32L431CC",
            "serverpath": "/opt/SEGGER/JLink/JLinkGDBServerCLExe",
            "rttConfig":
            {
                "enabled": true,
                "address": "auto",
                "decoders":
                [
                    {
                        "label": "RTT - STDOUT",
                        "port": 0,
                        "type": "console",
                        "timestamp": true,
                        "logfile": "${workspaceRoot}/build/rtt.log"
                    }
                ]
            }
        }
    ]
}
```

What I get from the VSCode Terminal is:

```text
[2022-09-08T17:18:54.963Z, +251016ms] Hello World
[2022-09-08T17:18:54.984Z, +251037ms] Good bye
```

What is stored in the rtt.log file:

```text
Hello World
Good bye
```

In case it is a missing feature, could it be considered/discussed as a feature request?
It will be a very nice to have feature.

Best Regards.

haneefdm, 2022-09-08:
Thanks for reporting this. It was an oversight. We added the timestamp after initial release and forgot all about the timestamp. As the implementation stands right now, this may take some doing. But I will add it to my todo list as I see that as a flaw.

Note that there will be no timestamps for binary type data as raw undecided bytes are written to the file. It will be only for `"type": "console"`

J-Rios, 2023-07-23:
Hi,

Here you can find a [Pull Request proposal](https://github.com/Marus/cortex-debug/pull/913) for getting this functionality.

Regards!

haneefdm, 2023-08-27:
Fix will be available in the next release. Changes already upstreamed

J-Rios, 2023-08-28:
Cool, thanks!

closed 2023-08-29 by haneefdm