Marus/cortex-debug #741: Regex for detecting GDB version does not work with GDB provided by fedora

DusterTheFirst, 2022-09-10:
The regex:

https://github.com/Marus/cortex-debug/blob/3a2079afacce600875b498cb4520ac1f0ed9a954/src/backend/mi2/mi2.ts#L143

fails to detect the version from the following GDB header:

```
GNU gdb (GDB) Fedora 12.1-1.fc36
```

which is the header from the provided (multiarch) gdb on fedora.

I propose the regex be updated to the following as it would allow for optional words proceeding the version string

```regex
^GNU gdb\s\(.*\)\s?(?:\w+\s)?(\d+)\.(\d+)[^\r\n]*
```

haneefdm, 2022-09-10:
Oooh. Thanks. This is the machine interface we are using. There is no spec on what it should return.

What do you think about this? It will allow petty much anything in front of the actual version.
```
^GNU gdb\s\(.*\)\s?(?:.*\s)?(\d+)\.(\d+)[^\r\n]*
```
or the following since I really don't care what is in the middle. Just looking for a version at the end of the line?
```
^GNU gdb\s.+\s(\d+)\.(\d+)[^\r\n]*
```

And thanks for introducing me to Debuggex

DusterTheFirst, 2022-09-10:
The simpler regex might be the better way to go since yah, as you said the program does not care what is between the `GNU gdb` and the version number.

haneefdm, 2022-09-20:
Should be fixed in V 1.6.4

closed 2022-09-20 by haneefdm