Marus/cortex-debug #745: Failed to launch OpenOCD GDB Server: Timeout.

ekawahyu, 2022-09-19:
The platform is Ubuntu 20.04. I can launch OpenOCD from terminal just fine, and able to debug from terminal with gdb-multiarch as well. But debugging with Cortex-Debug gives me the error as mentioned in the title.

Below you may find the raw debug output:
```
Cortex-Debug: VSCode debugger extension version 1.6.0 git(3026299). Usaage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from /usr/bin/objdump-multiarch --syms -C -h -w /home/zorin/Downloads/VSCode/libopencm3-examples/examples/stm32/f4/stm32f4-discovery/miniblink/miniblink.elf
Reading symbols from /usr/bin/nm-multiarch --defined-only -S -l -C -p /home/zorin/Downloads/VSCode/libopencm3-examples/examples/stm32/f4/stm32f4-discovery/miniblink/miniblink.elf
Launching GDB: /usr/bin/gdb-multiarch -q --interpreter=mi2
1-gdb-version
Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/zorin/Downloads/VSCode/libopencm3-examples -f /home/zorin/.vscode/extensions/marus25.cortex-debug-1.6.0/support/openocd-helpers.tcl -f /usr/share/openocd/scripts/interface/stlink-v2.cfg -f /usr/share/openocd/scripts/target/stm32f4x.cfg
    Please check TERMINAL tab (gdb-server) for output from openocd
Finished reading symbols from objdump: Time: 122 ms
Finished reading symbols from nm: Time: 141 ms
-> =thread-group-added,id="i1"
-> ~"GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2\n"
-> ~"Copyright (C) 2020 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"x86_64-linux-gnu\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<http://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source /home/zorin/.vscode/extensions/marus25.cortex-debug-1.6.0/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source /home/zorin/.vscode/extensions/marus25.cortex-debug-1.6.0/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-file-exec-and-symbols "/home/zorin/Downloads/VSCode/libopencm3-examples/examples/stm32/f4/stm32f4-discovery/miniblink/miniblink.elf"
-> 8^done
Failed to launch OpenOCD GDB Server: Timeout.
```

From the gdb-server terminal, I can see that OpenOCD is ready and waiting for connection:
```
[2022-09-19T23:21:44.338Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/zorin/Downloads/VSCode/libopencm3-examples -f /home/zorin/.vscode/extensions/marus25.cortex-debug-1.6.0/support/openocd-helpers.tcl -f /usr/share/openocd/scripts/interface/stlink-v2.cfg -f /usr/share/openocd/scripts/target/stm32f4x.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v40 API v2 SWIM v0 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 2.917254
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
[2022-09-19T23:21:54.390Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

What am I missing from my setup?

haneefdm, 2022-09-20:
```
Open On-Chip Debugger 0.10.0
```
OpenOCD 0.10.0. That is ancient. Many many things will fail including support for ST devices. OpenOCD is supposed to print something that says it is ready to accept gdb connections which I don't see at all.

Please install the latest OpenOCD and report back. Also use our latest extension as well.

ekawahyu, 2022-09-20:
@haneefdm, yep, I didn't realize how ancient 0.10.0 since it comes with Ubuntu 20.04 package manager. Everything works as expected after I build the latest OpenOCD from source (0.12.0-rc1). Thanks. I am closing this issue now.

closed 2022-09-20 by ekawahyu