Marus/cortex-debug #747: Can't start debug using stlink

XinyuKhan, 2022-09-28:
Config:

        {
            "name": "ST-Link Debug",
            "cwd": "`${workspaceFolder}",
            "executable": "${workspaceFolder}/Debug/ecu_v4_bootloader.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "stlink",
            "serverpath": "/Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.macos64_2.0.300.202203231527/tools/bin/ST-LINK_gdbserver",
            "device": "STM32F405RG",
            "interface": "swd",
            "runToEntryPoint": "main",
            "svdFile": "/Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.productdb.debug_2.0.300.202204011209/resources/cmsis/STMicroelectronics_CMSIS_SVD/STM32F405.svd",
            "showDevDebugOutput": "parsed"
        },

Error Log:

Cortex-Debug: VSCode debugger extension version 1.6.5 git(ed64786). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /Users/xinyuhan/Projects/lowerbody_exosuit/stm32_bootloader/Project/stm32cubeide/ecu_v4_bootloader/Debug/ecu_v4_bootloader.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /Users/xinyuhan/Projects/lowerbody_exosuit/stm32_bootloader/Project/stm32cubeide/ecu_v4_bootloader/Debug/ecu_v4_bootloader.elf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
Error: arm-none-eabi-nm failed! statics/global/functions may not be properly classified: Error: spawn arm-none-eabi-nm ENOENT
    Expecting `nm` next to `objdump`. If that is not the problem please report this.
Could not start GDB process, does the program exist in filesystem?
Error: spawn arm-none-eabi-gdb ENOENT

haneefdm, 2022-09-28:
You don't have the arm toolchain installed or have a setting to point to it. You can create a setting in your user settings (Use VSCode Settings...) or in your launch.json with `armToolchainPath`

Also, make sure you can run the arm gdb from the command line. Many times it is partially installed.

Please close this issue if satisfied.

haneefdm, 2022-09-28:
[image]

XinyuKhan, 2022-09-28:
> [image]

Hi haneefdm
Thank for you reply. I have tried the methond you mentioned above, but it still not worked. I set the path of the arm toolchains in the PATH env in my .zshrc and did not config the armToolchainPath in the vscode setting at beginning. It worked well, but this error occurred recently. After that I tried to add the toolchains path in the armToolchainPath config but the error was same.

haneefdm, 2022-09-28:
Yes, setting in PATH works well. But macOS is tricky in which file you use to set that PATH. As you know, we did not change anything and I use macOS all the time. I don't have issues either way. My .zshrc contains this
```
export PATH="$PATH:/Applications/ARM/bin:/Applications/CMake.app/Contents/bin"
```

I need to see exactly what you changed in settings and what you see in the Debug Console./

haneefdm, 2022-09-28:
Make sure you can launch alll thoss tools from the command-line. If you can't, neither can we.

haneefdm, 2022-10-03:
Closing. No response

closed 2022-10-03 by haneefdm