Marus/cortex-debug #749: Long start-up time due to target remote connection

a-cinar, 2022-09-28:
Hi,

We have been using `gdb` with `j-link debugger` for a while through the `CLI`. Now I am trying to integrate it to the `cortex-debug` extension in `vs-code`. I already had my own routine to start debugging and everything works fine and fast via the `CLI`. I completed the configurations of the `cortex-debug` extension like this:

```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Debug Main",
            "interface": "swd",
            "servertype": "jlink",
            "request": "launch",
            "device": "EFM32GG11B820F2048",
            "type": "cortex-debug",
            "cwd": "${workspaceRoot}",
            "armToolchainPath": "${workspaceRoot}/tools/gcc-arm-none-eabi-10.3-2021.10/bin/",
            "serverpath": "/usr/bin/JLinkGDBServer",
            "serverArgs": [
                "-speed",
                "8000",
            ],
            "showDevDebugOutput": "raw",
            "showDevDebugTimestamps": true,
            "runToEntryPoint": "main",
            "executable": "${workspaceRoot}/main-efm32gg11b820f2048il120.out",
        }
    ]
}
```

So there is nothing specific here. But I observe that `target-select extended-remote localhost:50000` command takes too much time compared to running it from the `CLI`. It takes around 16 seconds on `cortex-debug` and it takes less than a second when I run it on CLI. I attached the logs below, I am imitating the exact same command chain on the same machine `(Ubuntu)` by opening a `terminal` for the gdb server and another one for the gdb itself.

```md
0000000011: Cortex-Debug: VSCode debugger extension version 1.6.5 git(ed64786). Usage info: https://github.com/Marus/cortex-debug#usage
0000000013: Reading symbols from /tools/gcc-arm-none-eabi-10.3-2021.10/bin/arm-none-eabi-objdump --syms -C -h -w /dev/apps/firmware/.build/main-efm32gg11b820f2048il120.out
0000000016: Reading symbols from /tools/gcc-arm-none-eabi-10.3-2021.10/bin/arm-none-eabi-nm --defined-only -S -l -C -p /dev/apps/firmware/.build/main-efm32gg11b820f2048il120.out
0000000024: Launching GDB: /tools/gcc-arm-none-eabi-10.3-2021.10/bin/arm-none-eabi-gdb -q --interpreter=mi2
0000000030: 1-gdb-version
0000000040: Launching gdb-server: /usr/bin/JLinkGDBServer -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device EFM32GG11B820F2048 -speed 8000
0000000040:     Please check TERMINAL tab (gdb-server) for output from /usr/bin/JLinkGDBServer
0000000109: Finished reading symbols from objdump: Time: 96 ms
0000000275: -> =thread-group-added,id="i1"
0000000275: -> ~"GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git\n"
0000000275: -> ~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
0000000276: -> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
0000000276: -> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
0000000276: -> ~"This GDB was configured as \"--host=x86_64-linux-gnu --target=arm-none-eabi\".\n"
0000000276: -> ~"Type \"show configuration\" for configuration details.\n"
0000000276: -> ~"For bug reporting instructions, please see:\n"
0000000276: -> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
0000000276: -> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
0000000276: -> ~"\n\n"
0000000276: -> ~"For help, type \"help\".\n"
0000000276: -> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
0000000276: -> 1^done
0000000278: 2-gdb-set mi-async on
0000000181: -> 2^done
0000000181: 3-interpreter-exec console "set print demangle on"
0000000183: -> 3^done
0000000183: 4-interpreter-exec console "set print asm-demangle on"
0000000184: -> =cmd-param-changed,param="print asm-demangle",value="on"
0000000184: -> 4^done
0000000184: 5-enable-pretty-printing
0000000184: -> 5^done
0000000184: 6-interpreter-exec console "source /home/acinar/.vscode/extensions/marus25.cortex-debug-1.6.5/support/gdbsupport.init"
0000000188: -> 6^done
0000000188: 7-interpreter-exec console "source /home/acinar/.vscode/extensions/marus25.cortex-debug-1.6.5/support/gdb-swo.init"
0000000190: -> =cmd-param-changed,param="language",value="c"
0000000191: -> =cmd-param-changed,param="language",value="auto"
0000000193: -> 7^done
0000000193: 8-file-exec-and-symbols "/dev/apps/firmware/.build/main-efm32gg11b820f2048il120.out"
0000000282: -> 8^done
0000000406: Finished reading symbols from nm: Time: 390 ms
0000000523: Debug Time: GDB Ready...
0000001025: Debug Time: GDB Server post start events done...
0000001025: Debug Time: objdump and nm done...
0000001025: Debug Time: All pending items done, proceed to gdb connect...
HERE 
0000001026: 9-target-select extended-remote localhost:50000
0000001040: -> =thread-group-started,id="i1",pid="42000"
0000001040: -> =thread-created,id="1",group-id="i1"
0000027860: -> ~"0x00005f24 in EMU_EnterEM1 () at /third_party/gecko_sdk/platform/emlib/inc/em_emu.h:1311\n"
0000027860: 0x00005f24 in EMU_EnterEM1 () at /third_party/gecko_sdk/platform/emlib/inc/em_emu.h:1311
0000027861: -> ~"1311\t}\n"
0000027861: 1311	}
0000027861: -> *stopped,frame={addr="0x00005f24",func="EMU_EnterEM1",args=[],file="/third_party/gecko_sdk/platform/emlib/inc/em_emu.h",fullname="/third_party/gecko_sdk/platform/emlib/inc/em_emu.h",line="1311",arch="armv7e-m"},thread-id="1",stopped-threads="all"
0000027861: mi2.status = stopped
0000027862: Program stopped, probably due to a reset and/or halt issued by debugger
0000027862: -> 9^connected
TO HERE
0000027862: 10-interpreter-exec console "load  /dev/apps/firmware/.build/main-efm32gg11b820f2048il120-signed-full.s37"
0000027869: -> ~"Loading section .sec1, size 0x4b4cc lma 0x0\n"
0000027869: Loading section .sec1, size 0x4b4cc lma 0x0
0000027870: -> 10+download,{section=".sec1",section-size="308428",total-size="329960"}
0000027870: -> 10+download,{section=".sec1",section-sent="16032",section-size="308428",total-sent="16032",total-size="329960"}
0000027874: -> ~"Loading section .sec2, size 0x541c lma 0xfe10000\n"
0000027874: Loading section .sec2, size 0x541c lma 0xfe10000
0000027875: -> 10+download,{section=".sec2",section-size="21532",total-size="329960"}
0000027875: -> ~"Start address 0x00000000, load size 329960\n"
0000027875: Start address 0x00000000, load size 329960
0000032994: -> ~"Transfer rate: 53704 KB/sec, 14998 bytes/write.\n"
0000032994: Transfer rate: 53704 KB/sec, 14998 bytes/write.
0000032994: -> 10^done
0000032995: 11-interpreter-exec console "monitor reset"
0000033070: -> @"Resetting target\r\n"
0000033070: Resetting target
0000033070: -> 11^done
```

You can see that it is initiating the target connection on step 9 (timestamp: 0000001026) and completes it after ~26 seconds. I don't know what could be the difference between running it from the extension and running it from the external CLI as there is no parsing time included in that long delay, the time consumed is the time the GDB command responded. 

This might not be related to the extension itself and I am new to GDB so I might have missing something here. Any help appreciated.

haneefdm, 2022-09-28:
You can see that gdb is just waiting and doing nothing. Surprised it did not timeout.

```
            "serverpath": "/usr/bin/JLinkGDBServer",
```
Might be the problem. Isn't that the GUI version? It should have the letters `CL` in it.

https://github.com/Marus/cortex-debug/wiki/J-Link-Specific-Configuration

a-cinar, 2022-09-28:
> You can see that gdb is just waiting and doing nothing. Surprised it did not timeout.
>
> ```
>             "serverpath": "/usr/bin/JLinkGDBServer",
> ```
>
> Might be the problem. Isn't that the GUI version? [...]

For Linux, it is just `JLinkGDBServer` so it should be correct. It is sending the `-nogui` argument while running it. I checked the `J-Link Specific Configuration`s all around the web and the configurations were identical to mine. But I don't have any reference point to compare the startup times of cortex-debug on those projects.

haneefdm, 2022-09-28:
> For Linux, it is just JLinkGDBServerCLExe

Upto you but as far as I know unless SEGGER renamed it recently this statement is not true. The -nogui option is only valid on JLinkGDBServerCLexe and not some other executable.

Since you refuse to try it, there is not much I can do here.

a-cinar, 2022-09-28:
> > For Linux, it is just JLinkGDBServerCLExe
> 
> Upto you but as far as I know unless SEGGER renamed it recently this statement is not true. The -nogui option is only valid on JLinkGDBServerCLexe and not some other executable.
> 
> Since you refuse to try it, there is not much I can do here.

My bad there, you are right about that. So I changed my config to use `JLinkGDBServerCLexe` instead. It doesn't affect the result though.

haneefdm, 2022-09-28:
It may be `JLinkGDBServer` is symlink to `JLinkGDBServerCLExe` but I couldn't be certain.

Check to see what is going on in the `TERMINAL(gdb-server)` tab. This is where output from JLink server is printed.

a-cinar, 2022-09-29:
I checked the `gdb-server` logs. I get this message immediately after the `target remote` request:

```md
Waiting for GDB connection...Connected to 127.0.0.1
```

Then it issues some register reads, which takes all of the time:

```md
Reading all registers
Read 4 bytes @ address 0x0002038A (Data = 0x3010F89D)
Reading 64 bytes @ address 0x200098C0
Read 4 bytes @ address 0x000056AC (Data = 0xFFC1F001)
Reading 64 bytes @ address 0x20009900
```

After these, I get the response on the `Debug Console` of the extension:

```md
0000025589: -> ~"0x00005f24 in EMU_EnterEM1 () at /home/acinar/envio/repositories/embio/third_party/gecko_sdk/platform/emlib/inc/em_emu.h:1311\n"
0000025589: 0x00005f24 in EMU_EnterEM1 () at /home/acinar/envio/repositories/embio/third_party/gecko_sdk/platform/emlib/inc/em_emu.h:1311
0000025590: -> ~"1311\t}\n"
0000025590: 1311	}
0000025591: -> *stopped,frame={addr="0x00005f24",func="EMU_EnterEM1",args=[],file="/home/acinar/envio/repositories/embio/third_party/gecko_sdk/platform/emlib/inc/em_emu.h",fullname="/home/acinar/envio/repositories/embio/third_party/gecko_sdk/platform/emlib/inc/em_emu.h",line="1311",arch="armv7e-m"},thread-id="1",stopped-threads="all"
0000025591: mi2.status = stopped
0000025591: Program stopped, probably due to a reset and/or halt issued by debugger
0000025592: -> 9^connected
```

So it seems like the time consumed is not exactly because of the connection. It shouldn't be related to the reading of the registers because I don't see any reason why it should make a difference there. 

I think the issue is, it is trying to halt the program as you can see from the `Debug Console` logs. We don't issue a `halt` or `interrupt` command, so I checked the `J-Link GDB Server` docs. It says it will halt the target after the start of GDB server:

[image]

So I added the `-nohalt` argument to disable that but it halted anyway after the `target remote` request. I realized that the same thing also happens on CLI but it halts much faster. Is it because of the indexing on vs-code? It actually doesn't open that file as it was an automatic halt. But maybe it is trying to find the file anyway?

a-cinar, 2022-09-29:
Addition: I also observe this during debugging too. I can see the `breakpoint hit` message on `Debug Console` but sometimes it takes up to 10 seconds for the vs-code to go that file/line. Is that the time required to parse the message then find the file? The project is big so I think it may be the reason it takes too much time to do that. If that's the case, I would like to find a way to fasten it. Maybe play around some indexing etc.

haneefdm, 2022-09-29:
```
0000000406: Finished reading symbols from nm: Time: 390 ms
0000000523: Debug Time: GDB Ready...
0000001025: Debug Time: GDB Server post start events done...
0000001025: Debug Time: objdump and nm done...
0000001025: Debug Time: All pending items done, proceed to gdb connect...
```
Everyone has seen and groked the executable at this point. The gdb-server never sees the executable at all.

> but sometimes it takes up to 10 seconds for the vs-code to go that file/line

Can you expand on this. We do not report to VSCode the very first halt in some cases. And we do not parse the file name which is trivial anyways and unless your file is 100MB of C code, VSCode will not take that long either. Everyone is waiting on actually connection to form between gdb and the gdb-server and no one else is involved.

Until a connection is formed, the server has no idea what is about to happen. So, it is the connection time to focus on.

Can you see the time difference between `ping localhost` vs `ping 127.0.0.1`?

Could it be that JLink is taking a long time to discover the device? Could you post more of the gdb-server output?

a-cinar, 2022-09-29:
> Can you see the time difference between ping localhost vs ping 127.0.0.1?

They are similar.

> Could it be that JLink is taking a long time to discover the device? Could you post more of the gdb-server output?

It might be the case, I see that after the `breakpoint-hit`, it is trying to do some reads for registers and variables. You can see the logs below, it took ~16 seconds to reflect to the GUI after breakpoint hit:

```md
0000148764: -> *stopped,reason="breakpoint-hit",disp="keep",bkptno="1",frame={addr="0x0001beb2",func="PQ_MON_SampleChannel",args=[{name="aChannel",value="0x8004ad24 <sPqMonInstance>"}],file="/home/acinar/envio/repositories/embio/dev/libs/modules/src/modules/monitoring/pq_mon.c",fullname="/home/acinar/envio/repositories/embio/dev/libs/modules/src/modules/monitoring/pq_mon.c",line="231",arch="armv7e-m"},thread-id="1",stopped-threads="all"
0000148764: mi2.status = stopped
0000148767: 40-thread-list-ids
0000148768: -> 40^done,thread-ids={thread-id="1"},current-thread-id="1",number-of-threads="1"
0000148768: 41-thread-info 1
0000148768: -> 41^done,threads=[{id="1",target-id="Thread 57005",frame={level="0",addr="0x0001beb2",func="PQ_MON_SampleChannel",args=[{name="aChannel",value="0x8004ad24 <sPqMonInstance>"}],file="/home/acinar/envio/repositories/embio/dev/libs/modules/src/modules/monitoring/pq_mon.c",fullname="/home/acinar/envio/repositories/embio/dev/libs/modules/src/modules/monitoring/pq_mon.c",line="231",arch="armv7e-m"},state="stopped"}]
0000148769: 42-data-read-memory-bytes "0x40088000" 572
0000148778: -> 42^done,memory=[{begin="0x40088000",offset="0x00000000",end="0x4008823c",contents="500050004444444434440044800f0000000000000000000000000000ab0f0000ffff00000000000000000000000000005000500044441404004004404018000000000000000000000000000062180000ffff00000000000000000000000000007000700014144114414114148cfc00000000000000000000000000009cfb0000ffff000000000000000000000000000070007000142444aa40411114fb9e0000000000000000000000000000f81a0000ffff0000000000000000000000000000500050004000101444444444c2000000000000000000000000000000c2930000ffff0000000000000000000000000000500050000044444444114404347f0000000000000000000000000000f4770000ffff00000000000000000000000000005000500000000000000000000000000000000000000000000000000000000000ffff00000000000000000000000000005000500000000000000000000000000000000000000000000000000000000000ffff00000000000000000000000000005000500000000000000000000000000000000000000000000000000000000000ffff00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"}]
0000148778: 43-data-read-memory-bytes "0x40088400" 88
0000148782: -> 43^done,memory=[{begin="0x40088400",offset="0x00000000",end="0x40088458",contents="0030002000550300103210321032103200000000881e000000000000001863010000000000000000881e000000000000000000000000000000000000000000001f0000000000000000000000000000000300000000000000"}]
0000148782: 44-data-list-register-values N
0000148787: -> 44^done,register-values=[{number="0",value="536962012"},{number="1",value="3"},{number="2",value="1"},{number="3",value="0"},{number="4",value="0"},{number="5",value="2147790116"},{number="6",value="2147790116"},{number="7",value="117901063"},{number="8",value="2147835172"},{number="9",value="151587081"},{number="10",value="0"},{number="11",value="536923692"},{number="12",value="0"},{number="13",value="0x20016340 <TASK_IO_Stk+3940>"},{number="14",value="150823"},{number="15",value="0x1beb2 <PQ_MON_Execute+58>"},{number="25",value="553648128"},{number="91",value="536910088"},{number="92",value="536961856"},{number="93",value="0"},{number="94",value="0"},{number="95",value="0"},{number="96",value="6"},{number="97",value="536870928"},{number="98",value="0"},{number="99",value="0"},{number="100",value="0"},{number="101",value="0"},{number="102",value="0"},{number="103",value="0"},{number="104",value="0"},{number="105",value="0"},{number="106",value="0"},{number="107",value="0"},{number="108",value="0"},{number="109",value="0"},{number="110",value="2.5"},{number="111",value="1.47680569"},{number="112",value="12388344"},{number="113",value="8388608"},{number="114",value="-4.93868026e-40"},{number="115",value="0"},{number="116",value="0"},{number="117",value="0"},{number="118",value="0"},{number="119",value="0"},{number="120",value="0"},{number="121",value="0"},{number="122",value="0"},{number="123",value="0"},{number="124",value="0"},{number="125",value="0"},{number="126",value="0"},{number="127",value="0"},{number="128",value="0"},{number="129",value="0"},{number="130",value="0"},{number="131",value="0"},{number="132",value="0"},{number="133",value="0"},{number="134",value="0"},{number="135",value="0"},{number="136",value="0.11340285840560682"},{number="137",value="1.9156199630015454e+53"},{number="138",value="1.0611720220025939e-314"},{number="139",value="0"},{number="140",value="0"},{number="141",value="0"},{number="142",value="0"},{number="143",value="0"},{number="144",value="0"},{number="145",value="0"}]
0000148791: 45-stack-info-depth --thread 1 1000
0000164255: -> 45^done,depth="9"
0000164256: 46-stack-info-depth --thread 1 1000
0000164256: -> 46^done,depth="9"
0000164256: 47-stack-list-frames --thread 1 0 8
[... cut ...]
0000164887: 54-var-create --thread 1 --frame 0 var_aChannel_256 * "aChannel"
0000164887: -> 54^done,name="var_aChannel_256",numchild="8",value="0x8004ad24 <sPqMonInstance>",type="pq_mon_channel_t *",thread-id="1",has_more="0"
```

So I realized that some of the variables from the `Variables` tab were expanded. So everytime it hits a breakpoint, it is reading all of the expanded variables and registers. I also realized that it restores the expanded variables from the previous debug sessions so I didn't realize that they were expanded.

a-cinar, 2022-09-29:
Addition: I realized that even if I collapse the `Local` tab in the `VARIABLES`, it expands automatically when the breakpoint hits. So if there are big local variables inside the scope of the breakpoint, it increases the time to reflect the GUI as it reads those variables first.

haneefdm, 2022-09-29:
```
0000148791: 45-stack-info-depth --thread 1 1000
0000164255: -> 45^done,depth="9"
```
There is the culprit. gdb tries to figure out our current stack and it is taking a super long time. It relies on JLink to provide the needed register and memory reads to reconstruct your call stack. How slow is your connection anyways? Or maybe your JLink RTOS plugin is really bad. Badly written RTOS plugins can cause these kinds of problems. Try turning this off temporarily.

Where are these memory reads coming from? Our RTOSview? You may want to turn that off until we figure this out. Memory reads are not cheap.

Again, please post the output of the gdb-server. Humor me.

You can also have JLink dump gdb transaction log to a file

https://wiki.segger.com/J-Link_GDB_Server#-log

haneefdm, 2022-09-29:
> Addition: I realized that even if I collapse the Local tab in the VARIABLES, it expands automatically when the breakpoint hits. So if there are big local variables inside the scope of the breakpoint, it increases the time to reflect the GUI as it reads those variables first.

The number of variables matter -- you don't have thousands do you? Size of variables is less so. Yes, that is what VSCode does and I like it like that.

a-cinar, 2022-09-29:
> There is the culprit. gdb tries to figure out our current stack and it is taking a super long time. It relies on JLink to provide the needed register and memory reads to reconstruct your call stack. [...]

> Where are these memory reads coming from? Our RTOSview? You may want to turn that off until we figure this out. Memory reads are not cheap.

I am not using an RTOS plugin nor aware of it, I'd like to use that if I can so I will also check that :) But now I am curious why this read happens:

```md
0000148791: 45-stack-info-depth --thread 1 1000
0000164255: -> 45^done,depth="9"
```

How can I disable this? 

Here you can find the gdb server logs.

[JLinkGDBServerLog.txt](https://github.com/Marus/cortex-debug/files/9676962/JLinkGDBServerLog.txt)

haneefdm, 2022-09-29:
> How can I disable this?

Not use a debugger you mean? No stack info means zero debug info. No debugger can work without this.

> Again, please post the output of the gdb-server. Humor me.

What I mean is the text in the gdb-server window in the TERMINAL tab. You posted a couple lines above.
Contents of this window

[image]

haneefdm, 2022-09-29:
```
0000001026: 9-target-select extended-remote localhost:50000
0000001040: -> =thread-group-started,id="i1",pid="42000"
0000001040: -> =thread-created,id="1",group-id="i1"
0000027860: -> ~"0x00005f24 in EMU_EnterEM1 () at /third_party/gecko_sdk/platform/emlib/inc/em_emu.h:1311\n"
0000027860: 0x00005f24 in EMU_EnterEM1 () at /third_party/gecko_sdk/platform/emlib/inc/em_emu.h:1311
0000027861: -> ~"1311\t}\n"
0000027861: 1311	}
0000027861: -> *stopped,frame={addr="0x00005f24",func="EMU_EnterEM1",args=[],file="/third_party/gecko_sdk/platform/emlib/inc/em_emu.h",fullname="/third_party/gecko_sdk/platform/emlib/inc/em_emu.h",line="1311",arch="armv7e-m"},thread-id="1",stopped-threads="all"
0000027861: mi2.status = stopped
0000027862: Program stopped, probably due to a reset and/or halt issued by debugger
0000027862: -> 9^connected
```
Something is seriously wrong with your JLink setup which you have not told me about. The connection actually happens fast. The first thing after connection is that gdb asks JLINK SW to get it a status. GDB 1000% has to rely on gdb-server doing its job well. JLink is taking its sweet time reporting status. It took nearly 27 seconds for JLink to report its first status.

status queries are done by gdb, each time the program stops and we ask for thread/stack/frame information. Again, I believe it is JLink to be blamed here.

You can try switching to a newer version of GDB to see if it will help.

PS: Do you have a memory view open which can cause memory reads but that is not the issue right now.

haneefdm, 2022-09-29:
I looked at the logifle you attached. Very hard to read. Seems like it has data from multiple runs and I have no idea which command trigerred which transactions. You have to do this yourself.

Watch the logfile as you navigate through each step in VSCode. Everything has a cause and effect we need to determine some causality.

I am at the end of my investigative efforts. I am working almost blind here.

Let me know when you find out more on why getting program status is taking so much time. It should be in sub-millsiconds.

a-cinar, 2022-09-30:
Thank you for all of your efforts. You are right, it is hard for you to understand it. I am able to watch both logs at the same time so it would be easier if I do it. That's why I need to have an understanding of the whole process. 

I still didn't understand what you meant by this:

> Where are these memory reads coming from? Our RTOSview? You may want to turn that off until we figure this out. Memory reads are not cheap.

I don't have any view on vs-code related to RTOS. Am I missing something here? I also don't have any memory views open except the `Locals` which are read automatically anyway.

I will investigate the issue with the j-link. I will try to update it and also find another j-link debugger and test it. Also, one last question. I will watch through the gdb server logs to understand what happens during that period of time when it takes 27 seconds for JLink to report its first status. I can also share the logs of the both debug console and gdb server here, just for reference for you or for other people.

haneefdm, 2022-09-30:
Oh, I forgot, memory reads can also come from SVD/Peripheral viewer

> Also, one last question. I will watch through the gdb server logs to understand what happens during that period of time when it takes 27 seconds for JLink to report its first status. I can also [...]

Not sure what the question is.

a-cinar, 2022-09-30:
Forgot to explain how I got these logs:

- Start debugging and wait for it to break at `main`
- Clear the logs
- Click `Step Into`.
- Wait for it to step into the line.
- On the `Debug Console` logs we can see that it took ~12 seconds for it to `Step Into` the line.

If I `Restart` the debug session and do the same steps, it takes less than a second. So the logs above taken from the first debug session.

a-cinar, 2022-09-30:
I tried with a differen J-Link debugger hardware but it didn't change anything, FYI/

haneefdm, 2022-10-02:
Look at your file [GDB-LOGS.txt](https://github.com/Marus/cortex-debug/files/9684190/GDB-LOGS.txt) -- It should have been called JLink log btw, becuase it is create by them in their format.

One step at a time...Single-Step initiated
```
02-00000000-00-00100044-0022: TAB7FD700 100:053.035 JLINK_Step()
...
02-00000000-00-00100044-0026: TAB7FD700 100:053.112 JLINK_IsHalted()
```
GDB makes a two queries
```
01-0000000E-00-00100044-0017: $T05thread:0000DEAD;#74
00-0000000E-00-00100044-0005: $g#67
...
02-00000000-00-00100049-002A: TAB7FD700 100:058.675 JLINK_ReadReg(FPS31)
02-00000000-00-00100049-0032: TAB7FD700 100:058.692 - 0.018ms returns 0x00000000
01-0000000E-00-00100049-008C: $a41301200ae8030000000000a4130120c412012000000000000000000000000000000000000000000010ff1f0000000028e90000d80f0020a5ff0100d802020000000041#ee
```
Is the current thread still alive? Give me all the registers. This is followed by a bunch or register reads and last register read is `FPS31`. Elapsed time < 6ms. The last line is a return response to gdb for a previous request??

What follows is what is the PROBLEM. Thousands of SWO reads that read nothing and FLUSHes that probably flush nothing.
```
02-00000000-00-00100049-004C: TB4BC0700 100:058.754 JLINK_SWO_Read(..., Offset = 0x00, NumBytes = 0x80000)
02-00000000-00-00100049-001E: TB4BC0700 100:058.845   Data: 
02-00000000-00-00100049-003B: TB4BC0700 100:058.916 - 0.164ms returns NumBytesRead = 0x00
02-00000000-00-00100049-0044: TB4BC0700 100:058.938 JLINK_SWO_Control(JLINKARM_SWO_CMD_FLUSH, ...)
02-00000000-00-00100049-002C: TB4BC0700 100:058.971 - 0.034ms returns 0x00
```
And, finally after being busy for **12 seconds**, it accepts a request from gdb to read some memory
```
00-0000000E-00-00112701-000C: $m202d8,4#fd
02-00000000-00-00112702-003F: TAB7FD700 112:711.161 JLINK_ReadMem(0x000202D8, 0x4 Bytes, ...)
```
From here, everything looks normal (except for a few more spurious SWO things)

Question: Who is making JLink SWO requests, why so many? Note that in this log file, these can be requests initiated by JLink itself or another JLink client that is not gdb (one of the other SEGGER utilities).

We spent 12 seconds doing nothing.

I saw the same problem with the earlier log file you attached and it was also flooded with the same SWO transactions. Note that gdb has no idea what SWO is. And, we don't make such requests and you ahven't even enabled SWO trace. Only thing we did was to reserve a TCP port for SWO but didn't use it and this is the same for the last five years. We always reserve.

To interpret this log, you need to understand the remote serial procol

https://sourceware.org/gdb/onlinedocs/gdb/Remote-Protocol.html

But, since it is output by JLink, it has all sorts of JLink stuff -> More than just request/reponse with gdb. Useful, but I have to enable a filter in my head to see what is what and you can easily differentiate between a request vs response.

GDB can also output such logs but I can't remember how to enable it. I use it sometimes to verify what the gdb-server tells me.

a-cinar, 2022-10-04:
Thanks for all the insights. 

I don't have any idea why the j-link does that read-flush dumps for 12 seconds. I saw the same thing, but I thought it could be something like pinging when idle. I didn't think that it was the reason, not the result.

Honestly, I don't know how to proceed right now. Maybe, I can try to get help from segger to delegate it. Otherwise, we can not keep digging into this issue. It is not related to the `Cortex Debug` extension. By the way, I also tried initiating the debug process from the `Simplicity Studio IDE` with some debug prints enabled. It was much faster but there was no difference in the gdb commands it is using. It is weird, I don't have any other clients that might be causing these dummy reads-flushes. 

I will update the issue here if I get any results. I appreciate your support!

haneefdm, 2022-10-04:
There is one last thing to try.

Can you replace the gdbserver path with a shellscript that removes the args `-swoport 50001` and launch the real gdb server with the rest of the args?

This would eliminate any trace of SWO. But I do think at this point contacting SEGGER is in order

a-cinar, 2022-10-04:
> Can you replace the gdbserver path with a shellscript that removes the args `-swoport 50001` and launch the real gdb server with the rest of the args?

Sure. I'll do that right away. By the way, here is another log of a session where I just try to `step over` the first line on `main.c`. I disabled the `swo` from the `launch.json`. It took 8 seconds to step over a line. Again, if I restart the debug session and retry everything is fine. The same step over takes less than a second.

```md
0000073519: 35-exec-next --thread 1
0000073519: -> 35^running
0000073520: -> *running,thread-id="all"
0000073520: mi2.status = running
0000082139: -> *stopped,reason="end-stepping-range",frame={addr="0x00024722",func="main",args=[],file="/firmware/src/main.c",fullname="/firmware/src/main.c",line="51",arch="armv7e-m"},thread-id="1",stopped-threads="all"
0000082139: mi2.status = stopped
```

And here is the [ j-link logs](https://github.com/Marus/cortex-debug/files/9707046/j-link-logs.txt) of the same session.

a-cinar, 2022-10-04:
> Can you replace the gdbserver path with a shellscript that removes the args -swoport 50001 and launch the real gdb server with the rest of the args?

This didn't make any difference, unfortunately.

a-cinar, 2022-10-04:
> But I do think at this point contacting SEGGER is in order.

By the way, the debug adapter is actually an EV Board from Silicon Labs but it has the segger programmer IC on it.

haneefdm, 2022-10-04:
```
02-00000000-00-00073896-002C: TA803E700 073:897.110 JLINK_GetHWStatus(...)
02-00000000-00-00073896-0029: TA803E700 073:897.322 - 0.215ms returns 0
02-00000000-00-00076098-002C: TA803E700 076:098.465 JLINK_GetHWStatus(...)
02-00000000-00-00076098-0029: TA803E700 076:098.707 - 0.243ms returns 0
02-00000000-00-00078299-002C: TA803E700 078:299.695 JLINK_GetHWStatus(...)
02-00000000-00-00078299-0029: TA803E700 078:299.943 - 0.267ms returns 0
02-00000000-00-00080500-002C: TA803E700 080:501.104 JLINK_GetHWStatus(...)
02-00000000-00-00080500-0029: TA803E700 080:501.350 - 0.266ms returns 0
```
Each `JLINK_GetHWStatus` is taking ~3 seconds. That should be measured in fractions of milliseconds.

Restarts helping does not make sense to me because everything is restarted from scratch and there is no memory anywhere of what happened previously.

haneefdm, 2022-10-04:
Never mind my previous message. It is 0.3 ms.

a-cinar, 2022-10-04:
Let me send you the logs of the same session after the restart maybe we can find out what's skipped there?

a-cinar, 2022-10-04:
Wait a second, you were right about that. It takes ~3 seconds actually to process `JLINK_GetHWStatus`.

a-cinar, 2022-10-04:
Here is the logs of the same session after the restart. It takes less than a second to step over the same line. But it doesn't issue the `JLINK_GetHWStatus` as you can see from the log file.

```md
0000074739: 81-exec-next --thread 1
0000074739: -> 81^running
0000074739: -> *running,thread-id="all"
0000074739: mi2.status = running
0000074810: -> *stopped,reason="end-stepping-range",frame={addr="0x00024722",func="main",args=[],file="/home/acinar/envio/repositories/embio/dev/apps/cube/cube-main/firmware/src/main.c",fullname="/home/acinar/envio/repositories/embio/dev/apps/cube/cube-main/firmware/src/main.c",line="51",arch="armv7e-m"},thread-id="1",stopped-threads="all"
0000074811: mi2.status = stopped
```
And the `j-link` logs:

[j-link-logs-restart.txt](https://github.com/Marus/cortex-debug/files/9707654/j-link-logs-restart.txt)

haneefdm, 2022-10-04:
It did make calls to JLINK_SWO_Read/JLINK_SWO_Read but just once. Not the thousands of times that we saw earlier.

I don't have an explanation for any of this.

Can you please close this Issue. You can continue to post updates. I wan't to remove from my backlog.

closed 2022-10-04 by a-cinar