Marus/cortex-debug #754: Peripheral register range read causes exceptions

BigTwisty, 2022-10-07:
I am debugging an ARM R52 core using J-Link via JTAG.  Cortex-debug is working wonderfully, with one exception.  The peripherals in this core have "reserved" addresses interspersed throughout their register memory ranges which do not physically exist in silicon.  Any attempt to read from them triggers an abort exception, crashing the application.  Other debuggers request each peripheral register address individually, as defined in the SVD file.  Cortex-debug seems to be requesting the entire range of memory defined for the peripheral, triggering the exception.

Is there a way to configure cortex-debug to request individual register values rather than a memory range when viewing peripherals?

PhilippHaefele, 2022-10-07:
Hi @BigTwisty,

yes there's a setting named `svdAddrGapThreshold` which can be set to -1 to disable the combined reads of peripheral registers

PhilippHaefele, 2022-10-07:
In general it's always good to see what [IntelliSense](https://code.visualstudio.com/docs/editor/intellisense) says or to have a look into https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md to find out if there's a option available 🙂

haneefdm, 2022-10-07:
Yes what @PhilippHaefele might work out. How may I ask they are marking regions as reserved?

It depends on the vendor but once they specify a range for a peripheral, they should allow reads on the entire range (without throwing exceptions) and protect themselves from read side-effects (like FIFO pointers). Sadly, not all vendors pay attention to this -- the BUS interface tells IP that the read is coming from a debug request vs. from within the core.

BigTwisty, 2022-10-07:
Thank you guys for the quick response!

@PhilippHaefele Believe me, I did my best to RTFM!  I read the tool tips for every option in the json settings that looked remotely like it could help.  The debug_attributes.md page is very helpful though.

@haneefdm The SVD file does not list any reserved regions.  It defines peripheral at base addresses.  These contain clusters at offsets from the peripheral base.  These contain individual registers at offsets from the cluster.  Any "reserved" register addresses are simply not defined as registers.  Setting the gap threshold to zero seems to be allowing the debugger to group valid register addresses without scanning any undefined addresses, so this was a great help!  Setting to -1 spammed the GDB bus like you would not believe!!!

I am still getting some read issues, but they seem to be on my end at this point.  Oh, the joys of dealing with early silicon development micros!

closed 2022-10-07 by BigTwisty

haneefdm, 2022-10-07:
Yes `-1` is super extreme but one stupid (remain unnamed) silicon vendor needed that, and there was another who defined byte-wide registers in a 32-bit arch. Normally people define 32-bit registers even if they only use 1-bit. It was a case where you could not even read adjacent registers/bytes together.

But @PhilippHaefele did direct you to experiment with the right option. Glad zero worked out for you

PhilippHaefele, 2022-10-07:
I just mentioned it because i see a lot of people not knowing how to use IntelliSense (btw. the debug_attributes is just a generated markdown output of it, which indeed helps when searching for something)

Yep -1 was maybe a bit to harsh.

Happy we could help you 🙂