Marus/cortex-debug #755: defmt rttConfig using JLink

miathedev, 2022-10-09:
Hi,

how do i add a defmt message decoder to my rttConfig?
When using defmt in combination with rtt decoders the data stream only outputs not readable data, eg:
`�r�Gr'W%2k`

my current launch.json
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Debug Server",
            "request": "launch",
            "type": "cortex-debug",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceFolder}/target/thumbv7em-none-eabihf/debug/embassy-stm32wl-examples",
            "servertype": "jlink",
            "interface": "swd",
            "device": "STM32WLE5JC",
            "runToMain": true,
            "preLaunchTask": "Cargo build",
            "breakAfterReset": true,
            "showDevDebugOutput": "raw",
            "serialNumber": "XXXXXXXXXXXXX",
            "postLaunchCommands": [
                "help"
            ],            
            "swoConfig": {
                "enabled": true,
                "cpuFrequency": 8000000,
                "swoFrequency": 2000000,
                "source": "probe",
                "decoders": [
                    { "type": "console", "label": "ITM", "port": 0 }
                ],
            },
            "rttConfig": {
                "enabled": true,
                "address": "auto",
                "decoders": [{"port": 0, "type": "console", "noprompt": true, "clear": false, "encoding": "signed"}]
            },
        }
    ]
}
```

Because im new to RTT and to be honest Rust in general, i tryed writing over RTT in general without defmt. It worked, but while using **defmt** data is kind of crypted, thats why i guess there must be a decoder for that. Is there one for this plugin? Or is there a workaround?

Versions
```
defmt = "0.3"
defmt-rtt = "0.3"
```

I would appreciate any hints. I hope it cant be that hard to get this working somehow. I cant be the only one failing on this.

haneefdm, 2022-10-09:
Disclaimer: I know nothing of Rust or `defmt`. It appears `defmt` is deferred formatting meaning the data stream has to be decoded when read from the host side. Well, we don't do that.

You can possibly specify a `rttConfig`  without any decoders. This should (did not try) a TCP port for you and have JLINK do all the right stuff to start getting data in a terminal window but doing the following

```
nc localhost 19021 | defmt-print -f <path-to-your-elf-file>
```

Your TCP port number may vary. On your system, `nc` maybe called `netcat` and you may have to install it. The `-f` could be `-e`

See https://ferrous-systems.com/blog/gdb-and-defmt/#defmt-print In this page, they setup RTT manually instead of using our `rttConfig` but it is the decoding part that is the important part.

Now, we do support multiple encodings for the RTT stream. I have not idea how to decode the dffmt stream. If we did, we could build the decoder into Cortex-Debug. Please enlighten us as to how to decode.

I see the following but can we break it down into pseudocode that I can translate to Typescript or maybe we can pipe our RTT stream into `defmt-print`. I have no way of experimenting.

https://ferrous-systems.com/blog/gdb-and-defmt/#defmt-print

miathedev, 2022-10-10:
Using netcat and piping it into defmt-print worked like a charm!

Yesterday i tryed to use:
```
"postLaunchCommands": [
    "monitor rtt server start 8765 0",
    // should span the *right* part of RAM
    "monitor rtt setup 0x2003fbc8 0x30 \"SEGGER RTT\"",
    "monitor rtt start",
],
```

from your link as well, but the commands are not supported by the JLinkGDBServer.
I think using a pipe to defmt-print would be the most OS agnostic way actually.

haneefdm, 2022-10-10:
Those commands are for OpenOCD. JLink mechanism is very different. We support both and your `rttConfig` does not change. We abstract the differences out.

Even all those commands in that web page were not needed (not anymore) and they don't work for a launch (maybe for an attach). Our method works and there is no need for specifying hard coded addresses like `0x2003fbc8`

I wanted to leave the author of the blog a comment but I did not see a way.

closed 2022-10-10 by haneefdm

Georges760, 2022-11-17:
Look like ferrous-systems did document the [defmt deserialization](https://defmt.ferrous-systems.com/deserialization.html)

Maybe it can be implemented in Cortex-Debug plugin ?

haneefdm, 2022-11-17:
> I have no way of experimenting.

This is my problem. If someone works with me and helps me, I can do it. Or you can submit a PR but it will not be simple if you are not used to TypeScript.

I don't want to spend days/weeks learning Rust and creating a test case.

Georges760, 2022-11-18:
I may be able to provide you some raw values (byte array) to clear text (string) exemples that you can use them for your test cases. (I am not into TS myself)

defmt is THE library for rust embedded logging, so by supporting rust in Cortex Debug plugin, many rust users may want to use defmt.

wrs, 2024-12-25:
FYI, I have defmt output working by leveraging the `logfile` option on the decoder. I use SWO rather than RTT, but I think the decoders are the same? My setup is:

In `launch.json`, set the `logFile` option to a tmp file, and add a command to clear the file on launch:
```
            "preLaunchCommands": [
                "shell echo -n '' > /tmp/itm.txt"
            ],
 ...
                "decoders": [
                    { "type": "console", "label": "ITM", "port": 0, "logfile": "/tmp/itm.txt" },
                ]
```

Install `defmt-print` (I used `cargo install defmt-print` — not sure if there are binary releases available).

Then just leave this command running in a terminal:
```
tail -F /tmp/itm.txt | defmt-print -e target/thumbwhatever/your-elf-file-name
```

ckrenslehner, 2025-03-09:
Thank you @wrs for sharing. I also got it working with my JLink and a powershell command. Auto restarting the task whenever you start a new debug session would be nice (or at least clearing the output), however I did not find a solution for this.

Here is my configuration:
launch.json
```
        {
            "preLaunchTask": "${defaultBuildTask}",
            "type": "cortex-debug",
            "name": "JLINK Debug",
            "cwd": "${workspaceFolder}",
            "executable": "./target/thumbv7em-none-eabihf/debug/${workspaceFolderBasename}",
            "request": "launch",
            "servertype": "jlink",
            "windows": {
                "serverpath": "${env:JLINK_PATH}/JLinkGDBServerCL.exe",
                "armToolchainPath": "${env:ARM_NONE_EABI_TOOLCHAIN_PATH}",
            },
            "serverArgs": [
                "-s",
                "-endian",
                "little",
                "-speed",
                "12000",
                "-vd",
                "-s",
            ],
            "interface": "swd",
            "device": "STM32WB55xx",
            "svdFile": "${workspaceFolder}/stm32wb55.svd",
            "rttConfig": {
                "enabled": true,
                "address": "auto",
            }
        },
```

task.json
```
        {
            "label": "Defmt RTT log",
            "type": "process",
            "command": "powershell",
            "args": [
                "-Command",
                "$ErrorActionPreference = 'SilentlyContinue'; while ($true) { echo '[waiting]'; Start-Sleep -Seconds 1; defmt-print -w -e ./target/thumbv7em-none-eabihf/debug/${workspaceFolderBasename} tcp }"
            ],
            "isBackground": true,
            "problemMatcher": []
        } 
```

Anyway, I hope there will be some progress soon on #880. Regardless, thank you to all maintainers for this great project! 💯🚀