Marus/cortex-debug #756: Failed to launch GDB: Remote doesn't know how to detach (from interpreter-exec console "detach")

atsju, 2022-10-12:
I have a need identical #445 (embedded developer need to flash program often but not always single step debugging).

I use latest jflash 7.80d 
GDB from `GNU Arm Embedded Toolchain 10.3-2021.10`

my launch.json:
```json
        {
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "cwd": "${workspaceRoot}",
            // https://vector-of-bool.github.io/docs/vscode-cmake-tools/debugging.html
            "executable": "${command:cmake.launchTargetPath}",
            "name": "Flash program (J-Link)",
            "device": "STM32U575QI",
            "interface": "swd",
            "serverArgs": [
            ],
            "ipAddress": null,
            "serialNumber": null,
            "postLaunchCommands": [
                "monitor reset",
                "monitor go",
                "detach",
            ],
            "showDevDebugOutput": true,
        },
```

the output: 
```
-> 13^done
14-interpreter-exec console "monitor reset"
-> @"Resetting target\r\n"
Resetting target
-> 14^done
15-interpreter-exec console "monitor go"
-> 15^done
16-interpreter-exec console "detach"
-> ~"Detaching from program: C:\\FW.axf, Remote target\n"
Detaching from program: C:\FW.axf, Remote target
-> &"Remote doesn't know how to detach\n"
Remote doesn't know how to detach
-> 16^error,msg="Remote doesn't know how to detach"
Failed to launch GDB: Remote doesn't know how to detach (from interpreter-exec console "detach")
```

"Disconnect" instead of "detach" works but from what I read [here](https://www.embecosm.com/appnotes/ean4/html/ch03s02s10.html) disconnect will not let the target run. 

I want to be able to flash my target and if possible avoid to do a manual reset (power on or external) to effectively run the program. Can you help me out ?

haneefdm, 2022-10-12:
This is a known problem with JLink.

I have tried many times and there is no way to fix this. This is a SEGGER problem. They need to respect the `extended-remote` protocol and do the right thing. Something is not right in the communication between gdb and JLINK where we are  not involved. You need to lookmat the command sent by gdb remote protocol and not just what JLink reports. I am 99% certain they send a detach request which is nicely ignored. See sections 3.2.10/3.3 and the `D` command/packet from GDB. I think the following may be the command and JLink does nothing with it.

```
00-00000824-00-00004401-0005: $D#44
```

https://www.embecosm.com/appnotes/ean4/embecosm-howto-rsp-server-ean4-issue-2.html#id3081722

You can debug gdb transactions using some gdb variables like `remotelogfile`

https://sourceware.org/gdb/onlinedocs/gdb/Remote-Configuration.html

Please file an issue with them. I am not a customer of theirs, so I can't. 

I am closing this issue as there is nothing I can do. If you have information from SEGGER, let us know.

atsju, 2022-10-12:
I will try to open the issue with Segger, they are usually very reactive. However I'm not sure to have the technical level to answer their questions. Do you own a Jlink ? If so, may I put you in copy of the futur exchanges with Segger ?

atsju, 2022-10-12:
Well, through technical support email I always got on topic and fast answers in the past but maybe it's not a general/shared experience. My team/company regularly acquires Segger products so it might help. 

I will try to manage the thing an may do the interface with you if I need better understanding. If you are interested in playing/maintaining/debugging/investigating deeper I might have a spare Jlink base to offer. But it will not bring much compared to jlink flashed on ST eval board.

Note I will give a link to this thread in my email.

haneefdm, 2022-10-12:
Sure, you can attach this issue link and I will keep it open for a bit. I can help them test a fix if needed. I know first hand, some of their customer who will like this and will appreciate it.

PS: I edited my comment above.

atsju, 2022-10-12:
Thank you. 

I also confirm this is related to detach seen (and ignored?) by jlink log
```
02-00000000-00-00077284-0025: T6700 077:285.792 - 2.167ms returns 0
00-0000084C-00-00078001-0005: $D#44
01-0000084C-00-00078001-0004: $#00
02-00000000-00-00079493-0028: T6700 079:495.416 JLINK_GetHWStatus(...)
```

haneefdm, 2022-10-12:
Yup. Any line that does not contain a timestamp is either a request from gdb or a response from JLink. Sometimes, hard to tell. I wish they had arrows indicating direction. The rest is chatter created by JLink itself and may contain it talking to the JLink HW. For the following request,

```
00-0000084C-00-00078001-0005: $D#44
```

I expect something like
```
01-00000824-00-00004387-0006: $OK#9a
```
and I see none. Of course, I also expect it does the right thing before sending the `OK`

Side note:
BTW, this is very common on desktop applications since 1980's. We would attach to a running process (like daemons/services) and detach without killing the process by doing so. This is not something new.

SEGGER-Alex, 2022-10-17:
Hi,

SEGGER J-Link PM here.
Not sure where the "this is a known problem" comes from because I do not recall seeing anything in our release notes saying that this is a known problem.

What GDB Server currently does is sending an empty packet as response to the detach request, to indicate that it does not support this command. It is a quite common thing done by many GDB stubs when receiving an unsupported command, especially for commands where the online documentation does not explicitly state if a command is mandatory or optional to support.
Has been observed many many times over the past 10+ years with various GDB stubs.

Not sure if 'detach' is a mandatory command.
We will consider adding support for the detach packet 'D' in one of the next versions, but I cannot promise anything at this point.

Regarding the log file:
"I wish they had arrows indicating direction."
While the format is not documented for a good reason (log files are to be interpreted by SEGGER only and the format may change at any time), some hints here:
  _00-xxxxxxxx-xx-xxxxxxxx GDB -> GDB Server traffic
  01-xxxxxxxx-xx-xxxxxxxx GDBServer -> GDB traffic
  02-xxxxxxxx-xx-xxxxxxxx J-Link DLL API call related
  03-xxxxxxxx-xx-xxxxxxxx GDB Server log output (also shown in GDB Server I/O area_

Again: No guarantee that the log file format stays this way.

BR
Alex

atsju, 2022-10-17:
@SEGGER-Alex 

Thank you for your prompt answer.

> "this is a known problem" 

I think this is a known problem *for this repository*. Other users have already shared the missing `detach` command here, but I'm first person officially requesting it to Segger.

Please do your best to add the detach command to the roadmap as it would help also other users than me. I understand you cannot promise anything yet.

haneefdm, 2022-10-17:
Know to us for a couple of years. There is no missing detach here. In VSCode, you end a session two ways

1. Disconnect -- end session, No detach -- this will have a missing detach
2. Detach -- end session with detach.

The commands we send out are easily visible in our Debug Console.

In the first case, there is no detach. Yes we have had in the past with various gdb-servers trying to get detach work the way gdb documents and.

Please also see the JLink logs above. You will see that a detach was sent.

haneefdm, 2022-10-17:
PS: I edited my previous message.

haneefdm, 2022-10-17:
Sorry for my deleted messages. I replied to the wrong person.

To @SEGGER-Alex, thanks for looking into it. Yes, we have known about `detach` not being supported by JLink server for a couple of years. Users always come to us complaining. Not much we can do.

As for where it is required by GDB, I don't know, I doubt it, But this has been the convention documented by gdb for decades. Even in the 1990's I was using `detach` on desktop running programs (remote or local) and detach always meant let the program continue to run. There was no other good way of debugging daemons/services. We do the same thing today.

You are probably already familiar with the following links but this is the expected lifecycle of a debugged program.

https://sourceware.org/gdb/onlinedocs/gdb/Connecting.html
https://ftp.gnu.org/old-gnu/Manuals/gdb/html_node/gdb_22.html

And, VSCode went to great lengths to add the detach feature across all debuggers in the UI, APIs, documentation, etc. They make a clear distinction between just ending a session, and ending a session with a detach.

When `detach` is supported, it is very helpful. Even for running tests. We do this with other gdb servers.

haneefdm, 2022-10-21:
@atsju I am closing this. If SEGGER wants to engage on this can re-open. I don't have any hope for that.

SEGGER-Alex, 2022-11-09:
@haneefdm You are quite pessimistic here. We took the effort and got involved here, so of course we also considered to engage further...

Will be fixed in today's version (V7.82b) which is expected to go online within the next 4 hours from now.
https://www.segger.com/downloads/jlink/

haneefdm, 2022-11-09:
@SEGGER-Alex Thank you.

haneefdm, 2022-11-09:
Reopened as promised. I will test V7.82b this weekend. I don't think any changes will be required for this extension as we do the same thing for all gdb-servers.

Thanks again @SEGGER-Alex

@atsju, you can have a go at V7.82b when it is available...you may be able to get to it before I can.

atsju, 2022-11-10:
This is my config to simply flash and then stop the debug while board keeps running
```json
        {
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "cwd": "${workspaceRoot}",
            // https://vector-of-bool.github.io/docs/vscode-cmake-tools/debugging.html
            "executable": "${command:cmake.launchTargetPath}",
            "name": "Flash program (J-Link)",
            "device": "STM32U575QI",
            "interface": "swd",
            "serverArgs": [
            ],
            "ipAddress": null,
            "serialNumber": null,
            "postLaunchCommands": [
                "monitor reset",
                "monitor go",
                "detach",
                "disconnect",
            ],
        },
```
It seems to work (new program runs after detach) but I need to do `detach` + `disconnect` so that VScode goes out of debug mode (not just `detach`). I'm unsure if this is expected behavior.

haneefdm, 2022-11-10:
You should try detach+disconnect in `postStartSessionCommands` and not part of the launch., Let the session start and then end it. Or you will get an error saying gdb quit before the session started and we would not know why. Here is the entire sequence.

Sometimes, `overrideLaunchCommands` to setup the whole set of commands to start the session. Including programming if needed.

https://github.com/Marus/cortex-debug/wiki/Cortex-Debug:-Under-the-hood

haneefdm, 2022-11-11:
Oh, you may also want to add a 'quit' command after disconnect, if things are not completely quitting.

haneefdm, 2022-11-13:
I finished testing and it appears that JLink detach is working as expected. Mind you this is not an exhaustive test and Cortex-Debug does a few things beyond what gdb calls for for a clean detach and I didn't want to disable any of that. All gdb servers have some latent issues with detach and hence our cleanup prior to detach.

I know in my own company, they will appreciate this even though they may have to modify the tests to change expected false -ve to a +ve.

Thanks @SEGGER-Alex, @atsju 

I am closing this now.

haneefdm, 2022-11-13:
FYI: I just submitted an enhacement request wrt. RTT. We will see how it goes.

> Your request has been received and will be reviewed by our support staff.
> 
> Case: 60296575 (https://support.segger.com/#ticket/zoom/29689)
> To provide additional information you can simply reply to this e-mail or use the web interface via the link from above.
> 
> Our FAQ can be accessed here: https://wiki.segger.com/Main_Page
> 
> Your SEGGER Helpdesk Team

closed 2022-11-13 by haneefdm