Marus/cortex-debug #759: New MemoryView missing some features

Norm369, 2022-10-18:
Seems we can't copy a range of data to the clipboard. This was very useful since I don't see any other way of transferring large arrays to another app (eg. for plotting). Being able to specify the size was useful too: if you want 10000 bytes, you could enter 10000 for the size then Ctrl-A to select all and Ctrl-C to copy. Finally, specifying a C expression (eg. name of a global) doesn't seem to work, or at least not the way it did in the previous mem viewer.

haneefdm, 2022-10-18:
Do you want to file an issue against this extension or the MemoryView extension?

I did the C expression for both Cortex-Debug and MemoryView. It does the exact same thing. Again, if it is not working right in MemoryView, you should file an issue there.

MemoryView extension is brand new. We still haven't done the selection and clipboard work yet over there and editing is not done. I am glad you pointed it out. But in the wrong repo. I know it is a bit confusing

Needing a size was one big complaint because most IDEs don't ask for a size. Including Microsoft default viewer.

If you want the legacy memory viewer, you can have that, perhaps. I will think about it. It is end of life.

But if you want to move forward and make things better then MemoryView is where it will happen.

haneefdm, 2022-10-18:
Could I encourage you to file an issue (multiple ones if unrelated) over at

https://github.com/mcu-debug/memview

Norm369, 2022-10-18:
Thanks for the prompt response. I filed the issues (two separate ones now) at mcu-debug/memview. The C expression handling clearly does not work the same in cortex-debug 1.6.4 and memview.

Lack of any way to transfer large arrays into another program (eg. via clipboard) is a serious deficiency, in my opinion.  Most IDEs offer some way to do this. The problem is not with memview per se, just that cortex-debug doesn't offer any other way to get big arrays out, as far as I can see.

Thanks
Norm

https://github.com/mcu-debug/memview

Norm369, 2022-10-18:
moved to new issues at mcu-debug/memview

closed 2022-10-18 by Norm369

haneefdm, 2022-10-18:
I want to see if I want to expose the old memory viewer as a legacy function that I can deprecate later.