Marus/cortex-debug #762: JLink does not handle 'mon halt reset' from servertype 'external'

JojoS62, 2022-10-21:
cortex-debug version: 1.6.6
JLink: 7.98 64 Bit Windows (also 6.98 32 Bit Windows)

I've got a problem with debugging a Mbed target. When Cmake is generating the debug configuration, it uses servertype 'external'. This wants to halt and reset the target via 'mon halt reset', but the command is not accepted by the JLink probe (NUCLEO board with J-Link Firmware).
When I set the servertype to 'jlink', debugging works because attach/launch handle the commands different as two single commands.

The launch.json for both cases and the output is shown in:
https://github.com/mbed-ce/mbed-os/issues/77

Using servertype external maybe easier to use for a common configuration, or do you recommend to use the specific servertypes?

JojoS62, 2022-10-21:
thanks for your comment.
yes, I meant 'monitor halt reset', it is not overriden in the generated launch.json.
So your comment is very helpful and I see it is difficult to do the same work again for any external servertype.

haneefdm, 2022-10-21:
Yes, every server is different and we can only do so much.

Closing this for now.

closed 2022-10-21 by haneefdm