Marus/cortex-debug #763: Connection getting closed with pyocd

danielRep, 2022-10-24:
I started a configuration in Ubuntu 20.04 of a Cortex-Debug setup with pyocd. The connection between gdb and pyocd is being closed and I can't quite figure out why. I will share the outputs from both terminals.

These are the `launch.json` settings that I'm using:
 ```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "muscaa1: contention-eval",
            "cwd": "${workspaceFolder}",
            "executable": "contention-eval/bin/core0/core0.elf",
            "request": "launch",
            "type": "cortex-debug",
            "svdFile": "/home/daniel/workspace/Musca_A1.svd",
            "runToEntryPoint": "reset_handler",
            "servertype": "pyocd",
            "showDevDebugOutput": "raw",
            "debuggerArgs": ["-nx"],
            "preAttachCommands": ["monitor reset halt"],
            "serverArgs": ["-v", "-L*.cortex_m*=debug"]
        }
    ]
}
```
The output that I have from Cortex-Debug console:
```
Cortex-Debug: VSCode debugger extension version 1.6.6 git(9844153). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from /home/daniel/toolchains/xpack-arm-none-eabi-gcc-11.3.1-1.1/bin/arm-none-eabi-objdump --syms -C -h -w /home/daniel/workspace/predict-mech/contention-eval/bin/core0/core0.elf
Reading symbols from /home/daniel/toolchains/xpack-arm-none-eabi-gcc-11.3.1-1.1/bin/arm-none-eabi-nm --defined-only -S -l -C -p /home/daniel/workspace/predict-mech/contention-eval/bin/core0/core0.elf
Launching GDB: /home/daniel/toolchains/xpack-arm-none-eabi-gcc-11.3.1-1.1/bin/arm-none-eabi-gdb -q --interpreter=mi2 -nx
1-gdb-version
Launching gdb-server: /home/daniel/.local/bin/pyocd gdbserver --port 50000 --telnet-port 50001 -v "-L*.cortex_m*=debug"
    Please check TERMINAL tab (gdb-server) for output from /home/daniel/.local/bin/pyocd
Finished reading symbols from objdump: Time: 47 ms
Finished reading symbols from nm: Time: 46 ms
-> =thread-group-added,id="i1"
-> ~"GNU gdb (xPack GNU Arm Embedded GCC x86_64) 12.1.90.20220802-git\n"
-> ~"Copyright (C) 2022 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=x86_64-pc-linux-gnu --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source /home/daniel/.vscode/extensions/marus25.cortex-debug-1.6.6/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source /home/daniel/.vscode/extensions/marus25.cortex-debug-1.6.6/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-file-exec-and-symbols "/home/daniel/workspace/predict-mech/contention-eval/bin/core0/core0.elf"
-> 8^done
9-target-select extended-remote localhost:50000
-> 9^error,msg="Remote connection closed"
10-interpreter-exec console "set mem inaccessible-by-default off"
Failed to launch GDB: Remote connection closed (from target-select extended-remote localhost:50000)
[... 118 lines cut ...]
0000929 D halting core 0 [cortex_m]
0000929 D Setting vector catch to 0x00000001 [cortex_m]
0000931 I Client connected to port 50000! [gdbserver]
0000932 D GDB received query: [b'Supported', b'multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+;memory-tagging+#ec'] [gdbserver]
0000937 D uninit session <pyocd.core.session.Session object at 0x7fb254d2ccd0> [session]
0000938 D uninit board <pyocd.board.mbed_board.MbedBoard object at 0x7fb254ce60d0> [board]
0000942 D resuming core 0 [cortex_m]
0000943 D added=[] removed=[] [manager]
0000943 D bps after flush={} [manager]
0000943 D added=[] removed=[] [manager]
0000943 D bps after flush={} [manager]
0000949 D cannot resume: target not halted [cortex_m]
0000957 D closing interface [pyusb_backend]
0000959 D gdbserver shutdown event set; waiting for exit [gdbserver]
0000975 D gdbserver exiting connection loop [gdbserver]
0000975 I Client disconnected from port 50000! [gdbserver]
0000975 D gdbserver thread exiting [gdbserver]
0000975 D GDB server cleaning up [gdbserver]
0001075 D Error in packet IO thread: [Errno 9] Bad file descriptor [packet_io]
0001075 D GDB packet thread stopping [packet_io]
[2022-10-24T12:25:28.924Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

Am I missing something on the configuration? Thank you.

haneefdm, 2022-10-24:
An excellent question for the pyocd team.

We simply asked gdb to start a connection to pyocd and it returned an error. We are not involved in anything. Perhaps you need some special options on pyocd command line since you are dual-core.

```
9-target-select extended-remote localhost:50000
-> 9^error,msg="Remote connection closed"
```
We never talk to pyocd directly.

First, I would make sure that gdb and pycod run from the command line and are able to debug. We don't do anything that you cannot do in a shell and all commands are visible to you. Most of the command line options for pyocd are already shown in Debug Console. Modify and run accordingly.

haneefdm, 2022-10-24:
We check to make sure that any port we select is not in use. 

Did you try to set `numberOfProcessors` (should be `2`) and `targetProcessor` `(0 or 1)`? This will allocate the right amount of ports and thus avoid conflicts.

danielRep, 2022-10-24:
That did the trick @haneefdm . Adding those two options into the launch settings solved the problem. I wasn't about them. Thank you for the help.