Marus/cortex-debug #765: Force disassembly no longer works

Norm369, 2022-10-25:
On Cortex-debug 1.6.4 (on iMX RT1172) choosing Cortex-Debug: Set Force Disassembly then Forced: Always show Disassembly for Functions, no longer works. It used to bring up a window with the disassembled memory in it; now it does nothing as far as I can tell. I'm using this on functions for which the symbol table is available (since I'm able to display and modify variables using their names, and step through the C code line by line etc.).

Norm369, 2022-10-25:
> Sorry, our custom Disassembly functionality is going away. Sorry some remnants have remained as we haven't finished cleanup
>
> We have moved to the disassembly (infrastructure) provided by VSCode. They provide the UI and infrastructure and we provide the content. We worked with MS over the past two years to create this
>
> https://github.com/Marus/cortex-debug/wiki/Disassembly-Debugging
> https://devblogs.microsoft.com/cppblog/visual-studio-code-c-july-2021-update-disassembly-view-macro-expansion-and-windows-arm64-debugging/

Toggle Source Code in Disassembly View seems to do nothing at all.

Thanks
Norm

haneefdm, 2022-10-25:
Did you get a disassembly window? You can only seem something change if there is such a window

haneefdm, 2022-10-25:
The easiest way is to right-click on the stack frame and launch the disassembly window

Norm369, 2022-10-25:
No window appeared.

haneefdm, 2022-10-25:
No window will appear. It only change what a window looks like if it was there.

haneefdm, 2022-10-25:
This is not our feature. It is embedded deeply into VSCode now.

haneefdm, 2022-10-25:
> No window appeared.

I am sorry. I am confused. Tell me what steps you took. Like "I did this, followed by that and this is what happened..."

Norm369, 2022-10-25:
My mistake - I thought that Toggle Source Code in Disassembly command would act like the old Force Disassembly command and bring up a disassembly window if one wasn't open. But it doesn't. I followed your suggestion to right-click on the call stack and that does work properly.

Thanks
Norm

haneefdm, 2022-10-25:
It is like if you change tab-width or font-size nothing happens until you open an editor. To them that is just a setting in that window.

[image]

closed 2022-10-25 by haneefdm