Marus/cortex-debug #766: Debugger assumes arm-none-eabi-gdb file name

rerickson1, 2022-10-25:
When trying to use cortex-debug it assumes the GDB executable is named arm-none-eabi-gdb.
With the Zephyr RTOS and its SDK, the executable is named arm-zephyr-eabi-gdb. Is there a way to account for this in the extension settings? If not, could a new option be added?

haneefdm, 2022-10-25:
There are various ways to specify paths. Try `toolchainPrefix` (default is ''arm-none-eabi")

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

haneefdm, 2022-10-25:
Sorry, in your settings, it is called `armToolchainPrefix` which is not documented as part of debug attributes as that is part of settings.

rerickson1, 2022-10-25:
Thank you, setting "toolchainPrefix": "arm-zephyr-eabi" works perfectly.

closed 2022-10-25 by haneefdm

rerickson1, 2022-10-25:
Thanks, I tested the `armToolchainPrefix` approach as well and that works too.