Marus/cortex-debug #767: "request": "attach" always halts CPU

rerickson1, 2022-10-25:
When attaching it always halts the CPU.
What is the best way to attach to the target without halting the CPU?

In my use case, running the Zephyr RTOS with BLE, halting the CPU will cause a hard fault. Being about to attach without needing to reset the CPU every time is desirable.

haneefdm, 2022-10-25:
You can override the halt we perform with an attach by using `"overrideAttachCommands": []`, but what we did was probably redundant as the act of an `attach` via gdb probably already halted your CPU.

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

But, if things don't work right, please do not file any issues.

rerickson1, 2022-10-25:
Thank you for the input, it seems there's no way around the issue for my use case.
Tried `continue` in `postAttachCommands` and the `"overrideAttachCommands": []` and in both cases the CPU is halted long enough to cause the BLE controller to ASSERT due to timing requirements and cause a fault.

I was hoping I could replicate an attach like Segger Ozone does. An attach with that tool does no require halting the CPU.

closed 2022-10-25 by rerickson1

haneefdm, 2022-10-25:
Yeah, it is a limitation of gdb protocol.