Marus/cortex-debug #768: PERIPHERLA VIEW not working for chip DA14531

HugoPentalock, 2022-10-27:
Hi,
My PERIPHERAL VIEW just say "Waiting for register data from debugger..." when using a DA14531 chip from Dialog semiconductor.

I use the DA14531.svd file found here:
https://www.keil.com/dd2/dialogsemiconductor/da14531/#!#third-party-download-dialog

It worked some time ago.

haneefdm, 2022-10-27:
SVD has not changed in a while.

It appears that we are unable to read memory. Can you enable debug (See instructions in Debug Console) and show us the contents of the Debug Console

HugoPentalock, 2022-10-28:
Sure. Output form debug console attached

[Debug_console.txt](https://github.com/Marus/cortex-debug/files/9888469/Debug_console.txt)

haneefdm, 2022-10-28:
Peripherals are populated by reading memory when program in in stopped state. I see the following reads and all succeeded.
```
31-data-read-memory-bytes "0xe000e100" 4
-> 31^done,memory=[{begin="0xe000e100",offset="0x00000000",end="0xe000e104",contents="00000000"}]
32-data-read-memory-bytes "0xe000e180" 4
-> 32^done,memory=[{begin="0xe000e180",offset="0x00000000",end="0xe000e184",contents="00000000"}]
33-data-read-memory-bytes "0xe000e200" 4
-> 33^done,memory=[{begin="0xe000e200",offset="0x00000000",end="0xe000e204",contents="00000100"}]
34-data-read-memory-bytes "0xe000e280" 4
-> 34^done,memory=[{begin="0xe000e280",offset="0x00000000",end="0xe000e284",contents="00000100"}]
35-data-read-memory-bytes "0xe000e400" 32
-> 35^done,memory=[{begin="0xe000e400",offset="0x00000000",end="0xe000e420",contents="0000000000000000000000000000000000000000000000000000000000000000"}]
```

So, I don't see any issues here. You said you saw the following message and I cannot see such a message in our code. Are you sure you are using our SVD viewer? There are other extensions too.

> Waiting for register data from debugger...

Can you tell me the detailed steps to reproduce and screenshots.

HugoPentalock, 2022-10-28:
[image]

Attached a screen shot. I think the "PERIPHERAL VIEW" is part of Cortex-Debug extension.
To reproduce I simply start a debug session, where everything else like start/stop, single step, breakpoints and CORTEX registers are working fine :-).
Just the Peripheral registers are missing.

haneefdm, 2022-10-28:
That is not our window. Our window is called "CORTEX PERIPHERALS". That other window ("PERIPHERALS VIEW") is probably from the "Embedded Tools" extension from Microsoft.

closed 2022-10-28 by haneefdm