Marus/cortex-debug #772: OpenOCD server: Spelling mismatch for RTOS 'ChibiOS' vs. 'chibios'

orwell96, 2022-11-10:
I am trying to debug a ChibiOS application using the OpenOCD debug server (on an STM32 using ST devboard) and noticed a minor oversight in the config validation. My debug configuration looks like this:
```
{
            "type": "cortex-debug",
            "request": "attach",
            "name": "Debug (OpenOCD)",
            "servertype": "openocd",
            "rtos": "ChibiOS",
[...]
        }
```
However in this setup, openocd errors out with the message:
```
extensions/marus25.cortex-debug-1.6.7/support/openocd-helpers.tcl:33: Error: Unknown RTOS type ChibiOS, try one of: ThreadX, FreeRTOS, eCos, linux, chibios, Chromium-EC, embKernel, mqx, uCOS-III, nuttx, RIOT, hwthread,  or auto
```
Apparently, for openocd the RTOS has to be specified as `chibios`, not `ChibiOS`

Using `"rtos": "chibios"` in the launch.json does not work, the plugin displays an error message on the bottom right telling that 'chibios' is not a supported value for rtos:
```
The following RTOS values are supported by OpenOCD: ChibiOS eCos embKernel FreeRTOS mqx nuttx ThreadX uCOS-III auto
```

Changing "ChibiOS" to "chibios" [here](https://github.com/Marus/cortex-debug/blob/b0f5563a29f5dbe2529b6f49a3bfb1ba6a7c1b41/src/frontend/configprovider.ts#L10) and then providing `"rtos": "chibios"` fixes the issue and makes OpenOCD start up correctly.

haneefdm, 2022-11-10:
The list of names is taken from here (search for ChibiOS).

https://openocd.org/doc/html/GDB-and-OpenOCD.html

Looks like they are not documented right. It is also case-sensitive because "FreeRTOS" is how you ask for FreeRTOS? Now, we have to scour through openocd to find out what the actual names are.

https://github.com/openocd-org/openocd/blob/3ca7bc7354ea3f8db8384142c2bd8675789e7888/src/rtos/chibios.c#L101
https://github.com/openocd-org/openocd/blob/3ca7bc7354ea3f8db8384142c2bd8675789e7888/src/rtos/FreeRTOS.c#L98

Btw, I always use "auto". It is simple and fast.

haneefdm, 2022-11-10:
Had to go through openocd source code and get all of them right.

Will be fixed in next release. Until then, "auto" should work pretty well.

closed 2022-11-10 by haneefdm