Marus/cortex-debug #775: Cannot debug code from RAM memory...

LAPSyLAB, 2022-11-19:
Hello,

not sure if this is a missing feature or just something that is not working properly on latest versions, but seems that when you try to debug when code is placed in RAM memory, code window somehow sticks to 0x08000..... memory address which is usually in Flash. It seems that registers are updating properly (not sure) when single stepping, but code window seems to have predefined address that points to Flash.

It would be a greate feature - we are using debug from RAM under CubeIDE with no problem, CubeIDE generates FLASH and RAM linker scripts, so it seems not so hard to fix this.

Is starting address for debug hardcoded or is accessible as a setting ?

Thanks in advance,
regards,
LapsyLAB.

haneefdm, 2022-11-19:
Not sure what you mean by "code window"

We do not mess with start addresses or even care. We get the register values from gdb which in turn gets it from your gdb-server (OpenOCD, JLink, STLink, etc.)

All we do is issue a halt. Program. reset and continue. You may have bootloaders and whatnot and we do not know. We simply let it run until  it stops for whatever reason (breakpoint, exception, pause-button, etc.)

It is entirely possible that your linker script is wrong and the VMA, LMA address are not right and gdb is confused. Note that even CueIDE use gdb. Given the same executable, and same gdb-server you should get same results with both us nad CubeIDE -- unless there are some scripts running to modify GDB behavior.

haneefdm, 2022-11-19:
Btw, gdb simply reacts to the PC and SP returned by the gdb-server. It has no other way to know.

LAPSyLAB, 2022-11-19:
Hello, thanks for info and comments... Now I will be able to check this more in details. 
Will report back, probably this is not an cortex-debug issue...

haneefdm, 2023-03-13:
Closing this as there is not much actionable here.

closed 2023-03-13 by haneefdm