Marus/cortex-debug #776: Cortex-debug doesn't debug on Vs Code WSL

Dr-Itachi, 2022-11-21:
Windows:
Windows 11
WSL:
WSL 2 Ubuntu 20.04

My Vs code installation doesn't launch debugging no way what i did it didn't work. It builds the project successfully but after that the debug server doesn't launch. It stops after the Prelaunchtask. Here is my launch.json

"version": "0.2.0",
"configurations": [
{
"name": "Launch",
"type": "cortex-debug",
"request": "launch",
"cwd": "${workspaceRoot}",
"executable": "${workspaceRoot}/build/${workspaceRootFolderName}.elf",
"showDevDebugOutput": "raw",
"servertype": "openocd",
"interface": "swd",
"runToEntryPoint": "main",
"device": "STM32F746ZG",
"configFiles": [
"board/st_nucleo_f7.cfg"
],
"preLaunchTask": "Hermann STM32 : Build",
"svdPath": "${workspaceRoot}/STM32F746.svd"
}
]
}

haneefdm, 2022-11-21:
We do not do anything to support WSL. The main thing to remember is that the debugger runs on the Linux/WSL side.

Naturally, gdb works fine. The thing you have issues with is gdb-servers (like OpenOCD). They do not work. If they don't work, neither can we. You need proper USB support inside WSL and a proper gdb-server. More people have succeeded with JLink

See this thread (and others if you search the Issues list).

https://github.com/Marus/cortex-debug/issues/467

Scroll to the bottom. You can post your questions on StackOverflow and see if someone can help you. I use a Mac so, I can't help.

haneefdm, 2022-11-21:
Well, did you set the path for OpenOCD? In the Debug Console you should see the exact command being launched and try to run the same on the command-line.

Dr-Itachi, 2022-11-21:
> Well, did you set the path for OpenOCD? In the Debug Console you should see the exact command being launched and try to run the same on the command-line.

The problem is that after the building process it stops. it doesnn't call openocd

haneefdm, 2022-11-21:
Check the "Debug Console" and copy/paste the contents here.

Dr-Itachi, 2022-11-22:
> 

Exécution de la tâche : scan-build compiledb make -j4 

scan-build: Using '/usr/lib/llvm-10/bin/clang' for static analysis
## Building [make -j4]...
arm-none-eabi-gcc -c -mcpu=cortex-m7 -mthumb -mfpu=fpv5-sp-d16 -mfloat-abi=hard -DUSE_FULL_LL_DRIVER -DHSE_VALUE=8000000 -DHSE_STARTUP_TIMEOUT=100 -DLSE_STARTUP_TIMEOUT=5000 -DLSE_VALUE=32768 -DEXTERNAL_CLOCK_VALUE=12288000 -DHSI_VALUE=16000000 -DLSI_VALUE=32000 -DVDD_VALUE=3300 -DPREFETCH_ENABLE=0 -DART_ACCELERATOR_ENABLE=0 -DSTM32F746xx -ICore/Inc -IDrivers/STM32F7xx_HAL_Driver/Inc -IDrivers/CMSIS/Device/ST/STM32F7xx/Include -IDrivers/CMSIS/Include -I/usr/local/gcc-arm-none-eabi-10.3-2021.10/arm-none-eabi/include -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/hw_desc_comT02_stm32f7.d" -Wa,-a,-ad,-alms=build/hw_desc_comT02_stm32f7.lst Core/Src/hw_desc_comT02_stm32f7.c -o build/hw_desc_comT02_stm32f7.o
arm-none-eabi-gcc -c -mcpu=cortex-m7 -mthumb -mfpu=fpv5-sp-d16 -mfloat-abi=hard -DUSE_FULL_LL_DRIVER -DHSE_VALUE=8000000 -DHSE_STARTUP_TIMEOUT=100 -DLSE_STARTUP_TIMEOUT=5000 -DLSE_VALUE=32768 -DEXTERNAL_CLOCK_VALUE=12288000 -DHSI_VALUE=16000000 -DLSI_VALUE=32000 -DVDD_VALUE=3300 -DPREFETCH_ENABLE=0 -DART_ACCELERATOR_ENABLE=0 -DSTM32F746xx -ICore/Inc -IDrivers/STM32F7xx_HAL_Driver/Inc -IDrivers/CMSIS/Device/ST/STM32F7xx/Include -IDrivers/CMSIS/Include -I/usr/local/gcc-arm-none-eabi-10.3-2021.10/arm-none-eabi/include -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/hw_desc_timer_f746.d" -Wa,-a,-ad,-alms=build/hw_desc_timer_f746.lst Core/Src/hw_desc_timer_f746.c -o build/hw_desc_timer_f746.o
arm-none-eabi-gcc build/main.o build/gpio.o build/stm32f7xx_it.o build/system_stm32f7xx.o build/hw_desc_comT02_stm32f7.o build/hw_desc_timer_f746.o build/stm32f7xx_ll_gpio.o build/stm32f7xx_ll_rcc.o build/stm32f7xx_ll_utils.o build/stm32f7xx_ll_exti.o build/startup_stm32f746xx.o -mcpu=cortex-m7 -mthumb -mfpu=fpv5-sp-d16 -mfloat-abi=hard -specs=nano.specs -TSTM32F746ZGTx_FLASH.ld  -lc -lm -lnosys  -Wl,-Map=build/TimerSTM32.map,--cref -Wl,--gc-sections -o build/TimerSTM32.elf
arm-none-eabi-size build/TimerSTM32.elf
   text    data     bss     dec     hex filename
   2064      12    1660    3736     e98 build/TimerSTM32.elf
arm-none-eabi-objcopy -O ihex build/TimerSTM32.elf build/TimerSTM32.hex
arm-none-eabi-objcopy -O binary -S build/TimerSTM32.elf build/TimerSTM32.bin

scan-build: Removing directory '/tmp/scan-build-2022-11-22-101000-18824-1' because it contains no reports.
scan-build: No bugs found.
 *  Le terminal sera réutilisé par les tâches, appuyez sur une touche pour le fermer.

Dr-Itachi, 2022-11-22:
The waiting bar in the Vs code debug window stops right after the prelaunchtask and doesn't seems to look at the next step after that

haneefdm, 2022-11-22:
> The problem is that after the building process it stops. it doesn't call openocd

VSCode does not call openocd. They call us and give us the launch.json portion that matters. We take it from there and if your servertype is openocd, WE run openocd. We do all the work but VSCode does not call us until the pre-launch task finishes and actually exits. Comment that pre-launch task temporarily to experiment.

We have nothing to do with the build/pre-launch task.   We only know how to debug when asked. But it appears that no one is even asking.

From your build long, it appears that the build is finished., This maybe something you ask VSCode folks. Especially if the Debug Console is empty.

closed 2022-11-22 by haneefdm

Dr-Itachi, 2022-11-22:
> 

After uncommenting prelaunchtask i get this on the debug console:

Cortex-Debug: VSCode debugger extension version 1.6.7 git(b0f5563). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /home/voldemort/.STM32Codes/TimerSTM32/build/TimerSTM32.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /home/voldemort/.STM32Codes/TimerSTM32/build/TimerSTM32.elf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
Error: arm-none-eabi-nm failed! statics/global/functions may not be properly classified: Error: spawn arm-none-eabi-nm ENOENT

Dr-Itachi, 2022-11-22:
And this in the Terminal:

Waiting for gdb server to start...[2022-11-22T22:26:20.930Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
mnt/c/openocd/bin/openocd.exe -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/voldemort/.STM32Codes/TimerSTM32 -f /home/voldemort/.vscode-server/extensions/marus25.cortex-debug-1.6.7/support/openocd-helpers.tcl -f board/st_nucleo_f7.cfg
[2022-11-22T22:26:22.400Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...

Dr-Itachi, 2022-11-22:
When setting openocd without specifying the path (default to the WSL side) nothing happens. It seems that cortex dont find it but when using instead the windows version of openocd i get this output in the terminal:

Waiting for gdb server to start...[2022-11-22T22:26:20.930Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
mnt/c/openocd/bin/openocd.exe -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/voldemort/.STM32Codes/TimerSTM32 -f /home/voldemort/.vscode-server/extensions/marus25.cortex-debug-1.6.7/support/openocd-helpers.tcl -f board/st_nucleo_f7.cfg
[2022-11-22T22:26:22.400Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...

haneefdm, 2022-11-22:
> Error: arm-none-eabi-nm failed! statics/global/functions may not be properly classified: Error: spawn arm-none-eabi-nm ENOENT
This will not stop debugging but why is `nm` missing? It is there in every arm compiler release.

About openocd, we launch whatever we are told to launch via your settings or your PATH. Launching the windows version on Linux may even work with WSL. I would not know and it would be wrong to do that because those port numbers are for the Linux side. Run that same command on the command line and see what happens. Anything we do, you can do it on the command line to see why it is not working. Please the issue of how people got it working and post over there. Maybe, someone will help you. I can't as I don't know much about WSL. We do not support it but with the right settings and USB support, people have made it work.

https://github.com/Marus/cortex-debug/issues/467

Dr-Itachi, 2022-11-22:
> About openocd, we launch whatever we are told to launch via your settings or your PATH. Launching the windows version on Linux may even work with WSL. I would not know and it would be wrong to do [...]

using the windows version of openocd i get this output from the terminal: it is in the debugging state

voldemort@DESKTOP-FEB5RA9:~/.STM32Codes/TimerSTM32$ openocd.exe -f board/st_nucleo_f7.cfg
Open On-Chip Debugger 0.12.0-rc2 (2022-10-26-14:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Info : STLINK V2J40M27 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.243073
Info : [stm32f7x.cpu] Cortex-M7 r0p1 processor detected
Warn : [stm32f7x.cpu] Silicon bug: single stepping may enter pending exception handler!
Info : [stm32f7x.cpu] target has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f7x.cpu on 3333
Info : Listening on port 3333 for gdb connections

Dr-Itachi, 2022-11-22:
When i manually flash with the same config using a Vs Code task it also works. here are my tasks

 "version": "2.0.0",
    "tasks": [
        {
            "label": "Hermann STM32 : Build",
            "type": "shell",
            "command": "scan-build",
            "args": [
                "compiledb",
                "make",
                "-j4"
            ],
            "problemMatcher": [],
            "group": {
                "kind": "build",
                "isDefault": true
            }
        },
        {
            "label": "Hermann STM32 : Clean",
            "type": "shell",
            "command": "make",
            "args": [
                "clean"
            ]
        },
        {
            "label": "Hermann STM32 : Flash",
            "dependsOn":["Hermann STM32 : Build"],
            "type": "shell",
            "command": "/mnt/c/openocd/bin/openocd.exe",
            "args": [
                "-f",
                "board/st_nucleo_f7.cfg",
                "-f",
                "interface/timer.cfg"
            ]
        }
    ]
}

Dr-Itachi, 2022-11-22:
[image]

Dr-Itachi, 2022-11-22:
I got ramdoms errors

[image]

haneefdm, 2022-11-22:
You have so many other issues. Your gdb is not even installed properly/fully.

If you are trying to duplicate our command lines, you have to it exactly and not leave out arguments. And, from the same current working directory. I really don't think using Windows version of openoc on Linux is wise. Also, you have a typo (missing '/') in your openocd path.

I can't help any further. WSL is not our expertise. I keep asking you to post in that thread where experts are available but you are not doing that.

Dr-Itachi, 2022-11-22:
> You have so many other issues. Your gdb is not even installed properly/fully.
>
> If you are trying to duplicate our command lines, you have to it exactly and not leave out arguments. And, from the [...]

Oh thanks for the typo error. I think I found the issue. Let me download the latest version of gcc available and try something I think can fix the issue

Dr-Itachi, 2022-11-22:
> You have so many other issues. Your gdb is not even installed properly/fully.
>
> If you are trying to duplicate our command lines, you have to it exactly and not leave out arguments. And, from the [...]

The issue was with paths. i didn't set the bin directory of arm-none-eabi in my path variable. and gdb also needed libcurses5. I fixed both of them and the debugging launch but stop after few secondes i dont know why

-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-file-exec-and-symbols "/home/voldemort/.STM32Codes/TimerSTM32/build/TimerSTM32.elf"
-> 8^done
9-target-select extended-remote localhost:50000
-> 9^error,msg="localhost:50000: Connection timed out."
10-interpreter-exec console "monitor reset halt"
Failed to launch GDB: localhost:50000: Connection timed out. (from target-select extended-remote localhost:50000)

Dr-Itachi, 2022-11-22:
what is needed to set hostname:port for external connection please ? i dont know what to put on hostname and port

haneefdm, 2022-11-22:
https://github.com/microsoft/WSL/discussions/5861#discussion-17216
https://github.com/Marus/cortex-debug/issues/467#issuecomment-1046162717

Dr-Itachi, 2022-11-22:
> [microsoft/WSL#5861 (comment)](https://github.com/microsoft/WSL/discussions/5861#discussion-17216)
> [#467 (comment)](https://github.com/Marus/cortex-debug/issues/467#issuecomment-1046162717)

I use powershell to print the hostname. I tried with myPChostname.shome.net:3333 and 50000 but none worked. openocd is waiting for connexion but cortex debug doesn’t connect to it. I have nothing in the debug console from cortex.

haneefdm, 2022-11-22:
If you are using `external` you control the host name and the port-number. Openocd tells you which port(s) it used and what port number it is waiting on. gdb connects to whatever you tell it.

For us, we use 50000 because we told openocd to use 50000. I have no idea what you told openocd to do.

You can also launch gdb from the command-line and try to connect ... or use telnet or netcat

Dr-Itachi, 2022-11-22:
> You have so many other issues. Your gdb is not even installed properly/fully.
>
> If you are trying to duplicate our command lines, you have to it exactly and not leave out arguments. And, from the [...]

Ok so i create a Vs code task which will run openocd and set hostname and port in cortex debug to connect with but it seems to not connect to 
[image]
.it i dont know why

Dr-Itachi, 2022-11-22:
Here is another log. i set gdb_port to 50000 in openocd but it didn't wort either

'''7-interpreter-exec console "source /home/voldemort/.vscode-server/extensions/marus25.cortex-debug-1.6.7/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-file-exec-and-symbols "/home/voldemort/.STM32Codes/TimerSTM32/build/TimerSTM32.elf"
-> 8^done
9-target-select extended-remote DESKTOP-FEB5RA9.mshome.net:50000
-> 9^error,msg="DESKTOP-FEB5RA9.mshome.net:50000: Connection timed out."
10-interpreter-exec console "monitor halt"
Failed to launch GDB: DESKTOP-FEB5RA9.mshome.net:50000: Connection timed out. (from target-select extended-remote DESKTOP-FEB5RA9.mshome.net:50000)'''

tbnilsen, 2025-02-26:
Did you ever get this to work? I seem to be having the exact same problem With Ubuntu 24.10 and Cortex debugging.