Marus/cortex-debug #777: What does hw-rwpt mean? My program only returned hw-awpt.

jf168-stack, 2022-11-24:
[image]

haneefdm, 2022-11-24:
None of this was documented. We figured out but trial and error. If you are seeing hw-rwpt we can add that. The gdb I used when I was doing this only did those two.

a is for access (r/w), r is for read.

Can I see the entire line returned by gdb?

jf168-stack, 2022-11-28:
When I add a data breakpoint and select Break on Value Change, it returns hw-rwpt.

haneefdm, 2022-11-28:
Okay, I added hw-rwpt and lets see what happens. In the next release in a few days.

PhilippHaefele, 2023-02-02:
Was released with V1.6.9

@jf168-stack Can you please check if it is now working as expected?