Marus/cortex-debug #779: Any chance of adding probe-rs?

mubes, 2022-11-27:
...it seems to have very high quality target support because it leverages the PACKS (See https://github.com/probe-rs). Would be nice to have it available as a first-class-citizen.
(Keep up the good work!)

DAVE

PhilippHaefele, 2022-11-27:
Just had a quick look into the repos and found the following one: https://github.com/probe-rs/vscode

Haven't had a deeper look into the extension, but have you tried that? If so, what is missing over there, that can be found in Cortex-Debug?

haneefdm, 2022-11-27:
I don't have a single embedded rust executable to work with.

The problem I have is I need a complete project (on a Mac preferably) that runs using just command-line tools and a mkae/CMake. Many people think this is free but for me, it will take several days since I have zero Rust knowledge.

Since I don't have many boards, the project needs to target a board I have STM32F429ZI-NUCLEO144. I don't mind installing a few pieces of SW (I failed miserably every time I tried to install cargo and tools needed to compile an embedded example on Linux). Things (in Rust proper) would not compile and searching the internet did not help. I gave up.

haneefdm, 2022-11-27:
The https://github.com/probe-rs seems to have the entire debug adapter (DAP implementation). That is what we have. So, not sure what else you would need. Our frontend support the probe-rs does not make sense. The frontend/backend are a matched set.

Our [memory views](https://github.com/mcu-debug/memview), [RTOS views](https://github.com/mcu-debug/rtos-views) could potentially also work with that. We just have to add '[probe-rs-debug](https://github.com/probe-rs/vscode)' as a supported debugger.

Soon, the SVD/Peripheral viewer will also be factored out into its own extension.

closed 2022-12-04 by haneefdm