Marus/cortex-debug #780: Quit debugging without reason

hongshui3000, 2022-11-29:
How to reproduce:
Open the disassembly interface, keep sliding the mouse wheel up
[image]

haneefdm, 2022-11-29:
> Quit debugging without reason

What does that mean? VSCode quit? Debug session Quit. What was the last thing in Debug Console?

hongshui3000, 2022-11-30:
> > Quit debugging without reason
> 
> What does that mean? VSCode quit? Debug session Quit. What was the last thing in Debug Console?

Debug session Quit
[image]

haneefdm, 2022-11-30:
Nope, I can't tell what crashed. Any dialog boxes? After this happens, is your gdb or gdb-server still running when you look at your process/task manager? How do the debug tool bar look after this happens?

We scroll up all the time so, I don't get it. No way for me to duplicate this. Looks like you haven't enabled debug mode where we can see gdb interactions. Set it in "raw" mode

hongshui3000, 2022-12-01:
[LOG.txt](https://github.com/Marus/cortex-debug/files/10132295/LOG.txt)
There didn't seem to be any errors, but the debug session exited by itself anyway. This time, exit the debugging session by opening the disassembly window

haneefdm, 2022-12-01:
Check this file `%TMPDIR%/cortex-debug-server-exiting.log` and see what the last few lines say.

TMPDIR is your windows tmp dir and should have an env. var that tells you what that dir is.

haneefdm, 2022-12-01:
Also, try setting debug to "vscode" instead of "raw". You will see a lot more output in the console but only the last few matter.

hongshui3000, 2022-12-02:
> Also, try setting debug to "vscode" instead of "raw". You will see a lot more output in the console but only the last few matter.

[image]

haneefdm, 2022-12-02:
>Check this file %TMPDIR%/cortex-debug-server-exiting.log and see what the last few lines say.

> TMPDIR is your windows tmp dir and should have an env. var that tells you what that dir is.

I really need this info. gdb either quit or crashed.

hongshui3000, 2022-12-02:
> cortex-debug-server-exiting.log

[2022-12-02T06:54:54.177Z] ppid=20684 pid=20684 OCEAN JLink Launch: ******* Starting new session request type="launch"
[2022-12-02T06:54:54.256Z] ppid=20684 pid=5844 GDB started ppid=20684 pid=5844
[2022-12-02T06:55:01.767Z] ppid=20684 pid=5844 GDB: exited
[2022-12-02T06:55:01.768Z] ppid=20684 pid=5844 OCEAN JLink Launch: quitEvent: Killing server
[2022-12-02T06:55:01.769Z] ppid=20684 pid=5844 GDBServer(21640): forcing an exit with kill()
[2022-12-02T06:55:01.781Z] ppid=20684 pid=5844 OCEAN JLink Launch: quitEvent: sending VSCode TerminatedEvent
[2022-12-02T06:55:01.787Z] ppid=20684 pid=5844 GDBServer(21640): exited code=null signal=SIGTERM
[2022-12-02T06:55:01.828Z] ppid=20684 pid=5844 OCEAN JLink Launch: Begin disconnectRequest

hongshui3000, 2022-12-02:
https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.15.0
I use this gdb and I feel it is a bug of gdb.
armgcc's gdb to work fine.

haneefdm, 2022-12-02:
Yup, gdb crashed. I don't think it exited cleanly.

closed 2022-12-02 by haneefdm