Marus/cortex-debug #782: Breakpoints not working "No line xxx in file", no source view also

424778940z, 2022-11-30:
As title suggests
Not able to  get the breakpoint to work, also no source view, maybe they are related?
Looked through wiki and other examples, am I missing something?

The project is configured with cmake
```cmake
# MISC

# you should not need to set this when using VS Code
# set(CMAKE_BUILD_TYPE debug)
# set(CMAKE_BUILD_TYPE release)
# set(CMAKE_BUILD_TYPE sdram_debug)
# set(CMAKE_BUILD_TYPE sdram_release)
# set(CMAKE_BUILD_TYPE flexspi_nor_debug)
# set(CMAKE_BUILD_TYPE flexspi_nor_release)

set(ENV{ARMGCC_DIR} "/usr")
set(CMAKE_TOOLCHAIN_FILE "${CMAKE_CURRENT_SOURCE_DIR}/hal/nxp/mcux-sdk/tools/cmake_toolchain_files/armgcc.cmake")

# CHIP / BOARD / PATH SETTINGS
SET(BspDirPath "${CMAKE_CURRENT_SOURCE_DIR}/bsp/nxp/MIMXRT1170-EVK_1176")
SET(SdkDirPath "${CMAKE_CURRENT_SOURCE_DIR}/hal/nxp/mcux-sdk")
SET(SdkSdmmc_DirPath "${CMAKE_CURRENT_SOURCE_DIR}/hal/nxp/mcux-sdk-middleware-sdmmc")
SET(SdkUsb_DirPath "${CMAKE_CURRENT_SOURCE_DIR}/hal/nxp/mcux-sdk-middleware-usb")
SET(SrcDirPath "${CMAKE_CURRENT_SOURCE_DIR}/src")
SET(EXECUTABLE_OUTPUT_PATH ${CMAKE_CURRENT_BINARY_DIR}/${CMAKE_BUILD_TYPE})
SET(LIBRARY_OUTPUT_PATH ${CMAKE_CURRENT_BINARY_DIR}/${CMAKE_BUILD_TYPE})

# CROSS COMPILER SETTING
SET(CMAKE_SYSTEM_NAME Generic)
CMAKE_MINIMUM_REQUIRED(VERSION 3.10.0)

# THE VERSION NUMBER
SET(Tutorial_VERSION_MAJOR 1)
SET(Tutorial_VERSION_MINOR 0)

# ENABLE ASM
ENABLE_LANGUAGE(ASM)

SET(CMAKE_STATIC_LIBRARY_PREFIX)
SET(CMAKE_STATIC_LIBRARY_SUFFIX)

SET(CMAKE_EXECUTABLE_LIBRARY_PREFIX)
SET(CMAKE_EXECUTABLE_LIBRARY_SUFFIX)

project(bootloader)

set(MCUX_SDK_PROJECT_NAME bootloader.elf)

include(${BspDirPath}/flags.cmake)
include(${BspDirPath}/config.cmake)

set(FILE_LIST_SRC
    "${SrcDirPath}/main.c"
    "${SrcDirPath}/main.h"
    "${SrcDirPath}/hid_mouse.c"
    "${SrcDirPath}/hid_mouse.h"
    "${SrcDirPath}/hid_keyboard.c"
    "${SrcDirPath}/hid_keyboard.h"
    "${SrcDirPath}/usb_device_descriptor.c"
    "${SrcDirPath}/usb_device_descriptor.h"
    "${SrcDirPath}/usb_device_config.h"
    "${SrcDirPath}/usb_device_ch9.c"
    "${SrcDirPath}/usb_device_ch9.h"
    "${SrcDirPath}/usb_device_class.c"
    "${SrcDirPath}/usb_device_class.h"
    "${SrcDirPath}/usb_device_hid.c"
    "${SrcDirPath}/usb_device_hid.h"
)

[... 202 lines cut ...]
-> 17^done,register-values=[{number="0",value="0"},{number="1",value="0"},{number="2",value="0"},{number="3",value="0"},{number="4",value="0"},{number="5",value="0"},{number="6",value="0"},{number="7",value="0"},{number="8",value="0"},{number="9",value="0"},{number="10",value="0"},{number="11",value="0"},{number="12",value="0"},{number="13",value="0x2001ff20"},{number="14",value="0xffffffff"},{number="15",value="0x4f8 <Reset_Handler>"},{number="25",value="16777216"},{number="58",value="0"},{number="59",value="0"},{number="60",value="0"},{number="61",value="0"},{number="62",value="0"},{number="63",value="0"},{number="64",value="0"},{number="65",value="-nan(0xfffff00000000)"},{number="66",value="0"},{number="67",value="0"},{number="68",value="0"},{number="69",value="0"},{number="70",value="0"},{number="71",value="0"},{number="72",value="0"},{number="73",value="-nan(0xfffff00000000)"},{number="90",value="0"},{number="91",value="0x2001ff20"},{number="92",value="0x0 <__isr_vector>"},{number="93",value="0"},{number="94",value="0"},{number="95",value="0"},{number="96",value="0"},{number="97",value="0"},{number="98",value="0"},{number="99",value="0"},{number="100",value="0"},{number="101",value="0"},{number="102",value="0"},{number="103",value="0"},{number="104",value="0"},{number="105",value="0"},{number="106",value="0"},{number="107",value="0"},{number="108",value="0"},{number="109",value="0"},{number="110",value="0"},{number="111",value="0"},{number="112",value="-nan(0x7fffff)"},{number="113",value="0"},{number="114",value="0"},{number="115",value="0"},{number="116",value="0"},{number="117",value="0"},{number="118",value="0"},{number="119",value="0"},{number="120",value="0"},{number="121",value="0"},{number="122",value="0"},{number="123",value="0"},{number="124",value="0"},{number="125",value="0"},{number="126",value="0"},{number="127",value="0"},{number="128",value="-nan(0x7fffff)"}]
18-exec-continue --all
-> 18^running
-> *running,thread-id="all"
mi2.status = running
19-exec-interrupt
-> 19^done
-> ~"\nProgram"

Program
-> ~" received signal SIGINT, Interrupt.\n"
 received signal SIGINT, Interrupt.
-> ~"0x00223104 in ?? ()\n"
0x00223104 in ?? ()
-> *stopped,reason="signal-received",signal-name="SIGINT",signal-meaning="Interrupt",frame={addr="0x00223104",func="??",args=[],arch="armv7e-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
20-break-delete
-> 20^done
21-target-disconnect
-> =thread-exited,id="1",group-id="i1"
-> =thread-group-exited,id="i1"
-> 21^done
-gdb-exit
-> ^exit
```

haneefdm, 2022-11-30:
That message is from `gdb`, not from us. could mean
* code optimized out
* no debug info in executable (code not compiled/linked for debug)

I can't tell from your CMake file, what finally gets done when **compiling/linking** your code. You will need `-g` and an appropriate optimization level such as `-O0` or `Og`

https://gcc.gnu.org/onlinedocs/gcc/Debugging-Options.html

424778940z, 2022-11-30:
I checked those flags already, seems fine
they ware in the flags.cmake, here they are

```cmake
SET(CMAKE_ASM_FLAGS_DEBUG " \
    ${CMAKE_ASM_FLAGS_DEBUG} \
    -DDEBUG \
    -D__STARTUP_CLEAR_BSS \
    -D__STARTUP_INITIALIZE_NONCACHEDATA \
    -mcpu=cortex-m7 \
    -mfloat-abi=hard \
    -mfpu=fpv5-d16 \
    -mthumb \
")
SET(CMAKE_C_FLAGS_DEBUG " \
    ${CMAKE_C_FLAGS_DEBUG} \
    -DDATA_SECTION_IS_CACHEABLE=0 \
    -D_DEBUG=1 \
    -DDEBUG \
    -DCPU_MIMXRT1176DVMAA_cm7 \
    -DUSB_STACK_BM \
    -DFSL_OSA_BM_TASK_ENABLE=0 \
    -DFSL_OSA_BM_TIMER_CONFIG=0 \
    -DSDK_DEBUGCONSOLE=1 \
    -DMCUXPRESSO_SDK \
    -g \
    -O0 \
    -mcpu=cortex-m7 \
    -Wall \
    -mfloat-abi=hard \
    -mfpu=fpv5-d16 \
    -mthumb \
    -MMD \
    -MP \
    -fno-common \
    -ffunction-sections \
    -fdata-sections \
    -ffreestanding \
    -fno-builtin \
    -mapcs \
    -std=gnu99 \
")
SET(CMAKE_CXX_FLAGS_DEBUG " \
    ${CMAKE_CXX_FLAGS_DEBUG} \
    -DDEBUG \
    -DCPU_MIMXRT1176DVMAA_cm7 \
    -DMCUXPRESSO_SDK \
    -g \
    -O0 \
    -mcpu=cortex-m7 \
    -Wall \
    -mfloat-abi=hard \
    -mfpu=fpv5-d16 \
    -mthumb \
    -MMD \
    -MP \
    -fno-common \
    -ffunction-sections \
    -fdata-sections \
    -ffreestanding \
    -fno-builtin \
    -mapcs \
    -fno-rtti \
    -fno-exceptions \
")
SET(CMAKE_EXE_LINKER_FLAGS_DEBUG " \
    ${CMAKE_EXE_LINKER_FLAGS_DEBUG} \
    -g \
    -mcpu=cortex-m7 \
    -Wall \
    -mfloat-abi=hard \
    -mfpu=fpv5-d16 \
    --specs=nano.specs \
    --specs=nosys.specs \
    -fno-common \
    -ffunction-sections \
    -fdata-sections \
    -ffreestanding \
    -fno-builtin \
    -mthumb \
    -mapcs \
    -Xlinker \
    --gc-sections \
    -Xlinker \
    -static \
    -Xlinker \
    -z \
    -Xlinker \
    muldefs \
    -Xlinker \
    -Map=output.map \
    -Wl,--print-memory-usage \
    -Xlinker \
    --defsym=__stack_size__=0x2000 \
    -Xlinker \
    --defsym=__heap_size__=0x400 \
    -T${BspDirPath}/linker_script/MIMXRT1176xxxxx_cm7_ram.ld -static \
")

424778940z, 2022-11-30:
and here is the linker script `MIMXRT1176xxxxx_cm7_ram.ld`

```
/*
** ###################################################################
**     Processors:          MIMXRT1176AVM8A_cm7
**                          MIMXRT1176CVM8A_cm7
**                          MIMXRT1176DVMAA_cm7
**
**     Compiler:            GNU C Compiler
**     Reference manual:    IMXRT1170RM, Rev 1, 02/2021
**     Version:             rev. 1.1, 2022-04-02
**     Build:               b220402
**
**     Abstract:
**         Linker file for the GNU C Compiler
**
**     Copyright 2016 Freescale Semiconductor, Inc.
**     Copyright 2016-2022 NXP
**     All rights reserved.
**
**     SPDX-License-Identifier: BSD-3-Clause
**
**     http:                 www.nxp.com
**     mail:                 support@nxp.com
**
** ###################################################################
*/

/* Entry Point */
ENTRY(Reset_Handler)

HEAP_SIZE  = DEFINED(__heap_size__)  ? __heap_size__  : 0x400;
STACK_SIZE = DEFINED(__stack_size__) ? __stack_size__ : 0x2000;

/* Specify the memory areas */
MEMORY
{
  m_interrupts          (RX)  : ORIGIN = 0x00000000, LENGTH = 0x00000400
  m_text                (RX)  : ORIGIN = 0x00000400, LENGTH = 0x0003FC00
  m_data                (RW)  : ORIGIN = 0x20000000, LENGTH = 0x00040000
  m_data2               (RW)  : ORIGIN = 0x202C0000, LENGTH = 0x00080000
}

/* Define output sections */
SECTIONS
{
  __NCACHE_REGION_START = ORIGIN(m_data2);
  __NCACHE_REGION_SIZE  = 0;

  /* The startup code goes first into internal RAM */
  .interrupts :
  {
    __VECTOR_TABLE = .;
    __Vectors = .;
    . = ALIGN(4);
    KEEP(*(.isr_vector))     /* Startup code */
    . = ALIGN(4);
  } > m_interrupts

  /* The program code and other data goes into internal RAM */
  .text :
  {
    . = ALIGN(4);
    *(.text)                 /* .text sections (code) */
    *(.text*)                /* .text* sections (code) */
    *(.rodata)               /* .rodata sections (constants, strings, etc.) */
    *(.rodata*)              /* .rodata* sections (constants, strings, etc.) */
    *(CodeQuickAccess)       /* quick access code section */
    *(.glue_7)               /* glue arm to thumb code */
[... 135 lines cut ...]
    . = ALIGN(8);
    __end__ = .;
    PROVIDE(end = .);
    __HeapBase = .;
    . += HEAP_SIZE;
    __HeapLimit = .;
    __heap_limit = .; /* Add for _sbrk */
  } > m_data

  .stack :
  {
    . = ALIGN(8);
    . += STACK_SIZE;
  } > m_data

  /* Initializes stack on the end of block */
  __StackTop   = ORIGIN(m_data) + LENGTH(m_data);
  __StackLimit = __StackTop - STACK_SIZE;
  PROVIDE(__stack = __StackTop);

  .ARM.attributes 0 : { *(.ARM.attributes) }

  ASSERT(__StackLimit >= __HeapLimit, "region m_data overflowed with stack and heap")
}
```

haneefdm, 2022-11-30:
Your `MIMXRT1176xxxxx_cm7_ram.ld` is not relevant. To me, those make variables is not proof that they were the ones used. You have to see how gcc was **exactly** invoked. Or you can examine the output yourself like we do...

```
arm-none-eabi-nm --defined-only -S -l -C -p /home/adam/Desktop/bootloader/.build/debug/bootloader.elf
```
You should see all your symbols and files involved. You can also disassemble the executable and see if the file/lines exist.

You can also run gdb on the command line and set that breakpoint. We don't do any magic and neither can gdb if such a file/line does not exist in the executable.

Unlikely, If you compile for production, sometimes people don't use debug options and your make file could have also stripped the executable of debug info. You have to see what your make file actually did.

haneefdm, 2022-11-30:
You can use your gdb (arm-none-eabi-gdb) from command line manually. You can see every command we use in the debug console -- we use machine interface but you don't have to. This is between your gdb and your executable. We do not interfere in any way. There is nothing to configure for us or gdb AFAIK. Perhaps there is no code at/near that line or it got optimized out? I have no idea.

Did you disassemble to see what happened to that file/line? You can use gdb or objdump to disassemble? Did you try to set a breakpoint manually in gdb?

https://gist.github.com/jarun/ea47cc31f1b482d5586138472139d090
https://stackoverflow.com/questions/2511018/how-does-objdump-manage-to-display-source-code-with-the-s-option

I don't see this as a problem with this extension yet.

424778940z, 2022-12-01:
I tried, and those infos are there...

those "Non-debugging symbols" are weak function pointers

```
adam@adam-virtual-machine ~ % gdb-multiarch -q '/home/adam/Desktop/bootloader/.build/flexspi_nor_debug/bootloader.elf'
Reading symbols from /home/adam/Desktop/bootloader/.build/flexspi_nor_debug/bootloader.elf...
(gdb) info line main
Line 350 of "/home/adam/Desktop/bootloader/src/main.c" starts at address 0x300048d0 <main> and ends at 0x300048d4 <main+4>.
(gdb) info functions
All defined functions:

File ../../../../../../../../newlib/libc/misc/fini.c:
26:	void __libc_fini_array(void);

File ../../../../../../../../newlib/libc/misc/init.c:
30:	void __libc_init_array(void);

File ../../../../../../../../newlib/libc/stdlib/__atexit.c:
68:	int __register_exitproc(int, void (*)(void), void *, void *);

File ../../../../../../../../newlib/libc/stdlib/__call_atexit.c:
70:	void __call_exitprocs(int, void *);

File ../../../../../../../libgloss/libnosys/_exit.c:
9:	void _exit(int);

File ../../../../../../libgcc/libgcc2.c:
1015:	UDItype __udivmoddi4(UDItype, UDItype, UDItype *);

File /build/newlib-pB30de/newlib-3.3.0/newlib/libc/include/malloc.h:
136:	void __malloc_lock(struct _reent *);
50:	void _free_r(struct _reent *, void *);
42:	void *_malloc_r(struct _reent *, malloc_size_t);
45:	void free(void *);
37:	void *malloc(size_t);

File /build/newlib-pB30de/newlib-3.3.0/newlib/libc/include/reent.h:
153:	void *_sbrk_r(struct _reent *, ptrdiff_t);

File /build/newlib-pB30de/newlib-3.3.0/newlib/libc/include/stdlib.h:
76:	int atexit(void (*)(void));
93:	void exit(int);

File /build/newlib-pB30de/newlib-3.3.0/newlib/libc/include/string.h:
33:	void *memset(void *, int, size_t);

File /build/newlib-pB30de/newlib-3.3.0/newlib/libc/include/sys/errno.h:
15:	int *__errno(void);

File /home/adam/Desktop/bootloader/bsp/nxp/MIMXRT1170-EVK_1176/board.c:
262:	void BOARD_ConfigMPU(void);
24:	uint32_t BOARD_DebugConsoleSrcFreq(void);
36:	void BOARD_InitDebugConsole(void);

File /home/adam/Desktop/bootloader/bsp/nxp/MIMXRT1170-EVK_1176/clock_config.c:
275:	void BOARD_BootClockRUN(void);

File /home/adam/Desktop/bootloader/bsp/nxp/MIMXRT1170-EVK_1176/pin_mux.c:
34:	void BOARD_InitBootPins(void);
212:	void BOARD_InitPins_DEBUG(void);
--Type <RET> for more, q to quit, c to continue without paging--
132:	void BOARD_InitPins_LCD(void);
59:	void BOARD_InitPins_UART(void);
103:	void BOARD_InitPins_USB(void);

File /home/adam/Desktop/bootloader/hal/nxp/mcux-sdk-middleware-usb/device/usb_device_dci.c:
895:	usb_status_t USB_DeviceDeinit(usb_device_handle);
1111:	usb_status_t USB_DeviceDeinitEndpoint(usb_device_handle, uint8_t);
1217:	usb_status_t USB_DeviceGetStatus(usb_device_handle, usb_device_status_t, void *);
[... 584 lines cut ...]
0x3000cec0  __func__.0
0x3000d170  __func__.1
0x3000d184  __func__.0
0x3000d1f0  __func__.0
0x3000d238  __func__.4
0x3000d248  __func__.2
0x3000d260  __func__.1
0x3000d2bc  __func__.21
0x3000d2d0  __func__.16
0x3000d2dc  __func__.15
0x3000d2f4  __func__.11
0x3000d30c  __func__.10
0x3000d598  __func__.16
0x3000d5ac  __func__.14
0x3000d5bc  __func__.12
0x3000d5d0  __func__.11
0x3000d5e4  __func__.5
0x3000d5f8  __func__.4
0x3000d60c  __func__.3
0x3000d61c  __func__.2
0x3000d64c  __EH_FRAME_BEGIN__
0x3000d64c  _init
0x3000d658  _fini

```

haneefdm, 2022-12-01:
We cannot help if gdb is not cooperating. Please don't post such large listings... my emails are getting flooded.

closed 2022-12-01 by haneefdm