Marus/cortex-debug #783: Is there a way to set environment variable when invoking gdb?

424778940z, 2022-12-01:
I'm trying other version of the arm tool chain while troubleshooting https://github.com/Marus/cortex-debug/issues/782
The official ARM toolchain gdb requires python3.8, and it seems only look at a fixed location unless env PYTHONHOME is set
I checked document, but cannot find a way to set the env when cortex-debug invoking gdb

```
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = (not set)
  program name = '/usr/local/bld-tools/bld-tools-virtual-env/bin/python'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = '/usr/local/bld-tools/bld-tools-virtual-env/bin/python'
  sys.base_prefix = '/usr'
  sys.base_exec_prefix = '/usr'
  sys.executable = '/usr/local/bld-tools/bld-tools-virtual-env/bin/python'
  sys.prefix = '/usr'
  sys.exec_prefix = '/usr'
  sys.path = [
    '/usr/lib/python38.zip',
    '/usr/lib/python3.8',
    '/usr/lib/lib-dynload',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding

```

haneefdm, 2022-12-01:
Since env. vars are global, they will be automatically inherited by all programs. That is how all OSes/Shells work.

You can also wrap/replace the gdb command into a script to do what you need and/or specify the exact gdbPath. There are many ways of doing what you need.

Even better use a gdb that does not require Python.

closed 2022-12-01 by haneefdm

424778940z, 2022-12-01:
> Since env. vars are global, they will be automatically inherited by all programs.

well the PYTHONHOME is not global, if it is then that will defeat the purpose of having a different version of python

dimsanius, 2023-07-18:
Hello,

I might be slightly late, but I have recently figured out the way of sorting out Python 3.8 dependency issue. Detailed description available here https://github.com/Marus/cortex-debug/issues/907#issuecomment-1639752057