Marus/cortex-debug #784: Disassembly in ARM big-endian is wrong

nonarkitten, 2022-12-04:
This little bit of C code
```c
void InitDMTimer(void) {
    CM_WKUP_TIMER1_CLKCTRL->BIT.MODULEMODE = 2;
    while(CM_WKUP_TIMER1_CLKCTRL->BIT.IDLEST);
```

Compiles to this
```
402f4af8 <InitDMTimer>:
402f4af8:	e30034c4 	movw	r3, #1220	; 0x4c4
402f4afc:	e34434e0 	movt	r3, #17632	; 0x44e0
402f4b00:	e5932000 	ldr	r2, [r3]
402f4b04:	e3c22403 	bic	r2, r2, #50331648	; 0x3000000
402f4b08:	e3822402 	orr	r2, r2, #33554432	; 0x2000000
402f4b0c:	e5832000 	str	r2, [r3]
402f4b10:	e320f000 	nop	{0}
402f4b14:	e30034c4 	movw	r3, #1220	; 0x4c4
402f4b18:	e34434e0 	movt	r3, #17632	; 0x44e0
402f4b1c:	e5933000 	ldr	r3, [r3]
402f4b20:	e6bf3f33 	rev	r3, r3
402f4b24:	e1a03823 	lsr	r3, r3, #16
402f4b28:	e2033003 	and	r3, r3, #3
402f4b2c:	e6ef3073 	uxtb	r3, r3
402f4b30:	e3530000 	cmp	r3, #0
402f4b34:	1afffff6 	bne	402f4b14 <InitDMTimer+0x1c>
``` 

But during debug shows up like this:
[image]

haneefdm, 2022-12-04:
I don't think that is us. But, I have never seen a big-endian ARM before. Shouldn't matter though. We use GDB to do the disassembly. I am not sure you are using the right GDB. We are not clever enough to do our own disassembly. We rely on GDB.

In the Debug Console prompt, can you do `disassemble InitDMTimer` and see what you get?

nonarkitten, 2022-12-04:
disassemble InitDMTimer
44-interpreter-exec console "disassemble InitDMTimer"
{"output":"","token":44,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
=

nonarkitten, 2022-12-04:
Technically every ARM is bi-endian. The code is compiled be8 which means instructions are always little endian, but data can be either. I think it confuses GDB which is astonishing since this bi-endian mode of ARM has been around for a very, very long time. I mean, it runs (sort of), but JLINK doesn't seem to like it either and stability is rubbish.

haneefdm, 2022-12-04:
> Technically every ARM is bi-endian

I know that. Just haven't seen a big endian implementation, so have zero test cases.

So, there was no disassembly from gdb?

closed 2022-12-04 by haneefdm