Marus/cortex-debug #785: JLink debugging not working

JoshuaYonathan, 2022-12-04:
Cortex-Debug v-1.6.7
ARM GNU Toolchain 12.2.MPACBTI-Bet1
JLink v-7.82d

The JLink server appears and says "Waiting for GDB connection...". After ~5 seconds it times out. Debug console and terminal logs attached. The JLink server and GDB works when I run it manually in the terminal but not using the same arguments cortex-debug uses.
[debug_console.txt](https://github.com/Marus/cortex-debug/files/10147864/debug_console.txt)
[terminal.txt](https://github.com/Marus/cortex-debug/files/10147865/terminal.txt)

JoshuaYonathan, 2022-12-04:
Using the ARM GNU Toolchain v-11.3 with Cortex-Debug v-1.4.4 will work correctly. Using the latest version of cortex-debug with this version of the arm gnu toolchain will not work with the same issue I reported above. Using v-12.2 of the arm toolchain will not work with cortex-debug v-1.4.4, it hangs forever when I press the "play" button.

haneefdm, 2022-12-04:
Wrong JLink gdb server. Try JLinkGDBServerCL.exe

You are using the graphical version that is not meant to be used from command-line

haneefdm, 2022-12-04:
You should see a lot more information the terminal window....where JLink is running.

JoshuaYonathan, 2022-12-04:
> 

This fixed it for me, thanks.

closed 2022-12-04 by JoshuaYonathan