Marus/cortex-debug #790: Debugging STM32H745 dual core - almost there

jagauthier, 2022-12-15:
I'm trying to move away from STM32CubeIDE for several reasons. Slow, bloat, workspace management are a few of the issues I have with this.  So, I've started experimenting with VSCode, Cortex-debug, and the whole platform for development with STM32.  First, I started with a simple project and it worked great.

Then I moved to an empty project that I generated with STM32CubeMX for a dual core STM32H745.  One of the initial challenges I met was that CubeMX doesn't support Makefile generation for this MCPU. Fine, I exported the project to STM32CubeIDE and I worked in a Makefile for awhile to develop .elf files for both cores.  It needs more work, but I am able to build both.

I then used STM32CubeIDE and OpenOCD to figure out how ST is debugging with this platform and I stole their configs and implemented them into my launch files.

The way that the code is generated is there are 2 folders created for each core. A CM4 and a CM7 folder. Each contains source code specific to the core and at the top level is the HAL library and other shared files (LD, SVD, etc)

Now, here's where I am:  Debugging the Cortex-M4 works flawlessly,   I can launch the debugger, and it runs to main(). I am able to step through the code, set breakpoints, etc.

But when I do the same with the Cortex-M7 - openocd runs and goes into debug mode.  VSCode seems to go into debug mode but there is no ability to interact with the code.  It doesn't seem to know where it is, and I can't step through the code or visually see anything happening.

I'm not sure what my next steps to try and solve this are.

haneefdm, 2022-12-15:
Perhaps our Wiki could help you. Lots of topics. Multi-core is complicated but you can make it work and you have to understand how OpenOCD, etc. works.

https://github.com/Marus/cortex-debug/wiki
https://github.com/Marus/cortex-debug/wiki/Multi-core-debugging

In the second link scroll down to the bottom to see an example for ST although it uses STLink which is quite a bit different from OpenOCD (separate vs shared servers). The snippet at the top is for OpenOCD

jagauthier, 2022-12-15:
Thanks! I appreciate the pointer. I went through this and toyed with it a little. It's quite a different approach from using openOCD.  However, through some more trials I discovered my problem.  Long story short, it was a build issue and the binary was not being built with the correct symbols.  Once I figured this out I can debug both processors individually.   I have not tried to debug the at the same time yet - that is an exercise for a different day.

closed 2022-12-15 by jagauthier