Marus/cortex-debug #792: Multiple ST-Link devices

jagauthier, 2022-12-19:
I've been reading and searching but I have not figured out the best way to proceed.
First issue I have run into is that I cannot seem to specify which adapter to use in launch.json.
According to: https://github.com/Marus/cortex-debug/wiki/OpenOCD-Specific-Configuration there is no way to add additional command line arguments like "-c adapter serial XXXXX"

However, long term I have multiple projects open using different programmers.  Since these are Makefile-based projects I'd  like to be able to specify the adapter **somewhere** and have that carry through to the Makefile build AND the launch.json.

Any thoughts on how to achieve these tasks?

haneefdm, 2022-12-19:
How about `serverArgs`. `serialNumber` is for JLink only. Different gdb servers handle this differently.  Check their docs You can also send gdb `monitor` commands via `preLaunchCommands` 

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

If you are using openocd, you see section 8.1

https://openocd.org/doc/html/Debug-Adapter-Configuration.html

closed 2022-12-19 by haneefdm

jagauthier, 2022-12-19:
I guess I am looking for something (maybe a cortex-debug, or vscode thing) where I can set the serial number, and then reference that via some variable or argument to "make" and inside launch.json.

haneefdm, 2022-12-19:
I provided all the info I had. Our tool is heavily customizable and some of the probe servers are different and they have their own methods. You have to study all of them to see what makes sense.

jagauthier, 2022-12-19:
That's still hard coding the serial number into a config file.
I'm talking about something higher level. So that when I open a project and attach it to a new debugger I don't have to go and edit a Makefile (to flash) AND a launch file (to debug).

Maybe it's not something people don't do commonly.  But I've been known to have 4 debuggers at once running 4 different pieces of hardware.  

Thanks anyway. I'll try and figure something out!

haneefdm, 2022-12-19:
You can always have your own script act as a gdb-server, do whatever you want, massage arguments, etc. and finally launch (exec) the real gdb-server.