Marus/cortex-debug #793: STM32WLE5 debugging broke from v1.6.7 to v1.6.9

zfields, 2022-12-23:
This `launch.json` configuration has worked for several of the previous versions, and just broke with v1.6.9:

```json
{
    "name": "Cortex Debug",
    "type": "cortex-debug",
    "executable": "${workspaceRoot}/build/sparrow.elf",
    "cwd": "${workspaceRoot}",
    "request": "launch",
    "device": "STM32WLE5JC",
    "servertype": "stlink"
}
```

It gives the following error message:

> GDB executable "undefined-gdb" was not found.
> Please configure "cortex-debug.armToolchainPath" or "cortex-
> debug.gdbPath" correctly.

If I add the following entry...

```json
"gdbPath": "/opt/st/stm32cubeide_1.10.1/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.linux64_1.0.0.202111181127/tools/bin/arm-none-eabi-gdb"
```

Then I get this error message:

> GDB Server Console tcp port is undefined

If I downgrade to v1.6.7, the original unaltered `launch.json` works as expected.

zfields, 2022-12-23:
Upon further investigation, I noticed v1.6.9 claims dependencies on the following extensions:

- mcu-debug.debug-tracker-vscode
- mcu-debug.memory-view
- mcu-debug.rtos-views

It appears they are the source of the problem, because they are not loading correctly.

[image]

Once, they are loaded, debugging proceeds as expected, without requiring modification of the `launch.json` :thinking:.

These new extensions weren't required by v1.6.7, so I'm curious if they are absolutely necessary? If they are only adding features, then I would prefer to select the extensions myself, as opposed to having them installed on my behalf.

haneefdm, 2022-12-23:
They were part of this extension but had to be split up because it was getting too big and other debuggers can now use those extensions.

Out of the 25,000+ installs since last week, this is the instance of this kind of failure.

zfields, 2022-12-23:
I'm probably one of the first people using it in a `devcontainer`.

haneefdm, 2022-12-23:
Then this would be bug in VSCode. Will see if anyone else reports it.

zfields, 2022-12-23:
Are you certain it's not a bug in the way you are declaring your dependency tree?

This seems like it would happen for other extensions than just yours, but maybe not? :shrug:

haneefdm, 2022-12-23:
This is supposed to work. It is as simple as it gets...just 4-lines of JSON

https://github.com/Marus/cortex-debug/blob/4c381ef057fabe83b74b41eb870365891eef51db/package.json#L3111

Can you reproduce it?

I can flatten out the dependencies by adding more, but that would be ridiculous.

haneefdm, 2022-12-23:
It is a big deal if you can reproduce it and we can try a few things.

zfields, 2022-12-23:
Yes, it's easily reproducible.

I will help you try whatever you like.

haneefdm, 2022-12-23:
Can you install this release. You have to do this manually from the `vsix` file. The dependencies are generally taken care of at install (I think) and run time as well. VSCode tries to guess where to install the extension and perhaps it is guessing wrong

https://github.com/Marus/cortex-debug/releases/tag/v1.6.10-pre1

zfields, 2022-12-29:
@haneefdm Thank you for being so responsive! I'm sorry I was not able to work as quickly, I had a work project that needed finished before the holidays.

It turns out, a container does not simply activate the latest installed version from the host, but only downloads the latest published version.

However, while I was working on it, I noticed the follow status of the installed extensions.

[image]

Although it claims it "cannot activate" the **MemoryView** and **RTOS Views** extensions, they appear to be activated, and the **debug-tracker-vscode** extension appears to be deactivated. :shrug: 

I'll keep messing with it and report back here.

zfields, 2022-12-29:
When I install the .vsix in the container, I get the following error...

[image]

Perhaps there is something strange (cycle) with the dependencies surrounding **debug-tracker-vscode**?

haneefdm, 2022-12-29:
I have other users using containers and they are not seeing this problem. We definitely do not have a circular dependency issue

>> It turns out, a container does not simply activate the latest installed version from the host, but only downloads the latest published version.

Hmmm, we have no control over that. VSCode decides what goes on the host and what needs duplicating on the client.

Could you use this info to pre-install extensions?

https://code.visualstudio.com/docs/devcontainers/create-dev-container#_create-a-devcontainerjson-file

The extension seems to be installed but not activated. But, no one needs it. Just to get a debug session started, that should be okay.

<img width="593" alt="image" src="https://user-images.githubusercontent.com/41269583/209961680-c5114c04-b4a1-4146-8903-64a04f9e044e.png

You can have an extension running (activated) and still have the 'reload required'. Can you see the running extensions to see if any version of all the extensions are running. Maybe that dialog is lying to us.

haneefdm, 2022-12-29:
Could you try the following.

1. Start VSCode
2. Run the following command
[image]

3. List running extensions and see if the tracker and memory view are running? And, they are running in the container. While not ideal, the tracker could be running on the host which is a valid configuration so long as VSCode handles the remote procedure calls properly. Again, this is a decision VSCode makes.

haneefdm, 2022-12-29:
Okay, I made one change to the tracker extension (version 0.0.13). Give it an hour for the VSCode marketplace to propagate the new version. It now requests that it needs to be run where the workspace is and hopefully, that will make VSCode do the right thing.\

It is not the right thing for all clients but I want to see if it works better. You may have to uninstall all four extensions, restart VSCode and then only install cortex-debug. All the others should fall in line automatically.

zfields, 2022-12-29:
That worked, no more warnings/errors! 

[image]

Thank you for sticking with me on this. I really appreciate you working through the problem!

haneefdm, 2022-12-29:
May I ask what ended up working? :-)

zfields, 2022-12-29:
I waited for 20 mins.

I cleared the Docker daemon. Then I let VSCode rebuild it's wrapper container (VSCode context for remote containers), which performs a fresh install of CortexDebug, which subsequently pulled in **debug-tracker-vscode v0.0.13**, and now I don't get any warnings or errors.

haneefdm, 2022-12-29:
Okay, thanks. This is a bug in VSCode but not sure how to report it to them.

Gonna close this issue for now...thanks for helping.

closed 2022-12-29 by haneefdm

jnz86, 2023-11-10:
@haneefdm 

I'm getting the same thing he was. A year later, on debug-tracker-vscode 0.0.15 as well as Peripheral Viewer 1.4.6. Both say RELOAD REQUIRED after building. 

[image]

Linux Mint 21.2, ubuntu devcontainer, here is the relevant portion of my VSCode devcontrainer:

```
  "customizations": {
    "vscode": {
      "extensions": [
        "ms-vscode.cpptools-extension-pack",
        "mcu-debug.debug-tracker-vscode",
        "marus25.cortex-debug",
	"mcu-debug.memory-view",
	"mcu-debug.rtos-views",
	"mcu-debug.peripheral-viewer",        
        "fpopescu.vscode-unity-test-adapter",
        "ms-vscode.test-adapter-converter",
        "xaver.clang-format",
        "ms-vscode.vscode-serial-monitor",
      ]
    }
  },
```

I tried to remove as much as possible, but maybe that wasn't super reliable because of caching.

Any ideas on what I can try?

haneefdm, 2023-11-10:
Nope, not sure what is going on, I did the same thing you have. Perhaps you should rebuild the container to get all the latest. Maybe in a container, any extension updates get lost when you restart the container --- which VSCode auto updates and then asks to reload VSCode. Let us know what works.

jnz86, 2023-11-11:
I've rebuilt about 40 times today. I also flushed all cache. 

I don't have any version of the extensions listed, so my .devcon file should describe pulling the latest in. There should be nothing else to update to. I have a feeling they're being loaded, but maybe in the wrong order, or there is an issue that prevents proper execution at first. 

**This is the Installing Extensions part of my docker build:** I don't see any specific issues although I don't really know much about what I'm looking at.

```
[4579 ms] [01:28:40] Installing extensions...
[4580 ms] [01:28:40] Extension host agent started.
[4597 ms] Port forwarding 58840 > 39387 > 39387 stderr: Connection established
[4601 ms] [01:28:40] Started initializing default profile extensions in extensions installation folder. file:///root/.vscode-server/extensions
[4610 ms] [01:28:40] ComputeTargetPlatform: linux-x64
[4610 ms] [01:28:40] [127.0.0.1][787fa122][ManagementConnection] New connection established.
[4614 ms] Port forwarding connection from 58858 > 39387 > 39387 in the container.
[4614 ms] Start: Run in container: /root/.vscode-server/bin/1a5daa3a0231a0fbba4f14db7ec463cf99d7768e/node -e 
[4618 ms] [01:28:40] Log level changed to info
[4644 ms] [01:28:40] Completed initializing default profile extensions in extensions installation folder. file:///root/.vscode-server/extensions
[4676 ms] Port forwarding 58858 > 39387 > 39387 stderr: Connection established
[4743 ms] [01:28:40] [127.0.0.1][09e6b860][ExtensionHostConnection] New connection established.
[4759 ms] [01:28:40] [127.0.0.1][09e6b860][ExtensionHostConnection] <298> Launched Extension Host Process.
[5547 ms] [01:28:41] Installing extension 'ms-vscode.cpptools-extension-pack'...
[5548 ms] [01:28:41] Getting Manifest... ms-vscode.cpptools-extension-pack
[5600 ms] [01:28:41] Installing extension 'mcu-debug.rtos-views'...
[5600 ms] [01:28:41] Getting Manifest... mcu-debug.rtos-views
[5631 ms] [01:28:41] Installing extension 'ms-vscode.vscode-serial-monitor'...
[01:28:41] Getting Manifest... ms-vscode.vscode-serial-monitor
[5647 ms] [01:28:41] Installing extension 'ms-vscode.test-adapter-converter'...
[5648 ms] [01:28:41] Getting Manifest... ms-vscode.test-adapter-converter
[5655 ms] [01:28:41] Installing extension 'mcu-debug.memory-view'...
[01:28:41] Getting Manifest... mcu-debug.memory-view
[5662 ms] [01:28:41] Installing extension: ms-vscode.cpptools-extension-pack
[5669 ms] [01:28:41] Installing extension 'mcu-debug.peripheral-viewer'...
[01:28:41] Getting Manifest... mcu-debug.peripheral-viewer
[5689 ms] [01:28:41] Installing extension 'mcu-debug.debug-tracker-vscode'...
[01:28:41] Getting Manifest... mcu-debug.debug-tracker-vscode
[5709 ms] [01:28:41] Installing extension: mcu-debug.rtos-views
[5724 ms] [01:28:41] Installing extension 'shoaibshakeel.easyzoom-enhanced'...
[01:28:41] Getting Manifest... shoaibshakeel.easyzoom-enhanced
[5728 ms] [01:28:41] Installing extension 'fpopescu.vscode-unity-test-adapter'...
[5728 ms] [01:28:41] Getting Manifest... fpopescu.vscode-unity-test-adapter
[5734 ms] [01:28:41] Installing extension 'xaver.clang-format'...
[01:28:41] Getting Manifest... xaver.clang-format
[5762 ms] [01:28:41] Installing extension: mcu-debug.memory-view
[5795 ms] [01:28:41] Installing extension 'wraith13.zoombar-vscode'...
[5795 ms] [01:28:41] Getting Manifest... wraith13.zoombar-vscode
[5801 ms] [01:28:41] Installing extension: mcu-debug.debug-tracker-vscode
[5802 ms] [01:28:41] Installing extension: mcu-debug.peripheral-viewer
[5835 ms] [01:28:41] Installing extension: fpopescu.vscode-unity-test-adapter
[5848 ms] [01:28:41] Installing extension: shoaibshakeel.easyzoom-enhanced
[5862 ms] [01:28:41] Installing extension: ms-vscode.test-adapter-converter
[5867 ms] [01:28:41] Installing extension 'marus25.cortex-debug'...
[5867 ms] [01:28:41] Getting Manifest... marus25.cortex-debug
[5869 ms] [01:28:41] Installing extension: ms-vscode.vscode-serial-monitor
[5884 ms] [01:28:41] Getting Manifest... mcu-debug.debug-tracker-vscode
[5933 ms] [01:28:41] Installing extension: xaver.clang-format
[5945 ms] [01:28:41] Installing extension: wraith13.zoombar-vscode
[6080 ms] [01:28:41] Getting Manifest... mcu-debug.debug-tracker-vscode
[6146 ms] [01:28:41] Waiting for already requested installing extension mcu-debug.debug-tracker-vscode mcu-debug.rtos-views
[6186 ms] [01:28:41] Extracted extension to file:///root/.vscode-server/extensions/mcu-debug.debug-tracker-vscode-0.0.15: mcu-debug.debug-tracker-vscode
[6195 ms] [01:28:41] Renamed to /root/.vscode-server/extensions/mcu-debug.debug-tracker-vscode-0.0.15
[6196 ms] [01:28:41] Extracting extension completed. mcu-debug.debug-tracker-vscode
[6202 ms] [01:28:41] Extracted extension to file:///root/.vscode-server/extensions/mcu-debug.peripheral-viewer-1.4.6: mcu-debug.peripheral-viewer
[6205 ms] [01:28:41] Extension installed successfully: mcu-debug.debug-tracker-vscode
[6206 ms] [01:28:41] Finished waiting for already requested installing extension mcu-debug.debug-tracker-vscode mcu-debug.rtos-views
[6206 ms] [01:28:41] Extension 'mcu-debug.debug-tracker-vscode' v0.0.15 was successfully installed.
[6209 ms] [01:28:41] Renamed to /root/.vscode-server/extensions/mcu-debug.peripheral-viewer-1.4.6
[6210 ms] [01:28:41] Extracting extension completed. mcu-debug.peripheral-viewer
[6217 ms] [01:28:41] Extension installed successfully: mcu-debug.peripheral-viewer
[... cut ...]
[14984 ms] [01:28:50] Extension 'ms-vscode.cpptools-extension-pack' v1.3.0 was successfully installed.
[16210 ms] Port forwarding connection from 40988 > 39387 > 39387 in the container.
[16210 ms] Start: Run in container: /root/.vscode-server/bin/1a5daa3a0231a0fbba4f14db7ec463cf99d7768e/node -e 
[16276 ms] Port forwarding 40988 > 39387 > 39387 stderr: Connection established
[21280 ms] Port forwarding 40988 > 39387 > 39387 stderr: Remote close
[21284 ms] Port forwarding 40988 > 39387 > 39387 terminated with code 0 and signal null.
```

jnz86, 2023-11-11:
I haven't "SOLVED" it yet, but I'm close.

So, it very well may be something with VS Code, but potentially a workaround.

I have the debug-tracker-vscode extension installed in local AND my remote. This isn't OK it seems. If I remove the local, everything is fine, add it back in, and everytime the container loads it asks me to Reload Window on entry. 

The workaround is that if in `settings.json`, the VSCode/Profile one, not the workspace one, and use:
```
    "remote.extensionKind": {
        "mcu-debug.debug-tracker-vscode": ["workspace"]
    }
```

Then it seems fine so far. Otherwise, the extension seems to be marked as run GLOBALLY which at least in my case causes an issue. So I don't know how extensions mark themselves, but it seems there is global, workspace and ui.

Is there a case outside of containers it should not be marked as workspace? Because again, I haven't added the others in yet, but is seems like this may fix it.

Along those lines... I need to re-test, but I was not able to put the force-where-an-extension-runs lines in the workspace settings.json which is unfortunate since that is the one that will travel with the repo. Best I can tell that one is only looked at after the container is built. Where you need these settings in VSCode itself, before the container is created.

haneefdm, 2023-11-18:
Extensions mark themselves like the following in preference order

```JSON
    "extensionKind": [
        "workspace",
        "ui"
    ],
```

https://github.com/mcu-debug/debug-tracker-vscode/blob/7a9a06a737d2537dd1be33399af174ba5b247311/package.json#L22

We have no control beyond that as to what VSCode actually does

I think for a container it appears you have to force everything to the `workspace`. For WSL, that may not be true