Marus/cortex-debug #794: Confused, trying to use remote host with cortex-debug

duaneellissd, 2022-12-24:
Is there an example I can follow/reuse that shows how to configure a REMOTE OpenOCD session? Specifically I am getting this error message:

ERROR: At least one OpenOD Configuration File must be specified

I think - the tool believes I am running OpenOCD locally. I am not. And I think that is the problem - I think I need to tell the tool:  OpenOCD is already running on some other machine and GDB can reach/contact it via the HOSTNAME/PORT.

More detail:
     VS Code is running on a Linux server in a data center at work in the office.   
     OpenOCD is running on my laptop at home talking to my board.
     GDB (on linux) communicates to openocd on my laptop via a socket.

The above works - out side of VSCODE - I commonly use Emacs GUD mode, it works quite nicely.  It also works nicely if I launch Eclipse on the linux server and point Eclipse debugger at openocd on my laptop at home.

Based on what I see in:
https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

I setup:
   executable = My ELF file
   gdbPath = Path to my version of GDB
   servertype = openocd
   gdbTarget = HOSTNAME:PORT where GDB is running.
   configFiles = Empty list - Not Applicable, OpenOCD is remote.
   searchDir - Not applicable, OpenOCD is remote.
   request = attach
   server* = all non-existant - Not running the server locally.

  Why I need toolChainPrefix _ i do not know I am debugging right?
  type: cortex-debug

Question: What else should I specify? or how do I tell the tool "Don't worry about openocd" - it is already running and is waiting for a connection.

duaneellissd, 2022-12-24:
The configurations I have is this:
```
  { 
    "cwd": "${workspaceRoot}",
            "executable": "Build/debug.d/app_riscv_test.debug.elf",
            "name": "debug with OpenOCD",
            "gdbPath" : "/tools/Microchip/SoftConsole-v2021.1/riscv-unknown-elf-gcc/bin/riscv64-unknown-elf-gdb",
            "request" : "attach",
            "toolchainPrefix" : "/tools/Microchip/SoftConsole-v2021.1/riscv-unknown-elf-gcc/bin/riscv64-unknown-elf-",
            "type": "cortex-debug",
            "gdbTarget": "dellis.laptop.local:3333",
            "servertype": "openocd",
            "configFiles": [],
            "searchDir": [],
            "runToEntryPoint": "main",
            "showDevDebugOutput": "none",
        },

```

haneefdm, 2022-12-25:
If you are starting openocd externally (yourself), then `"servertype": "external"` should be used.

toolchainPrefix is not a path to the files. It is the prefix for every file in the tool chain. For you, it would be "riscv64-unknown-elf". These options are better set in the User/Workspace settings btw instead of each config.

It helps to look at the global settings in VSCode for Crotex-Debug. The settings that would interest you are the path to the bin dir of the toolchain executables `cortex-debug.armToolchainPath` and `cortex-debug.armToolchainPrefix`. While it says `arm...`, it is now a generic term for gnu tool chains regardless of the processor

[image]

> Why I need toolChainPrefix _ i do not know I am debugging right?

We need 3 to 4 tools for full functionality that may go beyond what you are used to in Eclipse. We need gdb, nm, objdump (you see these printed in the Debug Console) and we get bits of debug info from all these tools. This may change in the future. That is our design, so sorry if it doesn't meet your expectations.

You can also help with documentation. Our [Wiki](https://github.com/Marus/cortex-debug/wiki) is here for you to edit. I updated the Wiki just now to help explain it. I did not realize that people do not go through the VSCode settings for an extension before using it.

haneefdm, 2022-12-25:
Since you are using an external server, you are responsible for COMPLETELY setting up any pre/post launch or totally overriding the setting to make it work for you. Our defaults may not work. You are responsible for debugging your setup as well as we can't help. Hope you understand there are millions of ways people do this and there is no way to predict. But we provide customizations.

I am closing this now as I don't see a bug. This was more appropriate for StackOverflow. But feel free to add to docs or comment back. I am a single person supporting this effort. There are tons of users using `external` gdb-servers, btw.

Btw, you are welcome to use (and contribute) for RISC-V with our extension but please don't complain and try to contribute. You may see issues with 64 bit here and there. We are a 32 bit extension and may eventually support 64-bit archs

closed 2022-12-25 by haneefdm