Marus/cortex-debug #795: Seemingly random breakpoint hit without setting one

kistlin, 2022-12-26:
While trying to debug with WSL2/Docker, similar things discussed as in #467.

The setup looks like this (either (1) or (2))
[image]

**Versions**

Windows
- SEGGER J-Link GDB Server V7.84 Command Line Version

Ubuntu 22.04
- arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)
- gdb-multiarch GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1

Docker (Ubuntu 20.04)
- arm-none-eabi-gcc (Arm GNU Toolchain 11.3.Rel1) 11.3.1 20220712
- [GNU gdb (Arm GNU Toolchain 11.3.Rel1) 12.1.90.20220802-git](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads)

Extension
- marus25.cortex-debug-1.6.9

It seems to show some similarities to #310.

Strangely it works when using the Windows version of the toolchain (gdb) `Arm GNU Toolchain 11.3.Rel1`. Loading the same binary (`elf`). Not with any setup/toolchain mentioned above.

Potential problem in the debug output
```
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"target","content":"Expected an decimal digit (0-9)\r\n"}]}
Expected an decimal digit (0-9)
-> 12^done
```
On the command line I get the same issue. What works is
```
target remote <IP>:<PORT>
file /workspaces/nuttx-apps/nuttx/nuttx
break main
monitor reset 0
continue
```
(Using `monitor reset 0` or `monitor halt` separate)
After that it's possible, once the breakpoint is hit, to print local variables with `info locals`.

Full sample log output
```
Cortex-Debug: VSCode debugger extension version 1.6.9 git(4c381ef). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from /opt/arm/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-objdump --syms -C -h -w /workspaces/nuttx-apps/nuttx/nuttx
Reading symbols from /opt/arm/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-nm --defined-only -S -l -C -p /workspaces/nuttx-apps/nuttx/nuttx
Launching GDB: /opt/arm/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
Finished reading symbols from objdump: Time: 184 ms
-> =thread-group-added,id="i1"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-added","output":[["id","i1"]]}]}
-> ~"GNU gdb (Arm GNU Toolchain 11.3.Rel1) 12.1.90.20220802-git\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"GNU gdb (Arm GNU Toolchain 11.3.Rel1) 12.1.90.20220802-git\n"}]}
-> ~"Copyright (C) 2022 Free Software Foundation, Inc.\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Copyright (C) 2022 Free Software Foundation, Inc.\n"}]}
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."}]}
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"\nType \"show copying\" and \"show warranty\" for details.\n"}]}
-> ~"This GDB was configured as \"--host=x86_64-pc-linux-gnu --target=arm-none-eabi\".\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"This GDB was configured as \"--host=x86_64-pc-linux-gnu --target=arm-none-eabi\".\n"}]}
-> ~"Type \"show configuration\" for configuration details.\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Type \"show configuration\" for configuration details.\n"}]}
-> ~"For bug reporting instructions, please see:\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"For bug reporting instructions, please see:\n"}]}
-> ~"<https://bugs.linaro.org/>.\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"<https://bugs.linaro.org/>.\n"}]}
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."}]}
[... 171 lines cut ...]
hover -var-create: unable to create variable object (from var-create --thread 1 --frame 0 hover_684faaad9b3b5f8d5215b73f4de55b02b284bdaff69afde9283bf03a1ac16277 * "(uint32_t) &ch0.reglist")36-stack-select-frame --thread 1 0
-> 36^done
GDB -> App: {"output":"","token":36,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
37-stack-list-variables --thread 1 --frame 0 --simple-values
-> 37^done,variables=[{name="s",arg="1",type="const char *",value="0x15c <error: Cannot access memory at address 0x15c>"},{name="maxlen",arg="1",type="size_t",value="4294967295"},{name="sc",type="const char *",value="0x15c <error: Cannot access memory at address 0x15c>"}]
GDB -> App: {"output":"","token":37,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["variables",[[["name","s"],["arg","1"],["type","const char *"],["value","0x15c <error: Cannot access memory at address 0x15c>"]],[["name","maxlen"],["arg","1"],["type","size_t"],["value","4294967295"]],[["name","sc"],["type","const char *"],["value","0x15c <error: Cannot access memory at address 0x15c>"]]]]]}}
38-var-update --thread 1 --frame 0 --all-values var_s_256
-> 38^error,msg="Variable object not found"
GDB -> App: {"output":"","token":38,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Variable object not found"]]}}
39-var-create --thread 1 --frame 0 var_s_256 * "s"
-> 39^done,name="var_s_256",numchild="1",value="0x15c <error: Cannot access memory at address 0x15c>",type="const char *",thread-id="1",has_more="0"
GDB -> App: {"output":"","token":39,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["name","var_s_256"],["numchild","1"],["value","0x15c <error: Cannot access memory at address 0x15c>"],["type","const char *"],["thread-id","1"],["has_more","0"]]}}
40-var-update --thread 1 --frame 0 --all-values var_maxlen_256
-> 40^error,msg="Variable object not found"
GDB -> App: {"output":"","token":40,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Variable object not found"]]}}
41-var-create --thread 1 --frame 0 var_maxlen_256 * "maxlen"
-> 41^done,name="var_maxlen_256",numchild="0",value="4294967295",type="size_t",thread-id="1",has_more="0"
GDB -> App: {"output":"","token":41,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["name","var_maxlen_256"],["numchild","0"],["value","4294967295"],["type","size_t"],["thread-id","1"],["has_more","0"]]}}
42-var-update --thread 1 --frame 0 --all-values var_sc_256
-> 42^error,msg="Variable object not found"
GDB -> App: {"output":"","token":42,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Variable object not found"]]}}
43-var-create --thread 1 --frame 0 var_sc_256 * "sc"
-> 43^done,name="var_sc_256",numchild="1",value="0x15c <error: Cannot access memory at address 0x15c>",type="const char *",thread-id="1",has_more="0"
GDB -> App: {"output":"","token":43,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["name","var_sc_256"],["numchild","1"],["value","0x15c <error: Cannot access memory at address 0x15c>"],["type","const char *"],["thread-id","1"],["has_more","0"]]}}
```

haneefdm, 2022-12-26:
Thanks for such detailed reporting. You must be using an `"external"` type server. This is why you are seeing the following command being sent to GDB
```
12-interpreter-exec console "monitor reset halt"
```
That won't work for JLink and if you are server type was 'JLink'' then we would not have sent that command. When using an external server, you have to customize everything. The defaults will only work for some servers (like openocd, pyocd, etc.) There is no universal agreement between servers on how to reset or halt the device. Every one of them seems to want different commands.

If this is the case, you have multiple options. Can you confirm?

haneefdm, 2022-12-26:
You can use the following in your launch.json
```JSON
    "overrideLaunchCommands": [
        "monitor halt",
        "monitor reset",    // This line may be optional, but safer
        "-target-download",
        "monitor reset"     // You can use an optional reset-type but default is 0
    ],
```

If you run locally with server type set to "jlink" you would have seen those same set of commands

You will have to do something similar for other `overrideXXXCommands` for restart and if you need for reset (reset button defaults to restart commands). If not, the reset/restart buttons will have a similar problem.

kistlin, 2022-12-26:
Yes, I use `"servertype": "external"`.
And adding `overrideLaunchCommands` to `launch.json` worked.

Thanks a lot @haneefdm!

closed 2022-12-26 by kistlin