Marus/cortex-debug #796: error in debug

dayu26, 2022-12-28:
### **errors:**
openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /Users/yuda/rust/st/cortexm-app -f /Users/yuda/.vscode/extensions/marus25.cortex-debug-1.6.9/support/openocd-helpers.tcl -f /opt/homebrew/share/openocd/scripts/interface/stlink.cfg -f /opt/homebrew/share/openocd/scripts/target/stm32f3x.cfg
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK V2J37M26 (API v2) VID:PID 0483:374B
Info : Target voltage: 2.880000
Info : stm32f3x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f3x.cpu on 50000
Info : Listening on port 50000 for gdb connections
Info : accepting 'gdb' connection on tcp/50000
Info : device id = 0x10036422
Info : flash size = 256kbytes
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000400 msp: 0x20010000
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000400 msp: 0x20010000
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000400 msp: 0x20010000
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000400 msp: 0x20010000
  stm32f3x.cpu tpiu
    stm32f3x.cpu tpiu config (disable | ((external | internal (<filename> | <:port> | -)) (sync <port
              width> | ((manchester | uart) <formatter enable>))
              <TRACECLKIN freq> [<trace freq>]))
tpiu
  tpiu config (disable | ((external | internal (<filename> | <:port> | -)) (sync <port
            width> | ((manchester | uart) <formatter enable>)) <TRACECLKIN
            freq> [<trace freq>]))
Error: invalid subcommand "init"
  stm32f3x.cpu tpiu
    stm32f3x.cpu tpiu config (disable | ((external | internal (<filename> | <:port> | -)) (sync <port
              width> | ((manchester | uart) <formatter enable>))
              <TRACECLKIN freq> [<trace freq>]))
tpiu
  tpiu config (disable | ((external | internal (<filename> | <:port> | -)) (sync <port
            width> | ((manchester | uart) <formatter enable>)) <TRACECLKIN
            freq> [<trace freq>]))
Error: invalid subcommand "names"
/Users/yuda/.vscode/extensions/marus25.cortex-debug-1.6.9/support/openocd-helpers.tcl:15: Error: 
in procedure 'CDSWOConfigure' 
at file "/Users/yuda/.vscode/extensions/marus25.cortex-debug-1.6.9/support/openocd-helpers.tcl", line 15
Info : dropped 'gdb' connection
shutdown command invoked

### **Environment:**
I got error when I use vscode debug embedded rust  on M1 Mac os 
STM32F303

haneefdm, 2022-12-28:
You did not attach your launch.json.
Whose openocd are you using? I recommend using the one that ST ships with STM32CubeIDE, just in case.

See our wiki page below. ST may not have enabled SWO (requires TPIU, DW, ITM, etc.)

https://github.com/Marus/cortex-debug/wiki/SWO-Output#swo-configuration

You can copy the "/Users/yuda/.vscode/extensions/marus25.cortex-debug-1.6.9/support/openocd-helpers.tcl" and debug the issue yourself.

Let us know what worked.

dayu26, 2022-12-30:
my launch.json :
{
    /* 
     * Requires the Rust Language Server (rust-analyzer) and Cortex-Debug extensions
     * https://marketplace.visualstudio.com/items?itemName=rust-lang.rust-analyzer
     * https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug
     */
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug (QEMU)",
            "servertype": "qemu",
            "cwd": "${workspaceRoot}",
            "preLaunchTask": "Cargo Build (debug)",
            "runToEntryPoint": "main",
            "executable": "./target/thumbv7m-none-eabi/debug/cortexm-app",
            /* Run `cargo build --example hello` and uncomment this line to run semi-hosting example */
            //"executable": "./target/thumbv7m-none-eabi/debug/examples/hello",
            "cpu": "cortex-m3",
            "machine": "lm3s6965evb",
        },
        {
            /* Configuration for the STM32F303 Discovery board */
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug (OpenOCD)",
            "servertype": "openocd",
            "cwd": "${workspaceRoot}",
            "preLaunchTask": "Cargo Build (debug)",
            "runToEntryPoint": "main",
            "executable": "./target/thumbv7em-none-eabihf/debug/cortexm-app",
            /* Run `cargo build --example itm` and uncomment this line to run itm example */
            // "executable": "./target/thumbv7em-none-eabihf/debug/examples/itm",
            "device": "STM32F303VCT6",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32f303x.cfg"
            ],
            "svdFile": "${workspaceRoot}/.vscode/STM32F303.svd",
            "swoConfig": {
                "enabled": true,
                "cpuFrequency": 8000000,
                "swoFrequency": 2000000,
                "source": "probe",
                "decoders": [
                    { "type": "console", "label": "ITM", "port": 0 }
                ]
            }
        }
    ]
}

dayu26, 2022-12-30:
my launch.json

`{
    /* 
     * Requires the Rust Language Server (rust-analyzer) and Cortex-Debug extensions
     * https://marketplace.visualstudio.com/items?itemName=rust-lang.rust-analyzer
     * https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug
     */
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug (QEMU)",
            "servertype": "qemu",
            "cwd": "${workspaceRoot}",
            "preLaunchTask": "Cargo Build (debug)",
            "runToEntryPoint": "main",
            "executable": "./target/thumbv7m-none-eabi/debug/cortexm-app",
            /* Run `cargo build --example hello` and uncomment this line to run semi-hosting example */
            //"executable": "./target/thumbv7m-none-eabi/debug/examples/hello",
            "cpu": "cortex-m3",
            "machine": "lm3s6965evb",
        },
        {
            /* Configuration for the STM32F303 Discovery board */
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug (OpenOCD)",
            "servertype": "openocd",
            "cwd": "${workspaceRoot}",
            "preLaunchTask": "Cargo Build (debug)",
            "runToEntryPoint": "main",
            "executable": "./target/thumbv7em-none-eabihf/debug/cortexm-app",
            /* Run `cargo build --example itm` and uncomment this line to run itm example */
            // "executable": "./target/thumbv7em-none-eabihf/debug/examples/itm",
            "device": "STM32F303VCT6",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32f303x.cfg"
            ],
            "svdFile": "${workspaceRoot}/.vscode/STM32F303.svd",
            "swoConfig": {
                "enabled": true,
                "cpuFrequency": 8000000,
                "swoFrequency": 2000000,
                "source": "probe",
                "decoders": [
                    { "type": "console", "label": "ITM", "port": 0 }
                ]
            }
        }
    ]
}`

dayu26, 2022-12-30:
## the launch.json
{
            /* Configuration for the STM32F303 Discovery board */
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug (OpenOCD)",
            "servertype": "openocd",
            "cwd": "${workspaceRoot}",
            "preLaunchTask": "Cargo Build (debug)",
            "runToEntryPoint": "main",
            "executable": "./target/thumbv7em-none-eabihf/debug/cortexm-app",
            /* Run `cargo build --example itm` and uncomment this line to run itm example */
            // "executable": "./target/thumbv7em-none-eabihf/debug/examples/itm",
            "device": "STM32F303VCT6",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32f303x.cfg"
            ],
            "svdFile": "${workspaceRoot}/.vscode/STM32F303.svd",
            "swoConfig": {
                "enabled": true,
                "cpuFrequency": 8000000,
                "swoFrequency": 2000000,
                "source": "probe",
                "decoders": [
                    { "type": "console", "label": "ITM", "port": 0 }
                ]
            }
        }

## the openocd info
`
% openocd -v
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
`

haneefdm, 2022-12-30:
Like I said, if you use openocd from the ST tools, it does a good job of setting up the right stuff in the config files. You have to debug this yourself by editing a copy of the tcl file included and the function to edit is CDSWOConfigure. I already pointed you to the Wiki. How every device works with SWO is incredibly difficult to tell. We follow the general rules specified by openocd docs.

The failure you are seeing is never supposed to happen if `"target/stm32f303x.cfg"` if the proper config file. Perhaps there is no SWO in this device?

Can you make SWO work in Eclipse using ST tools?

haneefdm, 2022-12-30:
This is a community effort, so if you can figure out what is wrong, let us know. We do not have your board or program to be able to debug anything. If TPIU (which is from ARM) is missing or not configured, the scripts (configFiles) are supposed to do that.

haneefdm, 2022-12-30:
`openocd -v Open On-Chip Debugger 0.11.0` means nothing to me as there are many sub-versions of those. So, if you did an `apt get` or got it via some package manager, then I have no idea what you got.

With the ST distribution, you are likely to have the most up to date config files and ones that work.

https://github.com/STMicroelectronics/OpenOCD

dayu26, 2023-01-03:
Finally I found one of the param in memory.x is different from lib.rs. So it works, Thanks a lot!

closed 2023-01-03 by dayu26