Marus/cortex-debug #797: XRTOS Hangs on uC/OS-II if task is not created with extended information

CIPop, 2023-01-08:
The following code will cause the debugger to hang once a breakpoint is used after this task is created:

```C
err = OSTaskCreate(Consume, (void*)0, &Task3Stk[APP_CFG_TASK_START_STK_SIZE-1], APP_TASK_TEST3_PRIO);
```

Replacing this code with the extended version which populates stack information will work as expected:

```C
  err = OSTaskCreateExt(
    Consume, 
    (void*)0, 
    &Task3Stk[APP_CFG_TASK_START_STK_SIZE-1], 
    APP_TASK_TEST3_PRIO, 
    APP_TASK_TEST3_PRIO, 
    &Task3Stk[0], 
    APP_CFG_TASK_START_STK_SIZE,
    NULL, 
    OS_TASK_OPT_STK_CHK);
```

Getting the following output with the workaround:
[image]

**Expected** When information isn't provided, the debugger should not attempt to retrieve it.
**Actual** The debugger hangs for a few seconds, not accepting any further commands.

I tried disabling OpenOCD RTOS in `launch.json` (`//            "rtos": "auto",`) but it did not have any effect.

CIPop, 2023-01-08:
Immediate workaround if `OSTaskCreateExt` can't be used for any reasons: disable XRTOS by unchecking the option in VSCode. (Removing the OpenOCD RTOS parameter won't have any effect.)

[image]

PhilippHaefele, 2023-01-08:
Thanks for reporting the issue. Please note that the development of the RTOS View has moved to https://github.com/mcu-debug/rtos-views

Seems i didn't have foreseen the case with `OS_TASK_CREATE_EXT_EN` being enabled, but not using it.

Will have a look 👨‍💻

closed 2023-01-08 by haneefdm

CIPop, 2023-01-09:
Thank you @PhilippHaefele ! That was very fast :).

I will make sure to report any new issues in the correct repo next time.

PhilippHaefele, 2023-01-11:
No problem and thanks to @haneefdm a released version 0.0.3 is already available from the marketplace. Have fun 👨‍💻 

> I will make sure to report any new issues in the correct repo next time.

Thanks 😃