Marus/cortex-debug #798: XRTOS with uC/OS-II does not display task names

CIPop, 2023-01-08:
The following code will set the task name:

```C
  OSTaskNameSet(OSPrioCur, (INT8U*)"StartupTask()", &err);
```

The XRTOS viewer will display "???":
[image]

OS Task name can still be obtained exploring the TCB listed by XRTOS:

[image]

PhilippHaefele, 2023-01-08:
Thanks for reporting the issue. Please note that the development of the RTOS View has moved to https://github.com/mcu-debug/rtos-views

I will have a look into this. Would be nice if you can supply your `os_cfg.h` file here, so i can better reproduce the issue.

Thanks

PhilippHaefele, 2023-01-08:
Got it reproduced. Missed one expression when porting over to `mcu-debug/rtos-views`

closed 2023-01-08 by haneefdm

CIPop, 2023-01-09:
Thank you @PhilippHaefele ! That was very fast :).

PhilippHaefele, 2023-01-11:
No problem and thanks to @haneefdm a released version 0.0.3 is already available from the marketplace. Have fun 👨‍💻