Marus/cortex-debug #799: Reload after build

MKHen, 2023-01-10:
Is it possible to improve the debug process by loading a rebuild elf-file with a button to the target and run the debugger?

This would improve my "try and error" programming style :)

haneefdm, 2023-01-10:
That is what the current `Start Debugging` button does. However, loading the elf can be slow due to many factors and you should see to improve that aspect. Could you talk to your device manufacturer and how to speed things up?

A better/faster SWD connection, incremental loading, different gdb server or probe.

closed 2023-01-10 by haneefdm

MKHen, 2023-01-10:
The speed is fast enough for reloading.

The problem is that each time the ***Start Debugging*** button is hit a new instance is created.
Can I avoid this and just recycle the first instance?

haneefdm, 2023-01-10:
90% of the start time is towards programming. If you don't want programming and reloading the elf file, you can use the Reset button

PhilippHaefele, 2023-01-10:
I'm not 100% sure, but last time a colleague used the VSCode reset button (i do not use, auto-rebuild and a reset like that at all) it seemed that the elf was rebuild, but still the old one was loaded into the MCU. Also the rebuild is triggered as expected (auto build on).
As i didn't test it myself i'm not sure 🤔

haneefdm, 2023-01-10:
A rebuild not possible with the `Reset` button. I added that button and VSCode knows nothing about it.

VSCode controls almost everything about the `Restart` button. Hence the `Reset` button.

Of course, they are both configurable to make them do what you need but Restart is mostly controlled by VSCode upfront

PhilippHaefele, 2023-01-11:
Sorry i meant the VSCode `Restart` button in my above comment.
I tested it right now and it only seems that there's a build before run option of the CMake Tools extension, which only triggers on start and not on restart.
So overall i also do not see a solution and i agree with haneefdm that the current solution with stopping and starting the debug session is enough.
@MKHen Maybe you can improve your debug experience by using the shortcuts `F5` & `Shift + F5` (if you not already do this)