Marus/cortex-debug #802: attaching to a running target

abu-bakar-nu, 2023-01-12:
Hello,

OS and version: macOS M2 Ventura 13.1
Target MCU: Apollo 4 KBR

I have this `attach` configuration in my VSCode. 

**What I want:** 
To connect with the target with or without halting the running program.  I want to be able analyze the program state in its current state without reseting it.

**What is happening:**
After the target is attached, the program is stopped and when I press the run button, I do not see any output in the console, which should be printing "Hello World". I have to `reset` the target, and then press the `run` button again to see the output in the console. This starts the program from the beginning which I don't want.

Note that I do see the output in the console when I use the `launch` configuration.

```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
    {
        "type": "cortex-debug",
        "request": "launch",
        "name": "Debug Apollo 4",
        "cwd": "${workspaceRoot}",
        "preLaunchTask": "build-debug",
        "executable": "${workspaceRoot}/bin/hello_world.axf",
        "serverpath": "/Applications/SEGGER/JLink_V784a/JLinkGDBServerCLExe", 
        "servertype": "jlink",
        "device": "AMA4B2KP-KBR",
        "svdFile": "pack/SVD/apollo4p.svd",
        "interface": "swd",
        "runToEntryPoint": "main",
        "swoConfig": {
            "enabled": true,
            "swoFrequency": 1000000,
            "source": "probe",
            "decoders": [
                { 
                    "type": "console", 
                    "label": "Apollo4", 
                    "port": 0, 
                    "showOnStartup": true, 
                    "encoding": "ascii" 
                }
            ]
        }
    },
    {
        "type": "cortex-debug",
        "request": "attach",
        "name": "Attach Apollo 4",
        "cwd": "${workspaceRoot}",
        "executable": "${workspaceRoot}/bin/hello_world.axf",
        "serverpath": "/Applications/SEGGER/JLink_V784a/JLinkGDBServerCLExe", 
        "servertype": "jlink",
        "device": "AMA4B2KP-KBR",
        "svdFile": "/pack/SVD/apollo4p.svd",
        "interface": "swd",
        "swoConfig": {
            "enabled": true,
            "swoFrequency": 1000000,
            "source": "probe",
            "decoders": [
                { 
                    "type": "console", 
                    "label": "Apollo4", 
                    "port": 0, 
                    "showOnStartup": true, 
                    "encoding": "ascii" 
                }
            ]
        }
    }]
}
```

haneefdm, 2023-01-13:
> To connect with the target with or without halting the running program. I want to be able analyze the program state in its current state without reseting it.

That is what `atttach` does but your program is halted at some random location.

> After the target is attached, the program is stopped and when I press the run button, I do not see any output in the console

When you hit the run button, it does a `continue` from the currently halted state..so, if the `"Hello World"` was already executed, not sure where that went.

If you are seeing the words printed after a reset then you could be seeing the issue of lost characters. Make sure you single step over the print statement to see if future output is lost.

abu-bakar-nu, 2023-01-13:
The `"Hello World"` is in a while loop. It should be printed continuously, which is not happening with my `attach` configuration. Not sure if the issue is on the debugger side or on the target side.

haneefdm, 2023-01-13:
JLink has to pick up the bytes and deliver to VSCode over a TCP socket. We simply wait for data to arrive. We do not read the ITM/TPIU ourselves. That is a function of the gdb-server (in your case JLink).

We also do not make a distinction between `launch` and `attach`.

Single step at an instruction level to see where your bytes are going. If they are even being written to the HW FIFO or simply ignored....Since I can't see your code...

On the surface, it appears that this is a JLink problem that is requiring a reset to make things work. It should work without a reset.

haneefdm, 2023-01-13:
Some print routines are written so that if no debugger is attached, prints are ignored. But if this check is done only at the start and not on every char write, it may explain why a reset is required, and it would be considered an issue in the FW.

There are a many ways to implement SWO writes and people roll their own many times.

abu-bakar-nu, 2023-01-13:
what is the purpose for disabling the print routine if the debugger is not attached?

haneefdm, 2023-01-13:
> what could be the purpose of disabling the print routine if the debugger is not attached?

Not just the debugger (note that JLink is considered the debugger here), but it can also block if trace is not enabled.

otherwise the program may block until something (like JLink) drains the FIFO which is generally not very large.

See one sample implementation of `SWO_PrintChar`

https://mcuoneclipse.com/2016/10/17/tutorial-using-single-wire-output-swo-with-arm-cortex-m-and-eclipse/

I have no idea what your implementation looks like. You are asking me to work blind here.

closed 2023-01-13 by haneefdm