Marus/cortex-debug #803: In SWD viewer, the "Refresh" button updates the registers of only one peripheral in which the cursor is located

AndreySmirnov81, 2023-01-17:
[image]

thegecko, 2023-01-30:
@AndreySmirnov81 would you expect this refresh command to refresh all peripheral values or just the expanded ones?

AndreySmirnov81, 2023-01-31:
@thegecko
Only expanded.

AndreySmirnov81, 2023-01-31:
It would be nice if there was a "collapse all" button.

haneefdm, 2023-01-31:
> would you expect this refresh command to refresh all peripheral values

That would totally stall the debugger. We could be talking several megabytes of transfers. We should just refresh whatever is expanded.

AndreySmirnov81, 2023-01-31:
> @thegecko Only expanded.

Similar to how it is done for `Pause`

[image]

haneefdm, 2023-01-31:
> Similar to how it is done for Pause

Yes

thegecko, 2023-01-31:
> Similar to how it is done for Pause

Please see #813. There is a subtle difference between the pause (stopped) state of debug and the "global" refresh in that the refresh would act upon all running sessions rather than the one which paused (stopped).

haneefdm, 2023-01-31:
From a user point of view, it will appear like what happens when you Pause a session.

PR #813 (2023-02-01): Add peripheral tree refresh and collapse commands
As outlined in #803, the tree refresh command only refreshes the highlighted node. This is due to the command passing the selected node to the `peripheralsForceRefresh()` function.

This PR adds a new command for refresh which ignores the selected node. The functionality was also missing to recurse the trees and refresh the expanded nodes, so this has also been added.

As suggested, I've also exposed a `Collapse All` command.

Fixes #803
closed 2023-02-01 by haneefdm