Marus/cortex-debug #807: Undefined type of local variables in VSCode VARIABLES tab

klazklaz77, 2023-01-17:
Hi, everyone, I've found unexpected behavior of the VARIABLES tab in VSCode with local variables. The type of variable isn't correct in the hint when I hover over the some local variable in the VARIABLES tab.

Steps to reproduce: 
1. Launch some C-project;
2. Add some local variable in cycled C-code;
[image]
3. Add line breakpoint after some local variable in cycled code;
5. Continue the debug session;
6. Hower on the variable;
[image]
Actual result:
There is a message with additional information, the type of the variable is correct;
7. Perform continue few times;
8. Hower on variable:
There is a message with additional information, type of the variable is incorrect: undefined type

Expected result:
There is a message with additional information, the type of the variable is correct (volatile int a);

OS: Windows 10;

Versions:
VSCode 1.74.3;
C/C++ 1.13.9;
Cortex-Debug VS Code Extension 1.6.10;

I've attached a detailed video:
[Video_1.zip](https://github.com/Marus/cortex-debug/files/10436792/Video_1.zip)

klazklaz77, 2023-01-17:
Hi, the screen with incorrect variable type in VARIABLES:
[image]

Regards, Vitalii

haneefdm, 2023-01-17:
Would it be possible for you to enable debug in launch.json by setting `"showDevDebugOutput": "raw"` and attach the last few hundred lines (or the whole thing) from the Debug Console when it finally happens. We should be able to see what gdb is returning and why it is undefined. It is unlikely, I will be able to reproduce this easily and will save me a lot of time and hopefully a solution sooner for you.

klazklaz77, 2023-01-17:
Hi, @haneefdm , I've attached the file with debug_console output:
[Debug_console.txt](https://github.com/Marus/cortex-debug/files/10437005/Debug_console.txt)

1. Debug stops on main;
2. Perform the continue;
3. Debug stops on breakpoint;
4. Perform continue one more time;

You can see the video in the issue attachments. I hope It will help You.

Regards, Vitalii.

haneefdm, 2023-01-17:
Thanks Vitalii, I understand what is happening now. The Debug Console output is much more helpful. Vidoe requires me to have exact same setup/program/board/etc.

haneefdm, 2023-01-17:
Should be fixed in the next release. It was how gdb reported things before and how it does now. We should be able to handle both cases.

haneefdm, 2023-01-17:
I made a test release at https://github.com/Marus/cortex-debug/releases/tag/v1.6.11-pre1

Could you download and install the VSIX file (do not double click on it) using VSCode.

I will be removing this release/link once I make a final release to the marketplace

klazklaz77, 2023-01-19:
Hi, @haneefdm ,
[v1.6.11-pre1](https://github.com/Marus/cortex-debug/releases/tag/v1.6.11-pre1)
the issue doesn't reproduce now.

closed 2023-01-19 by haneefdm