Marus/cortex-debug #808: Sometimes does not show value on hover

geoffdudds, 2023-01-17:
Hello,

I occasionally cannot read a variable by hovering over it. Here is a screenshot from the debug console:
[image]

You can see in the screenshot that I was able to read `dampingFactor` and `K`, but the two subsequent reads of `config.Ka` and `config.kb` failed. Note that I can still see the values in the `Watch`.

I can fix this by reloading vscode. Restarting the debug session does not fix it. I have not been able to pin down when it will happen again. Often only 15 mins if I'm using the debugger actively.

Setup info:
- Windows 10
- Debugging cortex m7 via JTrace pro
- vscode v1.7.3
- cortex-debug v1.6.10
- c/c++ v1.14.0

thanks for any help,
Geoff

haneefdm, 2023-01-17:
Maybe VSCode does not know/understand what you are hovering on.  It may be giving us a partial query which won't work. The hover interface is built into cpptools/VSCode itself. We only respond to queries.

Instead of "raw", you can set the showDevDebugOutput to "vscode" or "both" and you will see the queries coming from VSCode to us and what we return.

But, it is strange that reloading fixes it. Each debug session is started from scratch with zero history of previous sessions.

geoffdudds, 2023-01-17:
Interesting.  The "debug-starts-each-session-from-scratch" sure makes it seem like a vscode or cpptools issue.
I will change the showDevDebugOutput to "both" and try to capture it again.

geoffdudds, 2023-01-17:
Ok. So in the following, the first read was successful and the second failed:
[image]

But I can't imagine this is very helpful...
I have had this occur where I was also not able to read the same variable from the "Watch" window. I will try to capture that as well - maybe it will be more helpful.

haneefdm, 2023-01-17:
That is not the output using "both" is it? I am not seeing any chatter from VSCode. Just from us.

We always try to see if an expression was previously used and try that first and then if that fails create a new gdb-variable (which are different from your variables/expressions) using your expression.

Are you not getting the value or does it display nothing on hover?

geoffdudds, 2023-01-18:
I get a tiny little empty box when I hover:
[image]

It does appear to be set as "both". I reloaded vscode and get the same output when I hover on various var's.

haneefdm, 2023-01-18:
I am sorry. Could you set the debug output format to "vscode". It should look something like this.
```
113-stack-select-frame --thread 1 0
-> 113^done
114-stack-list-variables --thread 1 --frame 0 --simple-values
-> 114^done,variables=[{name="result",type="cy_rslt_t",value="<optimized out>"},{name="a",type="volatile int",value="268457064"}]
115-var-update --thread 1 --frame 0 --all-values var_result_256
-> 115^done,changelist=[]
116-var-update --thread 1 --frame 0 --all-values var_a_256
-> 116^done,changelist=[]
To client: {"seq":0,"type":"response","request_seq":98,"command":"variables","success":true,"body":{"variables":[{"name":"result","evaluateName":"result","value":"<optimized out>","type":"cy_rslt_t","presentationHint":{},"variablesReference":0},{"name":"a","evaluateName":"a","value":"268457064","type":"volatile int a;\ndec: 268457064\nhex: 0x10005468\noct: 002000052150\nbin: 00010000 00000000 01010100 01101000","presentationHint":{},"variablesReference":0}]}}
From client: evaluate({"expression":"result","frameId":256,"context":"hover"})
117-var-update --thread 1 --frame 0 --all-values hover_1282714a99aac884e353537463450d7f6f4115ee8b01cb369d0313688116a179
-> 117^error,msg="Variable object not found"
118-var-create --thread 1 --frame 0 hover_1282714a99aac884e353537463450d7f6f4115ee8b01cb369d0313688116a179 * "result"
-> 118^done,name="hover_1282714a99aac884e353537463450d7f6f4115ee8b01cb369d0313688116a179",numchild="0",value="<optimized out>",type="cy_rslt_t",thread-id="1",has_more="0"
To client: {"seq":0,"type":"response","request_seq":99,"command":"evaluate","success":true,"body":{"result":"<optimized out>","type":"cy_rslt_t","presentationHint":{},"variablesReference":0}}
```

The "From Client:" and "To Client:" is the chatter between VSCode and us.

haneefdm, 2023-01-19:
Yes, `restart` and `reset` do not restart the debug session and you will see things cached.

That aside, this is what I see. Take `config.kb` for instance. In the final case (346-var-update) where it did not produce any result, we asked gdb for an update. The answer was nothing changed. So, we return whatever value we already had. But it appears that the cached value does not have a result field at all. This should not be possible but it did.

So, we have to track every request for `config.kb` and see what every result/transaction looked like. You can send me the entire Debug Console (attached a compressed text file).

Could you also use the latest pre-release (google how to install an extension using a VSIX file).

https://github.com/Marus/cortex-debug/releases

haneefdm, 2023-01-19:
https://github.com/Marus/cortex-debug/releases/tag/v1.6.11-pre2

Please make sure it is v1.6.11-pre2 which has a tiny change that could affect this issue in a good way.

geoffdudds, 2023-01-19:
pre2 gets closer but results in a different problem.
- I enter a function a hover over config.kb, returns  . This is correct.
- I enter a different function that operates on a different instance of type `config_t`. It also returns 876.923096. This is not correct. It appears to be using the cached value of the previous instance of the `config_t` type.
- If I then hover over `config` instead of `config.kb`, it shows the values of all members of the config struct, including kb. The value of kb is 0.0930841491, which is correct.

It seems like the cache is explicitly based on the name of the variable? ie the problem occurs when I try to read variables of the same name in different scopes?

Attached is a log of:
1. start debugger
2. breakpoint in function `calcMotorConfig`
3. hover over `config.kb` (kb = 876.923096)
4. continue
5. breakpoint in function `calcSpeedConfig`
6. hover over `config.kb` (kb = 876.923096)
7. hover over `config` (kb = 0.0930841491)

[log.zip](https://github.com/Marus/cortex-debug/files/10460799/log.zip)

haneefdm, 2023-01-19:
This is the problem. It appears that gdb things the value changed but does not even return a value
```
117-var-update --thread 1 --frame 0 --all-values hover_d48aa0f739e7ee01fcdfb09cbca36651d9843f8250e8ee3caba95dccbedb9d56
-> 117^done,changelist=[{name="hover_d48aa0f739e7ee01fcdfb09cbca36651d9843f8250e8ee3caba95dccbedb9d56",in_scope="false",type_changed="false",has_more="0"}]
```

So, I end up using the previous value. It should ALWAYS give me the value if it got updated. This is why before 'pre2' I did not even see a result. In pre2, you are seeing the previously cached value and which is wrong and should be wrong.

So, not sure what to do if GDB refuses to give me the value but says value changed.. We 1000% rely on GDB doing the right thing.

Would it be possible for you to try an older version on GDB. You seem to be on the most recent version. Perhaps try 11.x

geoffdudds, 2023-01-19:
Interesting. Ok, I'll try that - I'm going to be tied up for a bit so I might not have results until end of day, or possibly early tomorrow.

thanks,

haneefdm, 2023-01-19:
Okay, it may not be a GDB problem. It is a problem with the variable within scope or not. This depends on the current frame you selected in the STACK window. From GDB docs. We are not even checking to see if in_scope is true or not. But GDB is claiming that the 'expression' being given is not in the scope of the current frame. That is the problem. So, what was happening in the current production release is actually correct -- I need to backout the change in pre2.

[image]

haneefdm, 2023-01-19:
I am creating a workaround and will shortly publish it as pre3. This is difficult for me to test because I cannot reproduce it. All I can do is defensive programming and hope it works. We will have to analyze the debug output carefully though. I will let you know when pre3 is ready.

haneefdm, 2023-01-19:
Can you try the latest when you get a chance.

https://github.com/Marus/cortex-debug/releases

geoffdudds, 2023-01-20:
Do you still want me to try with gdb 11.x?

haneefdm, 2023-01-20:
No, lets try pre3 first.

geoffdudds, 2023-01-20:
This one crashes when it gets to the second function.
- when I break in function 1, it reads ok
- if I then break in function 2, it crashes after a few seconds even if I do not hover over anything
- if I don't break in function 1, but break in function 2, it reads ok

This log is of breaking in function one, hovering on config.kb, then breaking in function 2.

[log2.zip](https://github.com/Marus/cortex-debug/files/10462102/log2.zip)

haneefdm, 2023-01-20:
If I am not mistaken, that log file says 'pre2'. Not 'pre3'. The first line should look something liek
```
Cortex-Debug: VSCode debugger extension version 1.6.11-pre3 git(5b9d21b+dirty). Usage info: https://github.com/Marus/cortex-debug#usage

```

geoffdudds, 2023-01-20:
Sorry, I must have crossed my files there. I think I've attached the right log this time.

[log3.zip](https://github.com/Marus/cortex-debug/files/10462202/log3.zip)

haneefdm, 2023-01-20:
Crash should be fixed. Could you try again? It is now pre4

geoffdudds, 2023-01-20:
It worked! Looks like it deletes then recreates config.kb. I'm assuming due to it falling out of scope?

[image]

I attached a log in case there is anything you want to gleam from it:
[log4.zip](https://github.com/Marus/cortex-debug/files/10462444/log4.zip)

Really appreciate you helping with this - its a great extension!
I will do lots of "hovering" for the next week or so and let you know if I see this again.

thanks,
Geoff

haneefdm, 2023-01-20:
Oh, yeah. Thanks for the log. I have to study the log very carefully because I cannot reproduce this myself.

I have external clients (memory viewer, rtos viewer, MS Embedded Extension, others, etc.) that use the hover feature programmatically, I hope I didn't break it for them. They rely on the ID's being consistent/canonical.

haneefdm, 2023-01-20:
When you get a chance, can you also try it on a non-scalar value - like a struct or a class or pointer to one or an array. Why this is happening in the first place -- I have to think about it a lot. It affects other areas -- the variable window. Same logic is used over there as well.

I am glad we put the effort in creating the debug tools to self-debug our extension. 99% of the time I am working blind. And **thank you** for working with me patiently. I know the debug output can be hard to read...

If you are using an RTOS that the RTOS extension supports, we could test that too. It doesn't appear you are using one.

Oh, I see you are my neighbor to the North :-)

geoffdudds, 2023-01-20:
Ok, so I tried to collect a set of data that will hopefully give you lots to work with.
I tried to read a reference to a struct via:
1. hover
2. variable window
3. watch window

I did each of those with v1.6.10 and v1.6.11-pre4. I doubt the 1.6.10 was needed but it was quick to do.

A couple notes:
- when reading fails in the variable or watch window, no debug text is produced. I took a small screenshot of what I saw in those cases. 
- the `this` pointer also fails. You can see this in the screenshot. Presumably for the same reasons as the `config` struct I was actively trying to read.

[logs 5-10.zip](https://github.com/Marus/cortex-debug/files/10469651/logs.5-10.zip)

That's interesting that it can't be reproduced on your end. There are a number of us here that see the same thing. I'm not sure what we have in common yet. Could the debug h/w be involved somehow? We are all using JTrace's or JLinks.

Unfortunately, as you said, I am not using an RTOS. I can't think of any projects in our group I could readily test it out with. I could probably put together a shell of an application at some point, but I'm admittedly not very familiar with using RTOS's so it might take me a bit.

No thanks required! I am very grateful to have such a nice tool! 

Hey look at that, we are pretty close neighbors. My wife and I drove the Oregon coast a few years ago - beautiful area, nice people.

haneefdm, 2023-01-20:
Unfortunately, I have near zero tests with C++.

I wish I had some generic C++ stuff that I can insert into any project/device and try it out. Reproducing also means that the call stack has to be a certain way, and a calling pattern.

I have a feeling that the same problem that existed with hover also exists with variable/watch windows. It is a misunderstanding between my understanding, gdb documentation and what gdb actually does. They are not always in sync. We had to reverse engineer many things wrt. gdb. And, things change in GDB in subtle ways over versions because they felt it was safe. Technically they are right.

The screenshot though, is a problem with our code. It crashed but we caught the exception and blindly reporting back. Wish I could have attached the entire stack trace but VSCode would have barfed.

haneefdm, 2023-01-20:
I completely botched my current development environment and code base. I am working on providing `"Live Watch"` so you can have a set of expression that you can see updated as your program is running...

Required severe refactoring -- I am pretty sure I have to increment the minor version. Should be totally compatible but with this much refactoring, one can never tell.

Of course on a separate branch. Chances of this working out are less than 10% at the moment. Chances can change drastically soon :-)

Preview:
[image]

geoffdudds, 2023-01-20:
A "Live Watch" would be awesome! It sounds dangerously close to graphing too. Sounds like you have a lot on the go.

If its any help, it seems what you did for the hover feature also fixed the watch window. This fix also works for c++ `this`. As far as I can tell, everything reads correctly for hover and watch with pre4.

closed 2023-03-09 by haneefdm