Marus/cortex-debug #810: Enhancement: Implement Live Watch

haneefdm, 2023-01-25:
Prototype:

https://user-images.githubusercontent.com/41269583/214466216-014ef68a-8db4-4b62-a911-f4bbcba5078f.mp4

This has been on my list for a long long time. I finally got something. I don't know what people want to see and who can help. Here is some info about this implementation.

* Only global variables, please. And their member fields if they are structs -- coming soon.
* Only works with OpenOCD, currently. How it works internally
   * Configure OpenOCD to accept multiple GDB connections. This part is very fragile and it has to be done before `init`. This part is customizable, and if the default mechanism doesn't work, you need be comfortable with configuring the target CPU to allow multiple connections.
   * Launch a second instance of gdb that does not react/receive any events. This is also fragile because I have no guarantees on how this works. GDB should not get events like stopped/breakpoint/other from OpenOCD. If it does, then it is ruined. So far, so good
* It will probably work with JLink but I have no clue based on their docs.
  * The second gdb solution may not work at all.
  * We can probably do raw memory reads and writes and hopefully all gdb servers support this. Meaning we have to do the complex work gdb does for us and I am not up to that. That complex work is decoding/translating types, structs, struct layouts, unions, and the list goes on and on. Super error-prone and I dread this.
* Maybe STLink has the same story as JLink. Zero docs, proprietary
* I don't know what the minimum version of GDB this would require.
* Of course, the first request I will get is to plot. Again, I can create an interface/API but need help in creating the HTML/CSS views to plot the stream. We can probably reuse the SWO/RTT plotting we got but I am not comfortable with that. Actually, I am ignorant too. I didn't write that and I am the wrong person.

HELP Needed: Once we get OpenOCD working, I will need volunteers to help figure out stuff with JLink/STLink. Especially from those who bought JLink probes and make requests for docs. I doubt they would talk to me. Note that I did everything on the command line to prove the mechanism works before I put it in the extension. Much faster that way and no need to learn TypeScript/VSCode-APIs

**I have a feeling I am creating a HUGE support nightmare for myself.**

@PhilippHaefele care to comment?

PhilippHaefele, 2023-01-26:
🥳 
Very nice and happy to see some progress in this direction, even i see fewer use cases where i personally need this in my daily business (as live watch hasn't any context and i do not have many measurement values in the system).

Your screen cast looks very promising and i also think that globals and their struct members are enough

> It will probably work with JLink but I have no clue based on their docs.
> The second gdb solution may not work at all.

This solution may work as the link i previously posted in some other issue states this kind of memory reading: https://forum.segger.com/index.php/Thread/3905-SOLVED-GDB-server-read-data-structures-while-target-is-running/?postID=13853#post13853
Not sure if it follow the rules mentioned like `Launch a second instance of gdb that does not react/receive any events`, but should be a staring point.

> HELP Needed: Once we get OpenOCD working, I will need volunteers to help figure out stuff with JLink/STLink. Especially from those who bought JLink probes and make requests for docs. I doubt they [...]

Happy to help with J-Link, but i must say that i maybe do not have that much time in the near future (2nd child in the pipline with ETA beginning of Feb 👶). So not predictable when i have time for this. Maybe more or less, nobody can say that yet 😆.

> I have a feeling I am creating a HUGE support nightmare for myself.

Presumably you/we will get some more tickets regarding this, but this will remove one of the last points other (commercial) IDEs have and user say that they are missing it.

PhilippHaefele, 2023-01-26:
> @PhilippHaefele First congrats and Feb is almost here. Take care and focus on baby and mom.

Thanks and yes will focus on them in the first place

> With the link provided, it appears the method I used for OpenOCD will work for JLink as well. But it is scary that they mention using `monitor` commands which I am trying to avoid. Doesn't make sense.

The comment also states `and use the GDB Protocol commands to read memory` what ever this means. Also hoping we do not need to use special commands. I think he maybe mentioned that in case you use the same session for debugging. We'll see and i can make a request / forum post on that topic, once we've tested it in more depth.

> One way to support JLink is to use OpenOCD and the interface set to JLink.

Of course this is always an option

> With the link provided, it appears the method I used for OpenOCD will work for JLink as well. But it is scary that they mention using `monitor` commands which I am trying to avoid. Doesn't make sense.

> It turns out, the second instance of GDB is getting all the events but this is working so long as I don't mess with 2nd GDB to alter the program state. Why? I don't know. I did enable non-stop mode

https://forum.segger.com/index.php/Thread/1425-JLink-GDBServer-async-mode/?page=Thread&threadID=1425
But as the other thread was from 2017, is it somehow working, i think they implemented the mode in general.

haneefdm, 2023-01-26:
When I ask GDB to evaluate an expression, it always turns into one or more memory reads for the gdb-server. Except gdb interprets what the bytes mean. A gdb server does not even know what variables/expressions are.

The async mode doesn't do much. We always have that enabled and it still does not allow most commands while the target is running. It is the addition of non-stop mode that does the trick but none of the gdb-servers explicitly support non-stop mode.

https://sourceware.org/gdb/onlinedocs/gdb/Asynchronous-and-non_002dstop-modes.html

> EDIT: Is the not sending events to the non-stop mode instance documented somewhere, so i can ask Segger about that?

No, I am getting all the events FROM GDB. Every one of them that the 1st instance gets. But GDB is unfazed?!?!?!

What non-stop does is it allows gdb to examine the target while the program/threads running. More importantly, you can have other threads running while one is stopped. You can break/continue individual threads. This is the part most gdb-servers cannot support but the examining part still works. Since I don't query/control any of the threads in the 2nd instance, GDB assumes that there are other side effects (besides threads, like HW making changes) that can cause the program state to change. The effect is also for GDB not to cache memory reads -- and this is very important to us.

I think I just explained to myself why this is working

haneefdm, 2023-01-29:
Updated the first comment now that non-scalars (arrays, structs, unions) are supported

haneefdm, 2023-01-29:
**Good news!!! The GDB solution works with JLink as far as I can tell.**

However, I am seeing some stability issues. In some cases, suddenly breakpoints don't work and the pause button doesn't work, variables don't update, etc. If I just kill and restart the session it works again. I can't pin it down and cannot reproduce it at will.

haneefdm, 2023-01-29:
**And, it works with STLink as well!** Before I only allowed a certain set of gdb-servers and disabled live-watch for all others. Now, I simply print a warning and let the users try to see if it will work with their gdb-server -- and asking them to report back. I don't have the bandwidth to test with all the other servers.

So far I had to use some trickery only on OpenOCD to configure it for this. JLink and STLink just worked with no extra configuration.

I am glad I picked the GDB (lazy) way. Reading memory myself and constructing structs with various layouts would have been a nightmare.

OptifySudarshanPatil, 2023-01-31:
@PhilippHaefele It's extremely needed and useful feature you've added! 
Please point me to direction of documentation to enable this in windows using Jlink.
Regards,
sdrshnptl

PhilippHaefele, 2023-01-31:
@OptifySudarshanPatil It was @haneefdm, not me who added the feature. Just part of discussion and some kind of development counterpart / helper. You maybe have meant both of us with "you've" (that's a flaw in the english language though), then sorry for my misunderstanding 😖 

And yes there are usages for that, but the developers need to keep in mind that this information is asynchronous to code execution (what a lot of people do not think about).
This means that the value maybe had some other values between two reads (that was not shown or read), that a user should correlate two different values with a lot of care (one could be new where the other is still old).

That's why i try to debug such things with methods like RTT, as i can guarantee that it is consistent (right data set @ right time point of execution).

Never then less i will of course use it e.g. for a quick check if the values are plausible or for values that very rarely change their value and I'm only interested in the average value.

and thanks for testing stuff 👍

OptifySudarshanPatil, 2023-01-31:
> @OptifySudarshanPatil It was @haneefdm, not me who added the feature. Just part of discussion and some kind of development counterpart / helper. You maybe have meant both of us with "you've" (that's [...]

My bad for confusion!
@haneefdm  Thank to you as well!

haneefdm, 2023-01-31:
@OptifySudarshanPatil If you have built from the latest sources (as you appear to have done), there is no extra configuration needed. What you have is enough.
```
"liveWatch": { "enabled": true, "samplesPerSecond": 4 }
```

If you are running our extension from the debugger, the instructions are at the bottom of this readme (you will need one change to your launch.json

https://github.com/Marus/cortex-debug#how-to-debug

There is one global setting you can experiment with that controls the refresh rate. Using the '+' button on the `CORTEX LIVE WATCH` title bar, add new items to the panel. It is just like the regular watch window.

I haven't made a public release yet. I have to do some more work. It will be a Preview release and if you enable Previews for this extension, VSCode will auto-download updates.

PhilippHaefele, 2023-01-31:
@haneefdm Also did some testing on [latest version from yesterday](https://github.com/Marus/cortex-debug/commit/927f1a2e4ce27f9e95fb358256c5cda8ff981bdc) and it worked quite well with my J-Link setup 👍

haneefdm, 2023-01-31:
@PhilippHaefele Thanks for checking with JLink.  I am thinking of making a Pre-release on the marketplace so users can selectively opt-in.

I have many other dangerous changes coming :-) that affects Variables and Watch windows.

haneefdm, 2023-03-06:
Hmmm. Which gdb-server?

PhilippHaefele, 2023-03-06:
As usual the J-Link GDB Server. Just noticed i didn't tested the normal watch. Will do more test tomorrow and attach more information or maybe even debug into it.

Best regards
Philipp

haneefdm, 2023-03-06:
See this comment. JLink was not quite smooth for me.

https://github.com/Marus/cortex-debug/issues/810#issuecomment-1407742188

 It may display that message if there were no items in the liveWatch window as well or if liveWatch was disabled. But I see you have one variable so the window is not empty. Seems like the `OSTime` is also available

PhilippHaefele, 2023-03-06:
Mhh i'm not sure anymore if i did have it enabled in the `launch.json` 🫣.

So that's maybe a false alarm. If so we could maybe add a pop-up / message in the right corner if someone tries to add an expression while live-watch is disabled.

Will check the configuration in the first place tomorrow.

haneefdm, 2023-03-06:
Ok, I duplicated after a dozen tries.

[image]

1. First you added a variable to Live Watch at some point.
2. Then you disabled liveWatch. In my case, I used `liveWatch.enabled = false` in launch.json

Because of multi-core and multi-program issues, that is all I can say. It is not enabled. Not sure I can print a popup message and when to do it. Ideas welcome

PhilippHaefele, 2023-03-07:
Oh my fault, live watch was disabled in the `launch.json`.

I would like to see an error message (e.g. `vscode.window.showErrorMessage()`) when the expression is not created in live watch, as there's currently no visual feedback that something hasn't work (other than the expression not showing up in the live watch window).

I see two cases when this can happen:

1. Debug session is not started at all
2. live watch is not enabled in launch.json

haneefdm, 2023-03-07:
Try the latest pre-release. 1.9.1

hongshui3000, 2023-03-09:
I was pleasantly surprised to find that he was able to monitor the value in the address

[image]

haneefdm, 2023-03-09:
Yes, in the next release. Note in the` Watch` and `Live Watch`, you can always get hex values (only for scalars) by adding a `,x` to the expression. This has been true for a couple of years and we borrowed that convention from Visual Studio

[image]

The `0x` and `n` buttons is global to the entire extension

[image]

haneefdm, 2023-03-09:
Some more buttons to manipulate what you are watching

[image]

hongshui3000, 2023-03-11:
>Launch a second instance of gdb 

btw
@haneefdm Is it possible to only open a gdb connection and implement a live watch？

sadkotheguest, 2023-03-19:
I am trying to use LiveWatch with RISC-V (CH32V307 MCU) using settings:

```
   "configurations": [
        {
            "name": "Start debug on CH32V307",
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceFolder}/build/firmware.elf",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "servertype": "openocd",
            //"showDevDebugOutput": "raw",
            "liveWatch": { "enabled": true, "samplesPerSecond": 2 } ,
            "svdFile": "${workspaceFolder}/svd/CH32V307xx.svd",
            "configFiles": [
                "${workspaceFolder}/wch/wch-riscv.cfg"
            ],
            "postLaunchCommands": [
                "b main",
                "set mem inaccessible-by-default off",
                "set architecture riscv:rv32",
                "set remotetimeout unlimited"
            ],
            "preLaunchTask": "cmake build"
        }
```

Got some messages in terminal:

```
Failed to read priv register.
Target wch_riscv.cpu.0: Failed to read memory (addr=0x20000008)
  progbuf=skipped (target not halted), sysbus=skipped (unsupported size), abstract=failed
```

When I am using _Log message breakpoint_ value in LiveWatch window is updating but sometimes it shows all-zeroes.

haneefdm, 2023-03-19:
>   progbuf=skipped (target not halted), `sysbus=skipped (unsupported size)`, abstract=failed

@sadkotheguest  First, I have minimal experience with RV architecture. From what I know, your CPU must allow access to the system bus to access the memory while the CPU is running. This bypasses the HART. It is optional in RV. Live Watch will never work if such access is not allowed.

See page 5/7/18 https://riscv.org/wp-content/uploads/2019/03/riscv-debug-release.pdf (esp. section 3.9 page 18)

Also, most of what you are doing in `postLaunchCommands` should be done in `preLaunchCommands`. You may also be wasting a breakpoint by setting it launch sequence. It is also unmanaged. You should set a line breakpoint in VSCode so it can be managed

sadkotheguest, 2023-03-19:
>  First, I have minimal experience with RV architecture. From what I know, your CPU must allow access to the system bus to access the memory while the CPU is running. This bypasses the HART. It is optional in RV. Live Watch will never work if such access is not allowed.
 
 @haneefdm 
Thank you for your advice and for the great plugin. I'll continue to figure out what's going on.

haneefdm, 2023-04-18:
It is in the plans. Wasn't sure what format to use for log files. Plotting is different as it can be done as a time-series.

What does VisualGDB use for a format? Is it open-source?

lijon, 2023-06-05:
I tried liveWatch and it ended with my USB-C hub (built into my LG screen) loosing all connections and requiring a power cycle :)

I've just discovered that at least with openocd, one can print variables etc in the gdb console *without* using a 2nd instance of gdb:

1. pause the debugger
2. type `mon resume` in debug console
3. we can now `p g_some_variable` etc..
4. Also, clicking the reload button in XPERIPHERALS updates the register values correctly.

So, maybe this procedure could be used instead?

And, would it be possible to have a "reload" button on the VARIABLES panel just like the XPERIPHERALS?

haneefdm, 2023-06-05:
> xAnd, would it be possible to have a "reload" button on the VARIABLES panel just like the XPERIPHERALS?

Why? Also, the VARIABLES panel is owned and controlled by VSCode. We also don't own the XPERIPHERALS, btw. It is owned by a separate extension.

> I've just discovered that at least with openocd, one can print variables etc in the gdb console without using a 2nd instance of gdb:

Doesn't work with other gdb servers and doesn't work with my version of openocd. Not sure how it is working for you.m Also we cannot use `mon resume` as that is not supported universally. We HAVE to use gdb resume. If you use `mon resume` and even if that works for you seemingly, now gdb and VSCode and us are all our of sync as to the current state of the program/

> I tried liveWatch and it ended with my USB-C hub (built into my LG screen) loosing all connections and requiring a power cycle :)

Note that we do not interact with ANY HW on your computer. So, not sure how this is possible.

lijon, 2023-06-05:
> > xAnd, would it be possible to have a "reload" button on the VARIABLES panel just like the XPERIPHERALS?
> 
> Why? Also, the VARIABLES panel is owned and controlled by VSCode. We also don't own the XPERIPHERALS, btw. It is owned by a separate extension.

To refresh it in case the values changed.

> Doesn't work with other gdb servers and doesn't work with my version of openocd. Not sure how it is working for you.m Also we cannot use `mon resume` as that is not supported universally. We HAVE to [...]

I'm using openocd 0.12.0 with an ST-Link/V2-1 with latest st-link firmware. Yes I understand they become out of sync, in that the debugger thinks it is stopped at some program address while it's actually still running. However, gdb can still print the values of variables etc. Just thought this might be a viable option, to use the single gdb instance instead of a second one, but I guess not if `mon resume` is an openocd-only thing and there's no similar commands for other gdb servers.

> > I tried liveWatch and it ended with my USB-C hub (built into my LG screen) loosing all connections and requiring a power cycle :)
> 
> Note that we do not interact with ANY HW on your computer. So, not sure how this is possible.

Yeah, I didn't think it was directly caused by this extension. I have no idea how it happened, maybe some bug in openocd that triggered something in the ST-Link/V2-1 firmware, that in turn made the USB hub hardfault, or something..

haneefdm, 2023-06-05:
Debug is far higher priority than live variables. So, no we **cannot** have VSCode, us, gdb, other extensions out of sync. We HAVE to play by GDB rules. There is no choice here.

tblaha, 2023-08-24:
Hey all, thanks for implementing this great feature! I saw a popup asking me to report back to you guys if live-watch works on "external". On my setup it does!

My setup:
- Raspberry Pi running openocd 0.12.0 (from the standard Raspberry Pi OS repo)
- STM32H743 target
- sourced and called the CDLiveWatchSetup function from https://github.com/Marus/cortex-debug/blob/master/support/openocd-helpers.tcl
- worked right out of the box, although I haven't played around with it much.

mpekurin, 2023-09-06:
> Live watch is not supported for servertype 'external'. Only 'openocd', 'jlink', 'stlink' supported/tested. Report back to us if it works with 'external'

My setup is vscode connected to WSL + `ST-LINK_gdbserver.exe` running on the Windows side for debugging, so I guess it's not "very" external.

However, the `launch.json` config is as follows:

```json
    {
      "type": "cortex-debug",
      "name": "Launch",
      "request": "launch",
      "servertype": "external",
      "gdbTarget": "${env:HOSTNAME}.mshome.net:61234",
      "executable": "${workspaceFolder}/build/${workspaceFolderBasename}.elf",
      "svdFile": "${workspaceFolder}/STM32L552.svd",
      "preLaunchTask": "GDB Server",
      "postDebugTask": "Kill GDB Server",
      "overrideRestartCommands": [
        "monitor reset",
        "break main"
      ],
      "overrideLaunchCommands": [
        "monitor reset",
        "break main"
      ],
      "liveWatch": {
        "enabled": true
      }
    }
```

And thank you for this awesome feature!

haneefdm, 2023-09-06:
There are no plans to support "external" server types. We need to know a LOT about how the gdb-server works to make live-watch and we know zero for "external" server types

mpekurin, 2023-09-11:
> There are no plans to support "external" server types. We need to know a LOT about how the gdb-server works to make live-watch and we know zero for "external" server types

If this is regarding my comment, I just wanted to report that the feature works as expected with my setup. I guess it will also work with any "standard" external server as well (openocd, jlink or stlink used as external). So maybe it would be a good idea to add a way to disable the `not supported for servertype 'external'` popup (but still show it by default, of course).

haneefdm, 2023-09-12:
Thanks for getting back @mpekurin When using openocd, with an external server type, openocd does need some additional options/configuration in my testing. Did not work out of the box. It is true that JLink and STLink did not require additional configuration.

I don't want to add an option to disable the message. The number of options has become unmanageable and they are not cheap to add. I changed the message type to an Information Message which disappears after a while. Also, changed the message a bit for external types.

OptifySudarshanPatil, 2023-10-14:
Hello @haneefdm and all,
I've tried live watch earlier with Jlink, which work out of the box and fantastic feature!!

any chance this feature will be alive on "BMP"
[image]

[image]

```
{
            "name": "BMP Debug",
            "cwd": "${workspaceRoot}",
            "executable": ".pio/build/due/firmware.elf",
            "request": "launch",    
            "preLaunchTask": {
                "type": "PlatformIO",
                "task": "Pre-Debug"
            },
            "type": "cortex-debug",
            "servertype": "bmp",
            "interface": "swd",
            "device": "ATSAM3X8E",
            "BMPGDBSerialPort": "//./COM15",
            "runToEntryPoint": "setup",
            "liveWatch": {
                "enabled": true,
                "samplesPerSecond": 4
            }
        },
```

haneefdm, 2023-10-14:
Near zero chance it will work with `bmp`, if it didn't already work. They use a serial port and all the other servers use TCP ports., where the servers can accept multiple connections and, yet keep them separate.

OptifySudarshanPatil, 2023-10-14:
Agreed!
Thanks for the input!

dani9, 2023-10-18:
I tested the live watch feature with OpenOCD, and it's working great. Thanks!

Is there a way to modify the value of a watched variable without halting the target? Perhaps using a GDB command in the Debug Console?

This functionality is available via GUI in STM32Cube-IDE (Live Expressions) and MCUXpresso-IDE (Global Variables).

haneefdm, 2023-10-18:
@dani9 Thanks for trying it out. Yes, it is technically possible to modify but, I haven't had the time to implement it and test it. Will be a future addition. You can submit a separate issue for that so we can track it. Hopefully, you will be available to test it.

ALTracer, 2023-11-06:
> Near zero chance it will work with `bmp`, if it didn't already work. They use a serial port and all the other servers use TCP ports., where the servers can accept multiple connections and, yet keep them separate.

Greetings. What kind of support would be required from BMP side to implement Live Watch? They don't fully support Async mode or Non-Stop debugging yet, and it's true that there's only one serial gdb interface provisioned on upstream platforms. In other words, what gdbserver packets will you be throwing at /dev/ttyBmpGdb and when (is the target halted or running after `continue&`)? I don't think they'll be implementing `ST-LINKgdbserver`'s WriteDP and ReadAP remote monitor commands, because that violates encapsulation of internal ADIv5 handling structures, but once vCont and qNonStop packets become supported, I'd like to revisit this and maybe contribute some working solution when practically possible (I am not a maintainer there).

I mostly use Eclipse embedcpp CDT and its derivatives, but VSCode and Arduino IDE 2.0 (Eclipse Theia) look tolerable, too. Non-invasively reading RAM of capable Cortex-M targets seems to be a frequently asked question.

haneefdm, 2023-11-06:
@ALTracer It is upto the owners of BMP to look at OpenOCD, STLink, JLink to figure out if they can implement a second gdb connection for background reads. If not there is nothing I can do. We don't directly interact with the gdb-server to get live updates. gdb is the one doing background reads on our behalf which does not use things like WriteDP/ReadAP

flespark, 2023-11-15:
When live watch can support pyocd,. rather than openocd, I prefer pyocd for its cmdline toolbox trait. now enable live watch with pyocd will cause some trap like auto continue after paused

haneefdm, 2023-11-15:
@flespark Did you try it? It may just work. If so, let us know.

TomT0329, 2023-11-22:
Is it possible to change the registers value in the live watch windows when cpu is still running?

CaptainKAZ, 2024-01-26:
``hello i am using gdb-multiarch on ubuntu 20.04.
livewatch will not work if i set request to "launch", and the second gdb session seems crashed.
```
LiveGDB: GDB -> App: {"output":"","token":9,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
LiveGDB: 10-interpreter-exec console "set stack-cache off"
LiveGDB: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"cmd-param-changed","output":[["param","stack-cache"],["value","off"]]}]}
LiveGDB: GDB -> App: {"output":"","token":10,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
LiveGDB: 11-interpreter-exec console "set remote interrupt-on-connect off"
LiveGDB: GDB -> App: {"output":"","token":11,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
LiveGDB: 12-target-select extended-remote localhost:50000
LiveGDB: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-started","output":[["id","i1"],["pid","42000"]]}]}
LiveGDB: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-created","output":[["id","1"],["group-id","i1"]]}]}
LiveGDB: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"log","content":"/build/gdb-ZgDh0V/gdb-12.1/gdb/thread.c:1328: internal-error: switch_to_thread: Assertion `thr != NULL' failed.\nA problem internal to GDB has been detected,\nfurther debugging may prove unreliable."}]}
LiveGDB: /build/gdb-ZgDh0V/gdb-12.1/gdb/thread.c:1328: internal-error: switch_to_thread: Assertion `thr != NULL' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
LiveGDB: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"log","content":"\n"}]}
LiveGDB: 
LiveGDB: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"log","content":"----- Backtrace -----\n"}]}
LiveGDB: ----- Backtrace -----
LiveGDB: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"log","content":"Backtrace unavailable\n"}]}
LiveGDB: Backtrace unavailable
LiveGDB: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"log","content":"---------------------\n"}]}
LiveGDB: ---------------------
LiveGDB: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"/build/gdb-ZgDh0V/gdb-12.1/gdb/thread.c:1328: internal-error: switch_to_thread: Assertion `thr != NULL' failed.\nA problem internal to GDB has been detected,\nfurther debugging may prove unreliable."}]}
LiveGDB: /build/gdb-ZgDh0V/gdb-12.1/gdb/thread.c:1328: internal-error: switch_to_thread: Assertion `thr != NULL' failed.
A problem internal to GDB has been detected,
```
but if i switch request to "attach" it works normally
```
LiveGDB: 9-file-exec-and-symbols "/home/neo/Robocon2024-Embedded-BoardC/build/Robocon2024-Embedded-BoardC.elf"
LiveGDB: GDB -> App: {"output":"","token":9,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
LiveGDB: 10-interpreter-exec console "set stack-cache off"
LiveGDB: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"cmd-param-changed","output":[["param","stack-cache"],["value","off"]]}]}
LiveGDB: GDB -> App: {"output":"","token":10,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
LiveGDB: 11-interpreter-exec console "set remote interrupt-on-connect off"
LiveGDB: GDB -> App: {"output":"","token":11,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
LiveGDB: 12-target-select extended-remote localhost:50000
LiveGDB: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-started","output":[["id","i1"],["pid","42000"]]}]}
LiveGDB: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-created","output":[["id","1"],["group-id","i1"]]}]}
LiveGDB: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"prvCheckTasksWaitingTermination () at Middlewares/Third_Party/FreeRTOS/Source/tasks.c:3650\n"}]}
LiveGDB: prvCheckTasksWaitingTermination () at Middlewares/Third_Party/FreeRTOS/Source/tasks.c:3650
LiveGDB: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"3650\t\t\twhile( uxDeletedTasksWaitingCleanUp > ( UBaseType_t ) 0U )\n"}]}
LiveGDB: 3650			while( uxDeletedTasksWaitingCleanUp > ( UBaseType_t ) 0U )
LiveGDB: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"stopped","output":[["frame",[["addr","0x0800808c"],["func","prvCheckTasksWaitingTermination"],["args",[]],["file","Middlewares/Third_Party/FreeRTOS/Source/tasks.c"],["fullname","/home/neo/Robocon2024-Embedded-BoardC/Middlewares/Third_Party/FreeRTOS/Source/tasks.c"],["line","3650"],["arch","armv7e-m"]]],["thread-id","1"],["stopped-threads","all"]]}]}
LiveGDB: mi2.status = stopped
LiveGDB: Program stopped, probably due to a reset and/or halt issued by debugger
LiveGDB: GDB -> App: {"output":"","token":12,"outOfBandRecord":[],"resultRecords":{"resultClass":"connected","results":[]}}
Started live-monitor-gdb session
```
from the phenomenon described above, i guess the second gdb session only works when rtos kernel is started and "thread" is available?

haneefdm, 2024-01-26:
@CaptainKAZ I don't know how to help you.  I don't have a BMP or your HW. You can see the entire sequence from our logs/console. We don't have infinite resources. You can investigate further

> LiveGDB: /build/gdb-ZgDh0V/gdb-12.1/gdb/thread.c:1328: internal-error: switch_to_thread: Assertion `thr != NULL' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.

You should investigate why this is happening. You can even use command line tools to investigate. Post your solution/workaround so others can benefit

haneefdm, 2024-01-26:
Also, submit a new Issue with details.

closed 2024-01-26 by haneefdm

haneefdm, 2025-03-13:
> *is this "liveWatch" setting a common setting that all extensions have, or is it specific to this extension?

Unique to us, but other (commercial) debuggers have it. Not common in open source ones or VSCode extensions. Requires very special support from gdb-servers.

> *is the way "liveWatch" works, and how it should be set added to the wiki page/documentation? I don't find it anywhere, but maybe it is just me...

Reason it is called `liveWatch` is because how you use it is similar to the 'Watch` window that is builtin to VSCode. Difference is when it is updated. Most debuggers cannot update variables while the program is running. It only makes sense to global variables but anything gdb allows, we can allow but updates can fail if the variable/expression is not in global context. Unlike watch window, which has a context because of where the program is currently stopped, liveWatch has no context.

5S-OMAR-SWEIDAN, 2025-03-13:
> Unique to us, but other (commercial) debuggers have it. Not common in open source ones or VSCode extensions. Requires very special support from gdb-servers.
>
>
> Reason it is called `liveWatch` is [...]

Nice!
Can this info be added to the documentation maybe?

talpachen, 2025-07-02:
Great Job!
it works perfectly. DROP ozone and freemaster!