Marus/cortex-debug #811: XRTOS keeps turning on after restarting vscode

itzandroidtab, 2023-01-25:
After turning off the XRTOS view in the command window it turns itself back on after a restart of vscode.

haneefdm, 2023-01-25:
This issue belongs in the https://github.com/mcu-debug/rtos-views repo.

Can you please create this issue over there?

But, how are you turning off the XRTOS view? There should have been a setting to remember your preference. But there is a command in the Command Palette. Could you please try that as a workaround?

[image]

itzandroidtab, 2023-01-26:
@haneefdm Thank you for your response. I will create a issue in rtos-views.

That is also the only way I was able to turn it off. But every time it will turn back on
https://youtu.be/jFtUWc5dhno

closed 2023-01-26 by itzandroidtab