Marus/cortex-debug #812: SVD viewer: read-only fields do not have a `Copy Value` button

AndreySmirnov81, 2023-01-31:
Missing:
[image]

OK:
[image]

[image]

haneefdm, 2023-02-10:
This issue has been addressed in https://github.com/mcu-debug/svd-viewer

That will be the new home of SVD and will be removed from this repo when appropriate.

closed 2023-02-10 by haneefdm

AndreySmirnov81, 2023-02-13:
Thanks!

@haneefdm please look at the following comment:
https://github.com/mcu-debug/svd-viewer/commit/199e26eef71dbaeff1c06028506e851787feb51d#r100383410