Marus/cortex-debug #815: Question: How to get the Adapter Output Window to appear?

NOhs, 2023-02-01:
I am using cortex-debug to debug an RT600 evaluation board from NXP. Thanks for this great extension by the way 🙂. I am using the armclang compiler (6.14.1) to generate the image, the gcc-arm-embedded toolchain (10.2.1) for cortex-debug and JLink (V750) for communication with the device. My config is as follows:

```json
{
            "type": "cortex-debug",
            "request": "launch",
            "name": "armclang",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/mvp-app-hello/mvp-app-hello.elf",
            "serverpath": "C:/Program Files (x86)/SEGGER/JLink_V750/JLinkGDBServerCL.exe",
            "servertype": "jlink",
            "device": "XXXXXXXX",
            "serialNumber": "XXXXXXXX", //If you have more than one J-Link probe, add the serial number here.
            "runToEntryPoint": "main",
            "internalConsoleOptions": "openOnSessionStart",
            "postLaunchCommands": [
                "monitor reset halt",
                "monitor semihosting enable",
                "tbreak main",
                "continue"
            ],
            "svdFile": "XXXXXXXX",
            "preLaunchTask": "armclang_jlink_download"
        }
```

This works fine in the sense that I can start debugging, I can open an external Telnet terminal, connect to it and it displays all `cout` statements in that external Terminal.

However, from what I could read online, apparently, cortex-debug should automatically output into an `Adapter Output Window` under `Output`. But neither under `Output` nor under `Terminal` do I find anything that looks like the described `Adapter Output Window`. Is it possible to have the output in `vscode`, and if so, how do I activate this feature?

haneefdm, 2023-02-01:
Thanks for the feedback.

The Output window is now in a Terminal Tab. Look for `gdb-server`. Output windows are unidirectional. Terminals are bi-directional. The `gdb-server` should have your FW stdout and you can also use your stdin if you want

[image]

Please close the issue if you are satisfied.

NOhs, 2023-02-01:
Thank you for the promt reply! I see the gdb-server terminal tab. But my output does not appear inside. All I see are lines like these:

```
Read 4 bytes @ address 0x20180EF0 (Data = 0x201818F0)
Read 4 bytes @ address 0x201818F0 (Data = 0x201816A0)
Read 4 bytes @ address 0x201818F4 (Data = 0x20181628)
Read 4 bytes @ address 0x201818F8 (Data = 0x00000000)
Read 4 bytes @ address 0x201818FC (Data = 0x000003ED)
Read 4 bytes @ address 0x20181900 (Data = 0x00000064)
Read 4 bytes @ address 0x20181904 (Data = 0x00000064)
Read 4 bytes @ address 0x20181908 (Data = 0x00000000)
Read 4 bytes @ address 0x2018190C (Data = 0x00000000)
Read 4 bytes @ address 0x20181910 (Data = 0x20180EF0)
Read 4 bytes @ address 0x20181914 (Data = 0x00089A80)
Reading 8 bytes @ address 0x00089A80
Reading 8 bytes @ address 0x00089A88
Read 4 bytes @ address 0x20181918 (Data = 0x00000200)
Read 4 bytes @ address 0x2018191C (Data = 0x201814A0)
Reading 8 bytes @ address 0x201814A0
Read 4 bytes @ address 0x20181920 (Data = 0x00000001)
Read 4 bytes @ address 0x20181924 (Data = 0x00000000)
Read 4 bytes @ address 0x20181928 (Data = 0x00000000)
Read 4 bytes @ address 0x2018192C (Data = 0x00000000)
Read 4 bytes @ address 0x20181930 (Data = 0x00000000)
Read 1 bytes @ address 0x2018193C (Data = 0x04)
Read 1 bytes @ address 0x2018193D (Data = 0x00)
Read 1 bytes @ address 0x2018193E (Data = 0x02)
Read 1 bytes @ address 0x2018193F (Data = 0x0F)
```

Do you know what might be the cause of this?

haneefdm, 2023-02-01:
Not sure but for all other gdb-servers (OpenOCD), your stdout is part of the server's stdout. But, that is the entirety of what used to be in the Adapter output and what JLink has output on its stdout/stderr.

Just looked at the JLink documentation, and it appears they create a separate TCP port for that. We don't support that. However, there is another way.

[image]

https://wiki.segger.com/J-Link_GDB_Server#semihosting_enable

Add the following to your post-launch commands
```
monitor semihosting IOClient 2
```
We generally add such commands to pre-launch, but post-launch is also okay I guess. If it is working for you. The way you are doing it, it will cause an extra reset and potentially an extra wasted HW breakpoint.
```
            "preLaunchCommands": [
                "monitor semihosting enable",
                "monitor semihosting IOClient 2"
            ],
            "runToEntryPoint": "main",
            "postLaunchCommands": [
                "continue"
            ],
```
This way, our logic for `runToEntryPoint` actually gets used.

haneefdm, 2023-02-01:
Oh, sorry. You don't need the postLaunchCommans at all since that `continue` from the reset. If you remove postLaunchCommands, then it will work like you would expect

haneefdm, 2023-02-01:
Check updated comment

NOhs, 2023-02-01:
Thank you. Your suggestion works. The output now appears in the "DEBUG CONSOLE" tab. But I think I might stick with the external terminal as the "DEBUG CONSOLE" tab does not handle coloured output too well. Thanks also for the tip regarding the preLaunchCommand, that makes everything a bit faster 🙂.

Slightly off-topic: When I want to use the semihosting input feature, e.g. via `std::cin >> ...`, my program never waits for my input. How do I provide the input actually to my device?

haneefdm, 2023-02-01:
I am no good at semihosting. I actively avoid it. It is too slow and invasive. If you are using JLink, I recommend RTT.

https://www.segger.com/products/debug-probes/j-link/technology/about-real-time-transfer/

[image]

haneefdm, 2023-02-01:
Closing this as there is nothing more I can do? Want to keep my queue shorter

closed 2023-02-01 by haneefdm