Marus/cortex-debug #817: openocd.exe ENOENT

El-Bolas, 2023-02-04:
I've been looking at some of the solutions and for some reason I still can't get it to find the files (I think).
[image]
I'm getting that error seemingly because the path has not been set up correctly I guess, but when I try to set the paths in settings.json, it appears as though it's not possible to use these commands in my settings.json (I probably skipped something which is what I'd love if anyone could point out here)
[image]
[image]
I'm not entirely sure of why I'm getting that output, is it that I have something incorrectly configured in my extensions for Cortex-debug?
[image]
Those are the contents of my launch.json file, I've tried removing the  "gdbPath" : "arm-none-eabi-gdb" since some pointed out (if I understood correctly) that it should be configured in settings.json, but I can't do much of anything in my settings.json

I know that at the very least openocd.exe works because I runned it with src/openocd.exe from git and it seems to output an expected behaviour (I think)

El-Bolas, 2023-02-04:
Well, I'm able to write the paths in the settings.json that is used for all the vscode projects, but now the error is the following: 
[image]

I'm able to get a different error if I specifically set the path to the .exe file although I think I'm not supposed to do that, I get the following error then:
[image]
 ```
[2023-02-04T20:24:41.074Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:/Users/Julio/VSARM/sdk/openocd/src/openocd.exe" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "C:/Users/Julio/VSARM/sdk/openocd/tcl" -f "c:/Users/Julio/.vscode/extensions/marus25.cortex-debug-1.6.10/support/openocd-helpers.tcl" -f interface/picoprobe.cfg -f target/rp2040.cfg
Open On-Chip Debugger 0.11.0-g4f2ae61 (2023-02-01-17:20)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : clock speed 5000 kHz
Info : DAP init failed

[2023-02-04T20:24:41.215Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```
That's what I get in the terminal, any thoughts?

haneefdm, 2023-02-04:
> That's what I get in the terminal, any thoughts?

Nope. If you cannot run openocd from the command line neither can we. You can see the command line we use based on what settings you have. Try that. It is up to you and your device maker to specify how to configure openocd.

Things could be as simple as your device is not connected or powered on. It appears openocd cannot even find your device to be connected to your computer.

El-Bolas, 2023-02-04:
> Nope. If you cannot run openocd from the command line neither can we. You can see the command line we use based on what settings you have. Try that. It is up to you and your device maker to specify [...]

Forgive my ignorance here; the device that cannot be found is the debugger or the MCU being programmed to?

haneefdm, 2023-02-04:
It's okay. OpenOCD cannot find your MCU or board. You may want to read up on how most embedded debuggers work.

https://github.com/Marus/cortex-debug/wiki/Cortex-Debug-Under-the-hood

haneefdm, 2023-02-04:
Please look at the `Debug Console` Tab. Every external command we invoke, the exact command use is printed.

If you cannot launch openocd like we are trying to do then we need to figure that out. There is nothing we do special that you can't see in the Debug Console

El-Bolas, 2023-02-04:
> It's okay. OpenOCD cannot find your MCU or board. You may want to read up on how most embedded debuggers work.
> 
> https://github.com/Marus/cortex-debug/wiki/Cortex-Debug-Under-the-hood

Much appreciated, I'll read up on that and try again afterwards. Also, a completely unrelated and maybe out-of-place question: what did you guys study back in college? I'm struggling to take a decision these days so maybe it could help a little

El-Bolas, 2023-02-04:
> Please look at the `Debug Console` Tab. Every external command we invoke, the exact command use is printed.
> 
> If you cannot launch openocd like we are trying to do then we need to figure that out. There is nothing we do special that you can't see in the Debug Console

Ah my debug console outputs this: 
```
Cortex-Debug: VSCode debugger extension version 1.6.10 git(0458417). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from C:/Users/Julio/VSARM/armcc/10 2021.10/bin/arm-none-eabi-objdump.exe --syms -C -h -w O:/Documentos/.PICO/Projects/Picoprobe_1/build/Picoprobe_1.elf
Reading symbols from c:/users/julio/vsarm/armcc/10 2021.10/bin/arm-none-eabi-nm.exe --defined-only -S -l -C -p O:/Documentos/.PICO/Projects/Picoprobe_1/build/Picoprobe_1.elf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
    IMPORTANT: Set "showDevDebugOutput": "raw" in "launch.json" to see verbose GDB transactions here. Very helpful to debug issues or report problems
Launching gdb-server: "C:/Users/Julio/VSARM/openocd/openocd.exe" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "C:/Users/Julio/VSARM/sdk/openocd/tcl" -f "c:/Users/Julio/.vscode/extensions/marus25.cortex-debug-1.6.10/support/openocd-helpers.tcl" -f interface/picoprobe.cfg -f target/rp2040.cfg
    Please check TERMINAL tab (gdb-server) for output from C:/Users/Julio/VSARM/openocd/openocd.exe
Finished reading symbols from objdump: Time: 83 ms
Finished reading symbols from nm: Time: 87 ms
C:\Users\Julio\.eide\tools\gcc_arm\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
OpenOCD: GDB Server Quit Unexpectedly. See gdb-server output in TERMINAL tab for more details.
```

haneefdm, 2023-02-04:
Focus on this
```
Launching gdb-server: "C:/Users/Julio/VSARM/openocd/openocd.exe" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "C:/Users/Julio/VSARM/sdk/openocd/tcl" -f "c:/Users/Julio/.vscode/extensions/marus25.cortex-debug-1.6.10/support/openocd-helpers.tcl" -f interface/picoprobe.cfg -f target/rp2040.cfg
```

> I'm struggling to take a decision these days so maybe it could help a little

Wish I had a magic answer to pass down to my kids/grandkids/nieces/nephews/friends. Now it is time to shock you. My BS degree was in dairy science -- yes, I know all kinds of things about milk products, microbiology, etc. I can make cheese from scratch. Hated what would follow as a career in the food industry. Then I went into Computer science for a year and then merged into electrical engineering and VLSI design. I design HW, and SW, am involved in AI/ML, and a bunch of things. I just go with what interests me or what my company needs next. As you can probably tell, I am a very old person.

You said "you guys". It is mostly just me. I have a couple of lovely people pitch in. I just maintain this extension as a passion, not too much more.

So, develop a passion, focus, and go where it takes you and you will find new passions. Commit totally but be ready for course corrections. Always be confident and make the effort to back it up. I have changed my jobs every 2-4 years even working at the same company. Don't be stale.

Mind you, I am no authority in giving advice.

El-Bolas, 2023-02-04:
Thanks a ton for your personal experience :) it definitely helps. I'm still quite little so I don't have much experience to work off of (I guess that's how people say it)

> Focus on this
> 
> ```
> Launching gdb-server: "C:/Users/Julio/VSARM/openocd/openocd.exe" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_por
> ```

By that, do you mean the `-c "gdb_port 50000" -c"tcl_port 50001"`? Maybe I should try setting the ports manually?

haneefdm, 2023-02-05:
Ports have nothing to do with it. Read openocd doc. Your openocd cannot find your HW

El-Bolas, 2023-02-05:
Well here I come with an update: I "fixed it" by switching the picoprobe and the mcu that I was trying to program, basically I did the same thing but with a different rp2040 and it worked. I did change the picoprobe.cfg by cmsis-dat.cfg along with the program for the picoprobe but I was initially getting the same problem of openocd not finding the hardware. I guess this means it was a hw issue(? I think it could have been a problem with the target board since the swd pins I might have soldered poorly because all the other rp2040 that I have here work just fine as well.
Basically sorry for wasting your time, and thanks for the career advice :)

El-Bolas, 2023-02-05:
I checked the pins with the multimeter and they are soldered correctly so I don't know

closed 2023-02-05 by haneefdm