Marus/cortex-debug #819: Unable to get command return value?

hongshui3000, 2023-02-05:
I added the cmake-tool command to launch.json, and cortex-debug can get the path of the elf file
```
"executable": "${command:cmake.launchTargetPath}"
```
I wrote an extension myself, and I also hope that cortex-debug can be used, but it seems that cortex-debug cannot be recognized, I don't know why, I am a novice
```
"executable": "${command:my.launchTargetPath}"
```
I can see that my extension is indeed called and my command is executed, but the path string I get back doesn't seem to be picked up by the cortex-debug extension
my extension code
```
async function launchTargetPath(config: GlobalConfig, context: vscode.ExtensionContext):Promise<string | null>{

	let project: ProjectConfig = context.workspaceState.get("my.project") ?? { isInit: false };

	if (project.target === undefined) {

		await vscode.window.showErrorMessage(`You must choose a project to build.`);
		return null;
	}

	let exists = await fs.pathExists(path.join(project.target ?? "", "build", "zephyr"));
	if (exists) {
		let buildTarget = path.join(project.target.toString(), "build", "zephyr", "zephyr.elf");
		vscode.window.showInformationMessage(`build path ${buildTarget}`);
		return buildTarget;
	}
	// Error message
	await vscode.window.showErrorMessage('Error getting build dir. Check build.');
   return null;
}
```
Can anyone help me?

haneefdm, 2023-02-05:
Try making everything sync. As in no awaits and simply return a string. Try it with a constant string first.

haneefdm, 2023-02-05:
You should post such questions on Stackoverflow like Microsoft recommends. You are likely to find extension experts over there.

closed 2023-02-05 by haneefdm