Marus/cortex-debug #821: Feature request

Dr-Itachi, 2023-02-08:
I'm not reporting a bug. please can you add support for lldb ? I want to use lldb for the debugger but it seems that it is only possiible with gdb? Could be interresting if you add a support for lldb too ?

haneefdm, 2023-02-08:
May I ask why? What does LLDB do what gdb doesn't? The last time I checked, lldb's machine-interface (MI) was incomplete and incompatible. We use MI to communicate with GDB,

Microsoft even has a customized lldb that they used to ship because the stock one did not work.

Dr-Itachi, 2023-02-09:
> May I ask why? What does LLDB do what gdb doesn't? The last time I checked, lldb's machine-interface (MI) was incomplete and incompatible. We use MI to communicate with GDB,
> 
> Microsoft even has a customized lldb that they used to ship because the stock one did not work.

Ah thanks. Did not knwo that. for the question i just want to give a llvm toolchain a try

haneefdm, 2023-02-10:
You can (probably) use clang (or even commercial compilers) and still use GDB. LLDB does not really add that much to what GDB does. In fact it takes away a few things. I am closing this issue.

However, for native macos executables, you need to use lldb. We don't deal with native apps.

Closing this issue

closed 2023-02-10 by haneefdm

gonzalolarralde, 2025-12-29:
Hello, just adding a comment here to check if reopening this feature request or opening a new one would be considered. I'm building using Embedded Swift and lldb is the only (sustainable) debugging tool available for Swift. 

Just hoping to open the conversation. Aside from the extra cortex-specific features that would be great to have, cortex-debug is very reliable on keeping openocd healthy and understanding the MCU internal state.

I understand this request is very niche, I'd be happy to put some work towards this if there's an openness to add lldb support as a setting.

Thanks!

haneefdm, 2025-12-29:
Adding a second backend (lldb) is a non-trivial task. Especially testing and maintaining. You can take a look at MEEngine from Microsoft and see what it took. You have a couple of options. Use cppdbg from Microsoft and use lldb as the debugger. (cppdbg supports gdb, lldb and MS own vstudio debuggers). If Microsft is still shipping a custom version of LLDB (because the default one had issues), then that totally rules it out for us. Or you can use other lldb debuggers like the following

https://github.com/llvm/vscode-lldb#connect-to-a-debug-server-on-the-current-machine

In both cases, you can configure or launch a gdb-server like openocd and have it connect your gdb-server and have basic flow going.

Also, all our other extensions (memory view, peripheral view, RTOS views) work with cppdbg and other debuggers.

gonzalolarralde, 2025-12-30:
Appreciate your time and response! will take a look on the options you provided.