Marus/cortex-debug #822: Cannot start debug session on RT1170-EVK

mastupristi, 2023-02-09:
Hello,
I am trying to use vscode + cortex debug + JLink to develop/debug on **RT1170**.
JLink reports no particular problems, but the never seems to get to the (breackpoint at the beginning of the) main.
Pausing execution the ProgramCounter is internal to the ROM bootloader.
However if I reset the micro the firmware starts correctly, so I assume the flash was written and the firmware is good.

Here some data:

- OS: Xubuntu 22.04
- Toolchain: gcc-arm-none-eabi-10.3-2021.07
  - gcc: 10.3.1
  - gdb: 10.2.90.20210621-git
- vscode: 1.75.0
  - Commit: e2816fe719a4026ffa1ee0189dc89bdfdbafb164
Date: 2023-02-01T15:29:17.766Z
Electron: 19.1.9
Chromium: 102.0.5005.194
Node.js: 16.14.2
V8: 10.2.154.23-electron.0
OS: Linux x64 5.15.0-58-generic
Sandboxed: No
- Cortex-Debug: v1.6.10
- JLink: 7.84f

launch.json configuration is:

```
        {
            "request": "launch",
            "type": "cortex-debug",
            "name": "AmaOnEVK Start",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/FW/image/AmaOnEVK.elf",
            "servertype": "jlink",
            "device": "MIMXRT1176xxxA_M7",
            "interface": "swd",
            "rtos": "FreeRTOS",
            "svdFile": "${workspaceRoot}/FW/bsp/device/MIMXRT1175_cm7.svd",
            "runToEntryPoint": "main",
            "preLaunchTask": "build",
            "showDevDebugOutput": "raw",
            "preLaunchCommands": [
                "directory ${workspaceRoot}/FW"
            ]
        },
``````

A **MIMXRT1170-EVK** board is needed to reproduce the problem, jumpers J5, J6, J7 and J8 must be removed. You need to have a working FW compile it and try to start a debug session.

I would like to point out that my setup works correctly on other boards with RT1051 or on the RT1050-EVKB. I can also debug the RT1176 correctly using MCUXpressoIde.

I attach gdb traces taken from MCUXpressoIde (working) and vscode (not working), as well as the JLinkGdbServer log taken from vscode

[gdb_cortex_debug.txt](https://github.com/Marus/cortex-debug/files/10699396/gdb_cortex_debug.txt)
[gdb_expressoide.txt](https://github.com/Marus/cortex-debug/files/10699397/gdb_expressoide.txt)
[JLinkGDBServer_log.txt](https://github.com/Marus/cortex-debug/files/10699398/JLinkGDBServer_log.txt)

haneefdm, 2023-02-09:
You are hitting a breakpoint before main
```
-> ~"2 received signal SIGTRAP, Trace/breakpoint trap.\n"
2 received signal SIGTRAP, Trace/breakpoint trap.
-> ~"[Switching to Thread 536951996]\n"
[Switching to Thread 536951996]
-> ~"0x00223104 in ?? ()\n"
0x00223104 in ?? ()
-> *stopped,reason="signal-received",signal-name="SIGTRAP",signal-meaning="Trace/breakpoint trap",frame={addr="0x00223104",func="??",args=[],arch="armv7e-m"},thread-id="2",stopped-threads="all"
```

That stopped somewhere and that is not in main, but before main. We expect the first stop to be in main. `runToEntryPoint` does expect other things happening before we get to the entry point.

Try setting a breakpoint manually in main and remove `runToEntryPoint` and it might work.

In the MCUXpressoIde, the are resetting the device at that point. Their startup sequence is very different. You can also emulate their startup behavior -- they are actually doing a total restart of the debug after reset, not sure why or what was in your startup commands. You can emulate the same with Cortex-Debug -- see the various pre/post/override commands in our docs https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

```
042,160 ~"0x00223104 in ?? ()\n"
042,160 *stopped,frame={addr="0x00223104",func="??",args=[],arch="armv7e-m"},thread-id="1",stopped-t\
hreads="all"
042,160 22^connected
042,160 (gdb) 
042,160 &"\n"
042,161 ^done
042,161 (gdb) 
042,164 23monitor reset
042,164 &"monitor reset\n"
042,282 @"Resetting target\r\n"
042,282 23^done
```

mastupristi, 2023-02-10:
> You are hitting a breakpoint before main
>
> ```
> -> ~"2 received signal SIGTRAP, Trace/breakpoint trap.\n"
> 2 received signal SIGTRAP, Trace/breakpoint trap.
> -> ~"[Switching to Thread 536951996]\n" [...]

The address 0x00223104 is inside the micro's ROM (it's the RomBootloader). No one puts a breakpoint there.

> Try setting a breakpoint manually in main and remove runToEntryPoint and it might work.

I modified the launch.json like this (I also added the number of cores):

```
        {
            "request": "launch",
            "type": "cortex-debug",
            "name": "AmaOnEVK Start",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/FW/image/AmaOnEVK.elf",
            "servertype": "jlink",
            "device": "MIMXRT1176xxxA_M7",
            "interface": "swd",
            "rtos": "FreeRTOS",
            "svdFile": "${workspaceRoot}/FW/bsp/device/MIMXRT1175_cm7.svd",
            "numberOfProcessors": 2,
            "targetProcessor": 0,
            "preLaunchTask": "build",
            "showDevDebugOutput": "raw",
            "preLaunchCommands": [
                "directory ${workspaceRoot}/FW"
            ]
        },
```

The result is the same. The breakpoint put by hand in the main is never hit.
I am attaching the gdb trace before pressing the "pause" button and after pressing it.

best regards
Max

[gdb_cortex_debug_after_pause.txt](https://github.com/Marus/cortex-debug/files/10706970/gdb_cortex_debug_after_pause.txt)
[gdb_cortex_debug_before_pause.txt](https://github.com/Marus/cortex-debug/files/10706971/gdb_cortex_debug_before_pause.txt)

haneefdm, 2023-02-10:
> The address 0x00223104 is inside the micro's ROM (it's the RomBootloader). No one puts a breakpoint there.

You don't have to put a breakpoint. An exception/fault will look like a breakpoint to gdb. That is what a SIGTRAP is.

The processor count doesn't matter to JLink. It looks at each core like a separate processor 

You can run these commands(JLinkGDBServerCLExe and gdb) in a terminal command-line (separate terminals) and see for yourself. If you cannot do a proper reset, load your program, reset, set breakpoint in main, continue, then neither can we.

As I said before, in MCUXpressoIde, it was doing multiple resets. More importantly, why is that running a different program than cortex-debug. It has different sections/sizes, lot is different. We can handle axf files too as they are just elf files. So, how do I compare the two runs.

```
cortex-debug: "/home/max/Lavori/ama/srcs/repos/Ama-on-EVK/FW/image/AmaOnEVK.elf"
MCUXpressoIde:  "/home/max/Lavori/ama/wkps_startup_on_rt117x/evkmimxrt1170_freertos_hello_c\m7/Debug/evkmimxrt1170_freertos_hello_cm7.axf"
```

Please make sure you understand your startup sequence, even if you have to copy it from MCUXpressoIde, then do the same here. We have zero experience with this device.

haneefdm, 2023-02-15:
Closing this as I don't see any issue with the extension. If it is a startup issue specific to the device, then it needs to be figured out by the user. We can't

closed 2023-02-15 by haneefdm